espressif/arduino-esp32 #7136: Implementing ESP-IDF LEDC fading functions

davorvr, 2022-08-16:
### Related area

LEDC, PWM

### Hardware specification

ESP32

### Is your feature request related to a problem?

Aside from LED fading, it would be useful for adjusting servos more slowly.

### Describe the solution you'd like

I would like `[esp32-hal-ledc.c](https://github.com/espressif/arduino-esp32/blob/master/cores/esp32/esp32-hal-ledc.c)` to also include functions to wrap the existing fading functionality present in ESP-IDF's [`ledc.c` driver](https://github.com/espressif/esp-idf/blob/master/components/driver/ledc.c)

### I have checked existing list of Feature requests and the Contribution Guide

- [X] I confirm I have checked existing list of Feature requests and Contribution Guide.

davorvr, 2022-08-17:
Thanks guys, looking forward to 2.1.0! 😄

VojtechBartoska, 2022-08-18:
Hello @davorvr, yeah, we would like to implement this feature in the next major milestone 2.1.0.

P-R-O-C-H-Y, 2023-08-30:
Closing this as Fade implementation PR #8338 was already merged.

P-R-O-C-H-Y, 2023-08-30:
Closing this as Fade implementation PR https://github.com/espressif/arduino-esp32/pull/8338 was already merged.

PR #8338 (2023-06-29): LEDC Fade implementation
## Description of Change
Implemented LEDC Fade functionality into LEDC driver. 

There are 3 new APIs which can be used:
```
bool ledcFade(uint8_t pin, uint32_t start_duty, uint32_t target_duty, int max_fade_time_ms);`
bool ledcFadeWithInterrupt(uint8_t pin, uint32_t start_duty, uint32_t target_duty, int max_fade_time_ms, void (*userFunc)(void));
bool ledcFadeWithInterruptArg(uint8_t pin, uint32_t start_duty, uint32_t target_duty, int max_fade_time_ms, void (*userFunc)(void*), void * arg);
```

Added new example `LEDFade.ino` for demonstrating hardware LEDC fade.
Also  [... cut]
closed 2023-08-30 by P-R-O-C-H-Y