espressif/arduino-esp32 #7137: Wire displaying error messages when using I2C

i164j9, 2022-08-16:
### Board

ESP wroom 32 dev kit v1

### Device Description

This is probably unrelated... but previously https://github.com/platformio/platform-espressif32/issues/786 

I have not worked on my project in some time until today. As usual when I start vscode everything updates including platformio.
I am not sure if there is any relevancy to PIO, Espressif, Jeff Rowberg. After uploading my freshly compiled firmware I now get these repeating error messages
from Wire.cpp when I call initialize() from the MPU6050 library which makes calls to i2cdev.

***console output***
[581391][E][Wire.cpp:422] beginTransmission(): could not acquire lock
[581392][E][Wire.cpp:526] write(): NULL TX buffer pointer
[581392][E][Wire.cpp:448] endTransmission(): NULL TX buffer pointer
[581398][E][Wire.cpp:481] requestFrom(): NULL buffer pointer

My includes presently are:
#include "Arduino.h"
#include "MPU6050_6Axis_MotionApps20.h"
#include <WiFi.h>
#include <esp_now.h>
#include "Preferences.h"

These are the different variations of includes I have tried:
#include "Arduino.h"
#include "I2Cdev.h"
#include "Wire.h"
#include "MPU6050_6Axis_MotionApps20.h"
#include <WiFi.h>
#include <esp_now.h>
#include "Preferences.h"

***************************

#include "Arduino.h"
#include "Wire.h"
#include "MPU6050_6Axis_MotionApps20.h"
#include <WiFi.h>
#include <esp_now.h>
#include "Preferences.h"

All of the above variations have the same result because i2cdev is included in MPU6050.h and Wire.h is included in i2cdev.h. 

when I attempt to compile and upload via the arduino ide everything works and i have no errors. when i change my platformio.ini
;platform = https://github.com/platformio/platform-espressif32.git
;platform = espressif32
;platform = espressif32@3.4.0
platform = espressif32@3.5.0
I no longer get any errors and the IMU sensor does not work. it fails to initialize using 3.4.0 or 3.5.0.
The other two show the looped error messages from Wire.cpp

So this is where I am stuck. and unsure where to post this issue. so this is my starting point.
Below is all of the information I could collect to include:

I am using an esp wroom 32 dev kit v1
platformio dev 6.1.5a1
Windows 10 10.0.19044
vscode 1.70.1

***package list***
PS C:\Users\dev\Documents\PlatformIO\Projects\ESPNOW> pio pkg list
Resolving esp32doit-devkit-v1 dependencies...       
Platform espressif32 @ 5.1.0 (required: espressif32)
├── framework-arduinoespressif32 @ 3.20004.0 (required: platformio/framework-arduinoespressif32 @ ~3.20004.0)      
├── tool-esptoolpy @ 1.30300.0 (required: platformio/tool-esptoolpy @ ~1.30300.0)
├── tool-mkfatfs @ 2.0.1 (required: platformio/tool-mkfatfs @ ~2.0.0)
├── tool-mklittlefs @ 1.203.210628 (required: platformio/tool-mklittlefs @ ~1.203.0)
├── tool-mkspiffs @ 2.230.0 (required: platformio/tool-mkspiffs @ ~2.230.0)
├── tool-openocd-esp32 @ 2.1100.20220706 (required: platformio/tool-openocd-esp32 @ ~2.1100.0)
└── toolchain-xtensa-esp32 @ 8.4.0+2021r2-patch3 (required: espressif/toolchain-xtensa-esp32 @ 8.4.0+2021r2-patch3)
additionally MPU6050 and i2cdev by jeff rowberg https://github.com/jrowberg/i2cdevlib/tree/master/Arduino/MPU6050

***platformio.ini***
[env:esp32doit-devkit-v1]
platform = https://github.com/platformio/platform-espressif32.git
;platform = espressif32
;extra_scripts = post:extra_script.py
;platform = espressif32@3.4.0
board = esp32doit-devkit-v1
framework = arduino
monitor_speed = 115200
monitor_filters = esp32_exception_decoder, colorize

;***esp-prog settings***
; debug_tool = esp-prog
; upload_protocol = esp-prog
; debug_init_break = tbreak setup
; debug_speed = 2000

### Hardware Configuration

GPIO pins 22,21 are the only pins connected on the board

### Version

v2.0.4

### IDE Name

platformio

### Operating System

windows 10

### Flash frequency

40mhz

### PSRAM enabled

no

### Upload speed

115200

### Description

Wire has a series of error messages displayed when attempting to use I2C

### Sketch

```cpp
#include "Arduino.h"
#include "MPU6050_6Axis_MotionApps20.h"
#include <WiFi.h>
#include <esp_now.h>
#include "Preferences.h"
MPU6050_6Axis_MotionApps20 gyro(0x68);
void setup() {
  Serial.begin(115200);
  while(!Serial){};
  delay(100);
  Serial.println(WiFi.macAddress());
  
void setupGyro()
{
  gyro.initialize();
  if (gyro.testConnection())
  {
    Serial.println("gyro connection successful...");
  } else {
    Serial.print("gyro connection failed...\n");
    Serial.print("gyro will be disabled. check your sensor or your connections.");
  }
}
```

### Debug Message

```plain
no crashes.
[581391][E][Wire.cpp:422] beginTransmission(): could not acquire lock
[581392][E][Wire.cpp:526] write(): NULL TX buffer pointer
[581392][E][Wire.cpp:448] endTransmission(): NULL TX buffer pointer
[581398][E][Wire.cpp:481] requestFrom(): NULL buffer pointer
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

i164j9, 2022-08-28:
Problem solved. I blew away the drive partitions and started fresh. Things are back to normal.

closed 2022-08-28 by i164j9

AbuBakar-WEB1840, 2024-04-06:
In setup you missed "Wire.begin()"