espressif/arduino-esp32 #7139: Maybe someone should make the old download link invalid, remove the version pointed to!

ZinggJM, 2022-08-16:
### Board

all

### Device Description

any

### Hardware Configuration

irrelevant

### Version

v2.0.4

### IDE Name

Arduino IDE

### Operating System

any

### Flash frequency

any

### PSRAM enabled

no

### Upload speed

irrelevant

### Description

Some users are still reporting issues, because they are not notified about having the outdated version.
See https://github.com/espressif/arduino-esp32/issues/5043

### Sketch

```cpp
irrelevant
```

### Debug Message

```plain
irrelevant
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

ZinggJM, 2022-08-16:
See also: https://forum.arduino.cc/t/waveshare-e-paper-displays-with-spi/467865/3115

VojtechBartoska, 2022-08-16:
Hello @ZinggJM, I'm not sure if I understand. Which exactly link is outdated? Thanks for answer

ZinggJM, 2022-08-16:
Hello @VojtechBartoska. The download link that was earlier published in https://github.com/arduino/Arduino/wiki/Unofficial-list-of-3rd-party-boards-support-urls for Arduino IDE preferences still points to a download of version 1.0.6.
Users that have this link in preferences will never notice that it is outdated.
If this outdated link would point to something invalid, they might notice.

VojtechBartoska, 2022-08-17:
@ZinggJM Thanks for explanation. We will investigate this.

Parsaabasi, 2025-01-09:
Hello, 

I close this report since it contains the release that we no longer support. Please try the newer version and in case the issue reoccured, feel free to reopen it. 

Thanks

closed 2025-01-09 by Parsaabasi