espressif/arduino-esp32 #7142: serial problem

savagerex, 2022-08-17:
### Board

ESP32 S3

### Device Description

ESP32-S3-DevKitC-1

### Hardware Configuration

none

### Version

v2.0.3

### IDE Name

Arduino IDE

### Operating System

Windows 10

### Flash frequency

80Mhz

### PSRAM enabled

yes

### Upload speed

921600

### Description

 i write the program. (as below)
if i transmit one data ( 18byte ).I can receive more than 4000 pieces of data in one second.
but if i transmit one package ( 18 byte , 18byte ........ X10 + '\r' + '\n' = 182) . I can  only receive more than 3300 pieces of data in one second.

it lost  700 data. i don't understand.

### Sketch

```cpp
#include "SPI.h"
#include "BoardConfig.h"
#include "Sensor.h"

const char Version[]="1.0.0";
const char CompileDate[]= __DATE__ " " __TIME__;

const int TEMP2 = 0x06;
const int TEMP1 = 0x07;

long t[9];
double c;
//long values[20];
uint8_t values[20];
uint8_t DeviceId[2];
double x, y, z;
//uint8_t ui8txLogBuf[40];
uint16_t i, j, k, m;
uint8_t* ui8txLogBuf;
int DryFlag = 1;

#define PACKAGE 180    //180

unsigned long StartTime;
unsigned long EndTime;

const int DEVID_AD = 0x00;
const int XDATA3 = 0x08;
const int XDATA2 = 0x09;
const int XDATA1 = 0x0A;
const int YDATA3 = 0x0B;
const int YDATA2 = 0x0C;
const int YDATA1 = 0x0D;
const int ZDATA3 = 0x0E;
const int ZDATA2 = 0x0F;
const int ZDATA1 = 0x10;
const int RANGE = 0x2C;
const int POWER_CTL = 0x2D;

// Device values
const int RANGE_2G = 0x01;
const int RANGE_4G = 0x02;
const int RANGE_8G = 0x03;
const int MEASURE_MODE = 0x00;  //0x06; // Only accelerometer

// Operations
const int READ_BYTE = 0x01;
const int WRITE_BYTE = 0x00;

// Pins used for the connection with the sensor
const int CHIP_SELECT_PIN = 10;  //7

uint8_t ascii_trans(uint8_t ui8Code) {
  if((ui8Code == 0 || ui8Code > 0) && ui8Code <= 9){
    ui8Code += 48;  //0x30;
  }else if(ui8Code >= 10 && ui8Code <= 15){
    ui8Code += 55;  //0x37;
  }
  return ui8Code;
}

void IRAM_ATTR DryIsr() {
     DryFlag = 1; 
}

void setup() {
  // put your setup code here, to run once:
  //Serial.begin(115200);
  Serial.begin(1536000);
  SPI.begin(GPIO_OUTPUT_SCLK, GPIO_INPUT_MISO, GPIO_OUTPUT_MOSI, GPIO_OUTPUT_CS);
  SPI.beginTransaction(SPISettings(spiClk, MSBFIRST, SPI_MODE0));

  ui8txLogBuf = (uint8_t*)ps_calloc(1000, sizeof(uint8_t));
  // Initalize the  data ready and chip select pins:
  pinMode(CHIP_SELECT_PIN, OUTPUT);

  pinMode(GPIO_LED, OUTPUT);

  //Configure ADXL355:
  writeRegister(RANGE, RANGE_2G); // 2G
  writeRegister(POWER_CTL, MEASURE_MODE); // Enable measure mode

  DeviceId[0] = readRegistry(DEVID_AD);
  Serial.print("DeviceId = ");
  Serial.println(DeviceId[0],HEX);

  // Give the sensor time to set up:
  delay(100);

  attachInterrupt(DRY, DryIsr, RISING);
}

void loop() {

//  t[0] = readRegistry(TEMP2);
//  t[1] = readRegistry(TEMP1);
//
//  c=(t[0]<<8)+t[1];
//  c=((1852 - c)/9.05)+25.7;
//
//  Serial.print("temperature=");
//  Serial.print(c);
//  Serial.print("\t");

  if(DryFlag == 1) {
    DryFlag = 0;
  if(m == 0)
  {
     StartTime = millis();
  }
//  values[0] = readRegistry(XDATA3);
//  values[1] = readRegistry(XDATA2);
//  values[2] = readRegistry(XDATA1);
//  values[3] = readRegistry(YDATA3);
//  values[4] = readRegistry(YDATA2);
//  values[5] = readRegistry(YDATA1);
//  values[6] = readRegistry(ZDATA3);
//  values[7] = readRegistry(ZDATA2);
//  values[8] = readRegistry(ZDATA1);
  
//  for(i = 0; i < 9; i++) {
//      ui8txLogBuf[j] = ascii_trans((values[i] >> 4));
//      ui8txLogBuf[j+1] = ascii_trans((values[i] & 0x0F));
//      j+=2;
//  }

  for(i = 0; i < 9; i++) {
      ui8txLogBuf[j] = '0';
      ui8txLogBuf[j+1] = '0';
      j+=2;
  }

  if(j == PACKAGE) {
               ui8txLogBuf[j] = '\n';
               ui8txLogBuf[j+1] = '\r';
               Serial.write(ui8txLogBuf, 182);   //182
               m+=1;
               j=0;
            }
  }

  if (m==400){     //400
         EndTime = millis();
         Serial.print("**************UART Transfer Time*************");
         Serial.println(EndTime-StartTime);
         m=0;
  }
  

  
 

  // Next data in 100 milliseconds
  //delay(100);
  //delayMicroseconds(100);
}

void writeRegister(byte thisRegister, byte thisValue) {
  byte dataToSend = (thisRegister << 1) | WRITE_BYTE;
  digitalWrite(CHIP_SELECT_PIN, LOW);
  SPI.transfer(dataToSend);
  SPI.transfer(thisValue);
  digitalWrite(CHIP_SELECT_PIN, HIGH);
}

/* 
 * Read registry in specific device address
 */

unsigned char readRegistry(byte thisRegister) {
  unsigned char result = 0;
  byte dataToSend = (thisRegister << 1) | READ_BYTE;

  digitalWrite(CHIP_SELECT_PIN, LOW);
  SPI.transfer(dataToSend);
  result = SPI.transfer(0x00);
  digitalWrite(CHIP_SELECT_PIN, HIGH);
  return result;
}

//unsigned int readRegistry(byte thisRegister) {
//  unsigned int result = 0;
//  byte dataToSend = (thisRegister << 1) | READ_BYTE;
//
//  digitalWrite(CHIP_SELECT_PIN, LOW);
//  SPI.transfer(dataToSend);
//  result = SPI.transfer(0x00);
//  digitalWrite(CHIP_SELECT_PIN, HIGH);
//  return result;
//}

/* 
 * Read multiple registries
 */
void readMultipleData(int *addresses, int dataSize, int *readedData) {
  digitalWrite(CHIP_SELECT_PIN, LOW);
  for(int i = 0; i < dataSize; i = i + 1) {
    byte dataToSend = (addresses[i] << 1) | READ_BYTE;
    SPI.transfer(dataToSend);
    readedData[i] = SPI.transfer(0x00);
  }
  digitalWrite(CHIP_SELECT_PIN, HIGH);
```

### Debug Message

```plain
none
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

SuGlider, 2022-08-17:
@savagerex - Could you please try it with Arduino Core 2.0.4 and let me know the results? Thanks.
Please provide more details about the external SPI and UART peripherals connected to the ESP32-S3.

SuGlider, 2022-08-17:
It seems that some external sensor is raising GPIO pin name DRY. What is that exactly. How does it work. Please provide more information. Thanks.

savagerex, 2022-08-17:
I rearrange the program. as below.

MCU(ESP S3) send data ('0') by serial(uart).
PC(computer) receive the data.

you can try this code in esp32 S3 board.

#include "SPI.h"
#include "BoardConfig.h"
#include "Sensor.h"

const char Version[]="1.0.0";
const char CompileDate[]= __DATE__ " " __TIME__;

const int TEMP2 = 0x06;
const int TEMP1 = 0x07;

long t[9];
double c;
//long values[20];
uint8_t values[20];
uint8_t DeviceId[2];
double x, y, z;
//uint8_t ui8txLogBuf[40];
uint16_t i, j, k, m;
uint8_t* ui8txLogBuf;
int DryFlag = 1;

#define PACKAGE 180    //180

unsigned long StartTime;
unsigned long EndTime;

const int DEVID_AD = 0x00;
const int XDATA3 = 0x08;
const int XDATA2 = 0x09;
const int XDATA1 = 0x0A;
const int YDATA3 = 0x0B;
const int YDATA2 = 0x0C;
const int YDATA1 = 0x0D;
const int ZDATA3 = 0x0E;
const int ZDATA2 = 0x0F;
const int ZDATA1 = 0x10;
const int RANGE = 0x2C;
const int POWER_CTL = 0x2D;

// Device values
const int RANGE_2G = 0x01;
const int RANGE_4G = 0x02;
const int RANGE_8G = 0x03;
const int MEASURE_MODE = 0x00;  //0x06; // Only accelerometer

// Operations
const int READ_BYTE = 0x01;
const int WRITE_BYTE = 0x00;

// Pins used for the connection with the sensor
const int CHIP_SELECT_PIN = 10;  //7

uint8_t ascii_trans(uint8_t ui8Code) {
  if((ui8Code == 0 || ui8Code > 0) && ui8Code <= 9){
    ui8Code += 48;  //0x30;
  }else if(ui8Code >= 10 && ui8Code <= 15){
    ui8Code += 55;  //0x37;
  }
  return ui8Code;
}

void IRAM_ATTR DryIsr() {
   
}

void setup() {
  // put your setup code here, to run once:
  Serial.begin(1536000);

  ui8txLogBuf = (uint8_t*)ps_calloc(1000, sizeof(uint8_t));
  // Initalize the  data ready and chip select pins:
//
//  DryFlag = 1; 

}

void loop() {

//if (DryFlag==1){

//  if(m == 0)
//  {
//     StartTime = millis();
//  }
                
  for(i = 0; i < 9; i++) {
      ui8txLogBuf[j] = '0';
      ui8txLogBuf[j+1] = '0';
      j+=2;
  }   
  if(j == PACKAGE) {
               ui8txLogBuf[j] = '\n';
               ui8txLogBuf[j+1] = '\r';
               Serial.write(ui8txLogBuf, 182);   //182
               while (Serial.availableForWrite() <= 0) {}
               m+=1;
               j=0;
            }        

//  if (m>=400){     //400
//         EndTime = millis();
//         Serial.print("**************UART Transfer Time*************");
//         Serial.println(EndTime-StartTime);
//         m=0;
//         
//      DryFlag = 0;
//  }
 }
//}

void writeRegister(byte thisRegister, byte thisValue) {
  byte dataToSend = (thisRegister << 1) | WRITE_BYTE;
  digitalWrite(CHIP_SELECT_PIN, LOW);
  SPI.transfer(dataToSend);
  SPI.transfer(thisValue);
  digitalWrite(CHIP_SELECT_PIN, HIGH);
}

/* 
 * Read registry in specific device address
 */

unsigned char readRegistry(byte thisRegister) {
  unsigned char result = 0;
  byte dataToSend = (thisRegister << 1) | READ_BYTE;

  digitalWrite(CHIP_SELECT_PIN, LOW);
  SPI.transfer(dataToSend);
  result = SPI.transfer(0x00);
  digitalWrite(CHIP_SELECT_PIN, HIGH);
  return result;
}

//unsigned int readRegistry(byte thisRegister) {
//  unsigned int result = 0;
//  byte dataToSend = (thisRegister << 1) | READ_BYTE;
//
//  digitalWrite(CHIP_SELECT_PIN, LOW);
//  SPI.transfer(dataToSend);
//  result = SPI.transfer(0x00);
//  digitalWrite(CHIP_SELECT_PIN, HIGH);
//  return result;
//}

/* 
 * Read multiple registries
 */
void readMultipleData(int *addresses, int dataSize, int *readedData) {
  digitalWrite(CHIP_SELECT_PIN, LOW);
  for(int i = 0; i < dataSize; i = i + 1) {
    byte dataToSend = (addresses[i] << 1) | READ_BYTE;
    SPI.transfer(dataToSend);
    readedData[i] = SPI.transfer(0x00);
  }
  digitalWrite(CHIP_SELECT_PIN, HIGH);
}

SuGlider, 2022-08-18:
@savagerex - 
Please provide the minimum code necessary to demonstrate the issue. 
Please also explain how the issue can be seen and what is the expected output/execution.

At this point, I can't reproduce the issue.

savagerex, 2022-08-18:
ok!!! 
minimum code as below and run this code in esp32 S3 board.

you can install "tera term"on PC(computer). 
you can see it.  remember choose "timestamp" in "tera term".
you can see and "tera term" receive the data(0).

uint16_t i, j, k, m;
uint8_t* ui8txLogBuf;

#define PACKAGE 180    //180

void setup() {
  // put your setup code here, to run once:
  Serial.begin(1536000);
  ui8txLogBuf = (uint8_t*)ps_calloc(1000, sizeof(uint8_t));
}

void loop() {

  for(i = 0; i < 9; i++) {
      ui8txLogBuf[j] = '0';
      ui8txLogBuf[j+1] = '0';
      j+=2;
  }

  if(j == PACKAGE) {
               ui8txLogBuf[j] = '\n';
               ui8txLogBuf[j+1] = '\r';
               Serial.write(ui8txLogBuf, PACKAGE+2);   //182
               m+=1;
               j=0;
            }

}

SuGlider, 2022-08-18:
@savagerex - It runs fine... I see a lot of Zeros.
What is the issue that you have with this sketch?

savagerex, 2022-08-18:
this program, You can receive about three thousand in one second.

when you change the program,  #define PACKAGE 18
You can receive about four thousand in one second.

there is different in #define PACKAGE 18 & #define PACKAGE 180.

i means when i send the data. 

i wish #define PACKAGE 180. it can receive about four thousand in one second.

savagerex, 2022-08-18:
this code. 18byte is one data.  
#define PACKAGE 18 , means (18 +2('\r'+'\n'). 
#define PACKAGE 180 , means (18, 18,18.......X10 +2('\r'+'\n').

savagerex, 2022-08-18:
#define PACKAGE 180, this program, you can see. 335 X 10 = 3350.   about three thousand in one second.

[image]

savagerex, 2022-08-18:
#define PACKAGE 18, if this program use dry pin , it will  about four thousand in one second.

savagerex, 2022-08-18:
#define PACKAGE 18, if this program and don't use dry pin , it will about eight thousand in one second.

I mean when #define PACKAGE 180. it should be receive about eight thousand in one second.

the issue is why less???

SuGlider, 2022-08-18:
@savagerex - OK, I see the issue. 
A possible solution for that is to increase the TX Buffer and make writing to it faster.
Please use `size_t HardwareSerial::setTxBufferSize(size_t new_size)` function right before `begin(15360000)`
This change of Buffer Size will not affect `size_t HardwareSerial::availableForWrite()`, which in version 2.0.3, has an issue and will return some number from 0 to 128. This issue has been fixed within the latest Arduino Core from Github Master branch.

`setTxBufferSize()` usage Example:
``` cpp
void setup() {
  // put your setup code here, to run once:
  Serial.setTxBufferSize(1024);                    // <<<<==== must be set before Serial.begin()
  Serial.begin(1536000);
  ui8txLogBuf = (uint8_t*)ps_calloc(1000, sizeof(uint8_t));
}
```

savagerex, 2022-08-18:
i modified the code , as below. it does not work.

void setup() {
  // put your setup code here, to run once:
  Serial.setTxBufferSize(8192);
  Serial.begin(1536000);
  ui8txLogBuf = (uint8_t*)ps_calloc(1000, sizeof(uint8_t));

//  int wlen = Serial.availableForWrite();
//  Serial.println(wlen);
}

[image]

355 X 10 = 3550.  it still receive about three thousand in one second.

if it is possible, you can try this code.

savagerex, 2022-08-18:
//  int wlen = Serial.availableForWrite();
//  Serial.println(wlen);

when i print "wlen".
it is 128

savagerex, 2022-08-19:
I have updated to version 2.0.4

SuGlider, 2022-08-19:
> //  int wlen = Serial.availableForWrite();
> //  Serial.println(wlen);
> 
> when i print "wlen".
> it is 128

This is fixed only in the master branch.

2.0.4 doesn't fix it. It depends on a new IDF function that was added few time ago.

This fix will be added to 2.0.5.

SuGlider, 2022-08-19:
I have tested it and I get 153,600 bytes out per second.

SuGlider, 2022-08-19:
Try this sketch:

``` cpp
uint16_t i, j, k, m;
uint8_t* ui8txLogBuf;

#define PACKAGE 180 //180

void setup() {
  // put your setup code here, to run once:
  Serial.begin(1536000);
  //Serial.begin(115200);
  ui8txLogBuf = (uint8_t*)ps_calloc(1000, sizeof(uint8_t));
  if (!ui8txLogBuf) Serial.println("\n=========\n\nERROR\n\r");
  else Serial.println("\n=========\n\nOK\n\r");
  delay(1000);
}

void loop() {
  static uint32_t now = millis();
  static uint32_t counter = 0;

  for (i = 0; i < 9; i++) {
    ui8txLogBuf[j] = '0';
    ui8txLogBuf[j + 1] = '0';
    j += 2;
  }
  //  Serial.println(j);
  if (j == PACKAGE) {
    ui8txLogBuf[j] = '\n';
    ui8txLogBuf[j + 1] = '\r';
    Serial.write(ui8txLogBuf, PACKAGE + 2); //182
    counter += PACKAGE + 2;
    m += 1;
    j = 0;
  }
  //  delay(125);
  if (millis() - now >= 1000) {
    Serial.printf("\n\r\n\rTIME: %ld -- #Chars = %d\n\n\r", millis() - now, counter);
    delay(2000);   // time to see this message in the Serial Monitor.
    now = millis();
    counter = 0;
  }
}
```

SuGlider, 2022-08-19:
If the sketch waits to send all data (blocking write) by using a `Serial.flush()` right after `Serial.write()`, it can still send 143,234 bytes in One Second.

``` cpp
  if (j == PACKAGE) {
    ui8txLogBuf[j] = '\n';
    ui8txLogBuf[j + 1] = '\r';
    Serial.write(ui8txLogBuf, PACKAGE + 2); //182
    Serial.flush();                                // <<<<<<<<<<<==== Only returns when all bytes are sent out and UART FIFO is empty
    counter += PACKAGE + 2;
    m += 1;
    j = 0;
  }

```

SuGlider, 2022-08-19:
As you can see... This is not using `Serial.setTxBufferSize()`.
So, it seems that I just can't reproduce the issue you are reporting.

SuGlider, 2022-08-19:
Using `Serial.setTxBufferSize(4096)` and no blocking write (not using `Serial.flush()`), I get 157,248 bytes per second...

savagerex, 2022-08-21:
This method is .
do you have any other way ???

i need  receive about eight thousand in one second.

SuGlider, 2022-08-22:
> This method is . do you have any other way ???
> 
> i need receive about eight thousand in one second.

The message seems incomplete. Please clarify your need.

savagerex, 2022-08-22:
Using Serial.setTxBufferSize(4096). it does not work.
do you have any other way to solve???

i need PC(computer)receive about eight thousand in one second.

SuGlider, 2022-08-22:
I have read again the whole issue and I see that the sketch writes the ADXL355 data to UART only when DATA_READY (DRDY) pin raises.

Thus, the number of bytes it will write to UART depends on the frequency that the ADXL355 will make its data available for being read by raising DRDY pin.

It has nothig to do with UART/Serial issue. It is related to how the ADXL355 works.
The ADXL355 datasheet says that it must read the data from FIFO in order to raise the DRDY pin again. 
Your sketch doesn't do that.

Datasheet:
https://www.analog.com/media/en/technical-documentation/data-sheets/adxl354_adxl355.pdf

I suggest you this code for testing:
https://github.com/jyounnim/ESP32/blob/master/05.ADXL355/ADXL355_simple/ 

This other code (similar to the one you have posted) doesn't use DRDY to read ADXL355 data:
https://github.com/gpvidal/adxl355-arduino/blob/master/adxl355.ino

savagerex, 2022-08-22:
i ever use DATA_READY (DRDY) pin.
but result is the same. 

this code, i ruled out ADXL355.

i found esp32 send '0'.  it still receive about three thousand in one second.

so  i would doubt, maybe the uart have some problem.???

SuGlider, 2022-08-22:
I was taling about the initial Sketch posted in this issue.

About the last simplified code posted here, did you try this code:
https://github.com/espressif/arduino-esp32/issues/7142#issuecomment-1220577690

Clould you please send me the Log Output when running this code?

This is the output I get here with my S3 (using UART and not CDC):

#### with 115200 (right after boot):
```
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x1 (POWERON),boot:0x8 (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fce3808,len:0x43c
load:0x403c9700,len:0xbec
load:0x403cc700,len:0x2a3c
SHA-256 comparison failed:
Calculated: 3611f330726ad4edf64f62982b0568332a5800bc59425cab90be5379db92c108
Expected: ffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffff
Attempting to boot anyway...
entry 0x403c98d8
```

#### with 1536000 (right after booting):

===> The sketch says it has sent 153,608 bytes in 1 second!

```

=========

         OK

000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000
000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000
000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000
000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000
000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000
000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000
000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000
000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000
000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000
000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000
000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000
000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000
000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000
000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000
000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000
000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000
000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000
000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000
000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000
000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000
000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000
000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000
000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000
000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000
000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000
[... cut ...]
000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000
000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000
000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000

TIME: 1000 -- #Chars = 153608

```

SuGlider, 2022-08-22:
Please try the code from https://github.com/espressif/arduino-esp32/issues/7142#issuecomment-1220577690
and then post the output of your S3 as I did above.
Thanks.

savagerex, 2022-08-23:
you can see time stamp between ( 48.809 ~ 50.870).
the time is about 2  second.

[2022-08-23 11:08:48.809] 000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000
[2022-08-23 11:08:48.809] 000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000
[2022-08-23 11:08:48.809] 000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000
[2022-08-23 11:08:48.809] 000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000
[2022-08-23 11:08:48.809] 000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000
[2022-08-23 11:08:48.809] 000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000
[2022-08-23 11:08:48.824] 000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000
[2022-08-23 11:08:48.824] 000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000
[2022-08-23 11:08:48.824] 000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000
[2022-08-23 11:08:48.824] 000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000
[2022-08-23 11:08:48.824] 000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000
[2022-08-23 11:08:48.824] 000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000
[2022-08-23 11:08:48.840] 000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000
[2022-08-23 11:08:48.840] 000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000
[2022-08-23 11:08:48.840] 000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000
[2022-08-23 11:08:48.840] 000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000
[2022-08-23 11:08:48.840] 000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000
[2022-08-23 11:08:48.862] 000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000
[2022-08-23 11:08:48.862] 000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000
[2022-08-23 11:08:48.862] 000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000
[2022-08-23 11:08:48.862] 000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000
[2022-08-23 11:08:48.862] 000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000
[2022-08-23 11:08:48.862] 000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000
[2022-08-23 11:08:48.878] 000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000
[2022-08-23 11:08:48.878] 000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000
[2022-08-23 11:08:48.878] 000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000
[... cut ...]
[2022-08-23 11:08:50.870] 000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000
[2022-08-23 11:08:50.870] 0000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000
[2022-08-23 11:08:50.870] 000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000
[2022-08-23 11:08:50.870] 
[2022-08-23 11:08:50.870] 
[2022-08-23 11:08:50.870] TIME: 1000 -- #Chars = 153608
[2022-08-23 11:08:50.870]

SuGlider, 2022-08-24:
@savagerex - 
I think that the issue is in the Terminal and the TimeStamp processing.... 
It may be adding some processing time?!
I used a simple `Putty` application as "Terminal" to receive the Serial data.... It received all the data in one second.

The data sent by the ESP32 to PC can be buffered in the PC and this 2 seconds could be, actually, some "overhead time" it takes when the Terminal program adds the time stamp to the output. (just a guess)

Another way to test it would be by using 2 ESP32-S3 (or 1 S3 and some other MCU) connected  each other.
Measure the time and quantity of bytes received directly in the second MCU, instead of the PC - using another sketch for that.

Or try it with no Time Stamp or try another Terminal software.... 
I really think that the ESP32S3 sends the data in the right time, but the PC Program (Terminal) doesn't deal well with it...

SuGlider, 2022-08-24:
I have just have run the sketch for 2 minutes... 120 seconds and I counted how many times I saw the "TIME: 1000 -- #Chars = nnnn" times.... 
Using "Putty" as my Teminal Software, I counted 40 times. Which is correct because it takes 1 second to push all the Zeros and then it pauses for 2 seconds before starting over. Total of 3 seconds per "message" in the screen.

40 x 3 seconds per cycle = 120 seconds.... no delay.

SuGlider, 2022-08-24:
@savagerex - To add more...

>you can see time stamp between ( 48.809 ~ 50.870).
the time is about 2 second.

That means, in the worst case, that the S3 has sent 153,608 bytes in 2 seconds.... Thus: more than 75,000 bytes per second.

>This method is .
do you have any other way ???
i need receive about eight thousand in one second.

Well... problem solved. As you can see in the test you have done, the S3 has sent, in the worst case, seventy thousand in one second.

SuGlider, 2022-08-24:
@savagerex - Could you use a Data Analyzer to verify the baudrate of the ESP32-S3 UART?

savagerex, 2022-08-25:
the baudrate is 1536000. it is no problem.

SuGlider, 2022-08-25:
I think that the problem is not the ESP32S3.

It is in the PC Terminal Software.

SuGlider, 2022-08-25:
Can we close this issue?

savagerex, 2022-08-25:
OK!!  Thank you!!!

SuGlider, 2022-08-26:
You are welcome! Feel free to open any new issue you may have, we are here to help whenever we can.

closed 2022-08-26 by SuGlider