espressif/arduino-esp32 #7143: PICO-D4 fails to read flash in DIO mode

pkendall64, 2022-08-17:
### Board

ESP32 PICO Kit V4.1

### Device Description

Problem also occurs on other boards with PICO-D4 on them.

### Hardware Configuration

Nothing attached

### Version

v2.0.4

### IDE Name

PlatformIO

### Operating System

MacOS 12.5

### Flash frequency

40MHz

### PSRAM enabled

no

### Upload speed

460800

### Description

When flashing with QIO mode all is well, when flashing with DIO mode the sketch fails to talk the the built-in flash correctly.

here is the `platformio.ini` file as well
```
[env:esp32]
platform = espressif32@5.1.0
board = esp32dev
framework = arduino
board_build.partitions = min_spiffs.csv
board_build.flash_mode = dio
board_build.f_cpu = 240000000L
build_flags =
    -DTARGET_RX=1
monitor_speed = 115200
```

### Sketch

```cpp
#include <Arduino.h>
#include <SPIFFS.h>

void setup() {
  Serial.begin(115200);
  Serial.printf("Started...\n");
  Serial.printf("%ld\n", ESP.getSketchSize());
  SPIFFS.begin(true);
}

void loop() {
  delay(1000);
  Serial.println(".");
}
```

### Debug Message

```plain
rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 188777542, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:1184
load:0x40078000,len:13132
load:0x40080400,len:3036
entry 0x400805e4
Started...
E (11) esp_image: image at 0x10000 has invalid magic byte (nothing flashed here?)
0
E (13) SPIFFS: mount failed, -10025
E (1950) SPIFFS: mount failed, -10025
[  1953][E][SPIFFS.cpp:114] format(): Formatting SPIFFS failed! Error: -1
[  1953][E][SPIFFS.cpp:89] begin(): Mounting SPIFFS failed! Error: -1
.
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

Jason2866, 2022-08-17:
Platformio crew has updated the framework how the firmware is builded. The new version includes the PR (slightly modified) https://github.com/espressif/esp32-arduino-lib-builder/pull/80
To use just delete the hidden `.platformio` folder and restart VSC/Platformio.
The issue should be solved.

pkendall64, 2022-08-18:
Thanks @Jason2866 thats certainly seems to have fixed the issue.
I was a little worried that all the devices would need to updated via USB/UART rather than OTA, which would have been a right pain.

closed 2022-08-18 by pkendall64