espressif/arduino-esp32 #7145: Wifi.softAPgetStationNum() does not work if no client is connect?

Dork57, 2022-08-17:
### Board

ESP32 DEV Kit 1

### Device Description

ESP32 DEV Kit 1

### Hardware Configuration

no

### Version

v2.0.4

### IDE Name

PlattformIo

### Operating System

Windows 10

### Flash frequency

40Mhz

### PSRAM enabled

no

### Upload speed

115200

### Description

If the ESP32 Wifi run in 
`WiFi.mode(WIFI_AP_STA);`

and the PC is connected, the function `WiFi.softAPgetStationNum()` returns 1.
But if the PC disconnected it returns furthermore 1.

Can anybody check this? 

### Sketch

```cpp
Serial.printf("Connections: %u\r\n", WiFi.softAPgetStationNum());
```

### Debug Message

```plain
.
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

svdrummer, 2022-12-30:
december 2022

As a temp work arround, if I did something like  december 2022
int a = WiFi.softAPgetStationNum();
a = a -1;
Serial.print("Stations connected = ");
Serial.println (a);

Dork57, 2023-01-02:
I tested the Issue again in Version 2.0.6 with idf 4.4.3 (Arduino as part in idf). 
Now, the issue is gone.

mrengineer7777, 2023-04-13:
@Dork57 Thanks for the update

closed 2023-04-13 by mrengineer7777