espressif/arduino-esp32 #7148: BLE notify a single client

JulienLecoq, 2022-08-17:
### Related area

Bluetooth LE

### Hardware specification

ESP32

### Is your feature request related to a problem?

I would like to be able to notify a single client instead of notifying all clients from a Bluetooth LE server. This would allow request/response pattern without notifying clients which shouldn't be notified.

### Describe the solution you'd like

Add a parameter to the notify function of a BLE characteristic precising the client to notify.

### Describe alternatives you've considered

- Using L2CAP communication instead of GATT communication.
- Using the current notification system with the risk of clients being notified whereas they shouldn't.

### I have checked existing list of Feature requests and the Contribution Guide

- [X] I confirm I have checked existing list of Feature requests and Contribution Guide.

SuGlider, 2022-08-18:
>I would like to be able to notify a single client instead of notifying all clients from a Bluetooth LE server. 

Not sure what you mean with "to notify a single client". 
BLE is about Characteristics, which in turn can notify whoever is paying attention to it.
That said, what if your server provides 3 characteristics: A, B and C.
Each client connected to this server will "listen" to exclusively one of those 3 characteristics notification.
By this way the server will be able "to notify" a single client at a time.

Your server can also add a property to each characteristic that will tell the client if that is already "taken" by a client, forcing the client to look for the next characteristic with this property still "not taken" and then change it to "taken" to avoid other client to "use it" as "notification channel".

SuGlider, 2022-08-18:
Going out of the box... again, I don't know what is your project about, but why not use ESP-NOW?
ESP-NOW can be used to connect a Mesh Network of ESP32 boards and allow each one to communicate to the others at any time and any way. Using any sort of topology...

closed 2025-01-16 by Parsaabasi