espressif/arduino-esp32 #7149: Watchdog not working on code with buffer overflows

garageeks, 2022-08-18:
### Board

ESP32 Dev Module

### Device Description

Custom pcb with an ILI9341 SPI display

### Hardware Configuration

SPI display on SPI bus
TFT_MISO=19
TFT_MOSI=23
TFT_SCLK=18
TFT_CS=5
TFT_DC=17
TFT_RST=4

### Version

v2.0.3

### IDE Name

PlatformIO

### Operating System

Windows 10

### Flash frequency

80Mhz

### PSRAM enabled

no

### Upload speed

921600

### Description

I'm using what it looks like a buggy graphics library that leads to buffer overflows and sometimes it hangs.
I implemented standard task watchdog as in the example code below, but after a few days of running it appears to hang and doesn't restart it.
I also implemented a periodic 48h reboot with `ESP.restart();` as well as a restart if free heap memory goes below a certain threshold; as a safeguard from memory leaks and memory corruption.
I assumed the watchdog is in hardware and therefore shouldn't be perturbed by software glitches.
Could buffer overflows disable or make the task watchdog not working?

### Sketch

```cpp
#include <esp_task_wdt.h>

void setup {
   Serial.begin(115200);
   Serial.println("WDT test");
   esp_task_wdt_init(30, true);
   esp_task_wdt_add(NULL); 
}

void loop {
  esp_task_wdt_reset();
  Serial.println("Testing...");
  delay(1000);
}
```

### Debug Message

```plain
I don't have error messages: the crashes happen after days or weeks and the unit is in a remote location
```

### Other Steps to Reproduce

The task watchdog works with other firmwares I have written, however these firmwares have no buffer overflows.

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

VojtechBartoska, 2022-08-18:
Hello @garageeks, can you please retest your code on Arduino ESP32 core version 2.0.4? I I remember correctly we did some improvements to this. Thanks

garageeks, 2022-08-18:
Thank you @VojtechBartoska for the quick reply. I will compile a firmware with 2.0.4 and leave it there on the test bench running to see what happens, but it may require a long time.

I've been digging how the task watchdog works ([link](https://github.com/espressif/esp-idf/blob/master/components/esp_system/task_wdt.c)), and it looks like the watchdog itself is hardware-based, but the task monitoring is in FreeRTOS and thus vulnerable to memory corruption, am I right?
Therefore shall I place an external hardware watchdog to make it bulletproof?

SuGlider, 2022-08-21:
This is a good question. I'd say that an external watchdog would make it bulletproof, but it is an extra cost.
One possible way to solve it is to place WDT code to run in Core 0.
By default, Arduino tasks and code run in Core 1.

Try creating a task pinned to Core 0 that sets and treats the WDT. I have a feeling that it may solve the issue.

garageeks, 2022-08-31:
Hi guys, I have tested the code on 2.0.4 and it didn't hang for days, then I had to reflash it to test new features. 
I will leave a unit on and see what happens, in the meantime I'll put a hardware watchdog just to be sure, it is an additional cost, sure, but 1 USD to save a field trip is well spent.

@SuGlider it is a good idea, but isn't memory shared between the two cores? If yes, is there a way to assign protected memory in Core 0 that cannot be accessed from Core 1?

garageeks, 2023-02-27:
I thought about @SuGlider suggestion to feed WDT through a task in Core 0, but I don't think it is bulletproof, as, first, the memory between two cores is shared and there is no MMU that prevents memory overwrites (am I right?)
Additionally, the application code on Core 1 may misbehave but the unit won't be reset because the task in Code 0 is happily feeding the watchdog. This could work if we can have protected memory in Core 0 and a function in the code run by Core 1 is feeding the task in Core 0, which is in turn feeding the hardware watchdog.

In the end I simply added an external hardware watchdog with a long enough reset time that doesn't interfere with firmware uploading/OTA.

mrengineer7777, 2023-04-13:
I believe the ESP32 has watchdog monitoring that can be enabled on both cores.  I'm not aware of any memory protection on the ESP32. RAM corruption will cause unexpected behavior.  It may or may not crash the affected core.  A hardware watchdog is a wise choice.  I may add one on our next rev of PCBs.

closed 2023-04-13 by mrengineer7777