espressif/arduino-esp32 #7150: ESP32-S3 wakes up in deep sleep without touch.

SimGallery, 2022-08-18:
### Board

ESP32-S3

### Device Description

ESP32-S3-WROOM-1 (N16R8)

### Hardware Configuration

Use the gpio3 (Touch3) pin as the touch pin.

### Version

v2.0.4

### IDE Name

Arduino IDE

### Operating System

win10

### Flash frequency

QIO 120MHz

### PSRAM enabled

yes

### Upload speed

921600

### Description

Examples of deep sleep included timer, external, and touch, which worked very well on esp32. ESP32-S3 had problems with the Touchwakeup example. Even if you don't touch it, it wakes up as soon as you enter the deep sleep.

### Sketch

```cpp
//#define Threshold 1500 
#define Threshold 40

RTC_DATA_ATTR int bootCount = 0;
touch_pad_t touchPin;

void print_wakeup_reason(){
  esp_sleep_wakeup_cause_t wakeup_reason;
  wakeup_reason = esp_sleep_get_wakeup_cause();
  switch(wakeup_reason)
  {
    case ESP_SLEEP_WAKEUP_EXT0 : Serial.println("Wakeup caused by external signal using RTC_IO"); break;
    case ESP_SLEEP_WAKEUP_EXT1 : Serial.println("Wakeup caused by external signal using RTC_CNTL"); break;
    case ESP_SLEEP_WAKEUP_TIMER : Serial.println("Wakeup caused by timer"); break;
    case ESP_SLEEP_WAKEUP_TOUCHPAD : Serial.println("Wakeup caused by touchpad"); break;
    case ESP_SLEEP_WAKEUP_ULP : Serial.println("Wakeup caused by ULP program"); break;
    default : Serial.printf("Wakeup was not caused by deep sleep: %d\n",wakeup_reason); break;
  }
}

void print_wakeup_touchpad(){
  touchPin = esp_sleep_get_touchpad_wakeup_status();
  switch(touchPin)
  {
    case 0  : Serial.println("Touch detected on GPIO 4"); break;
    case 1  : Serial.println("Touch detected on GPIO 0"); break;
    case 2  : Serial.println("Touch detected on GPIO 2"); break;
    case 3  : Serial.println("Touch detected on GPIO 15"); break;
    case 4  : Serial.println("Touch detected on GPIO 13"); break;
    case 5  : Serial.println("Touch detected on GPIO 12"); break;
    case 6  : Serial.println("Touch detected on GPIO 14"); break;
    case 7  : Serial.println("Touch detected on GPIO 27"); break;
    case 8  : Serial.println("Touch detected on GPIO 33"); break;
    case 9  : Serial.println("Touch detected on GPIO 32"); break;
    default : Serial.println("Wakeup not by touchpad"); break;
  }
}

void callback(){
  //placeholder callback function
}

void setup(){
  Serial.begin(115200);
  delay(1000); //Take some time to open up the Serial Monitor

  ++bootCount;
  Serial.println("Boot number: " + String(bootCount));

  print_wakeup_reason();
  print_wakeup_touchpad();

  touchAttachInterrupt(T3, callback, Threshold);

  esp_sleep_enable_touchpad_wakeup();

  Serial.println("Going to sleep now");
  esp_deep_sleep_start();
}

void loop(){
}
```

### Debug Message

```plain
00:21:38.928 -> ESP-ROM:esp32s3-20210327
00:21:38.928 -> Build:Mar 27 2021
00:21:38.928 -> rst:0x1 (POWERON),boot:0x8 (SPI_FAST_FLASH_BOOT)
00:21:38.976 -> SPIWP:0xee
00:21:38.976 -> mode:DIO, clock div:1
00:21:38.976 -> load:0x3fce3808,len:0x43c
00:21:38.976 -> load:0x403c9700,len:0xbec
00:21:38.976 -> load:0x403cc700,len:0x2a3c
00:21:38.976 -> SHA-256 comparison failed:
00:21:38.976 -> Calculated: 74cb8a0835ff948a23b1fa30f5641bd3d3fc50d3e848150763aab2d791fb9d34
00:21:38.976 -> Expected: ffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffff
00:21:38.976 -> Attempting to boot anyway...
00:21:38.976 -> entry 0x403c98d8
00:21:40.027 -> Boot number: 1
00:21:40.027 -> Wakeup was not caused by deep sleep: 0
00:21:40.027 -> Wakeup not by touchpad
00:21:40.075 -> Going to sleep now
00:21:40.171 -> ESP-ROM:esp32s3-20210327
00:21:40.171 -> Build:Mar 27 2021
00:21:40.171 -> rst:0x5 (DSLEEP),boot:0x8 (SPI_FAST_FLASH_BOOT)
00:21:40.171 -> SPIWP:0xee
00:21:40.171 -> mode:DIO, clock div:1
00:21:40.171 -> load:0x3fce3808,len:0x43c
00:21:40.171 -> load:0x403c9700,len:0xbec
00:21:40.171 -> load:0x403cc700,len:0x2a3c
00:21:40.171 -> SHA-256 comparison failed:
00:21:40.171 -> Calculated: 74cb8a0835ff948a23b1fa30f5641bd3d3fc50d3e848150763aab2d791fb9d34
00:21:40.171 -> Expected: ffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffff
00:21:40.171 -> Attempting to boot anyway...
00:21:40.171 -> entry 0x403c98d8
00:21:41.218 -> Boot number: 2
00:21:41.218 -> Wakeup caused by touchpad
00:21:41.218 -> E (1006) TOUCH_SENSOR: touch_pad_get_wakeup_status(222): Touch channel error
00:21:41.218 -> 
00:21:41.218 -> assert failed: esp_sleep_get_touchpad_wakeup_status sleep_modes.c:931 (ret == ESP_OK && "wakeup reason is RTC_TOUCH_TRIG_EN but SENS_TOUCH_MEAS_EN is zero")
00:21:41.218 -> 
00:21:41.218 -> 
00:21:41.218 -> Backtrace:0x403771ea:0x3fce2c700x4037b211:0x3fce2c90 0x403807cd:0x3fce2cb0 0x4200d315:0x3fce2de0 0x4200150b:0x3fce2e10 0x42001603:0x3fce2e30 0x4200219a:0x3fce2e90 
00:21:41.265 -> 
00:21:41.265 -> 
00:21:41.265 -> 
00:21:41.265 -> 
00:21:41.265 -> ELF file SHA256: 0000000000000000
00:21:41.265 -> 
00:21:41.265 -> Rebooting...
00:21:41.265 -> ESP-ROM:esp32s3-20210327
00:21:41.265 -> Build:Mar 27 2021
00:21:41.265 -> rst:0xc (RTC_SW_CPU_RST),boot:0x8 (SPI_FAST_FLASH_BOOT)
00:21:41.265 -> Saved PC:0x42020262
00:21:41.265 -> SPIWP:0xee
00:21:41.265 -> mode:DIO, clock div:1
00:21:41.265 -> load:0x3fce3808,len:0x43c
00:21:41.265 -> load:0x403c9700,len:0xbec
00:21:41.265 -> load:0x403cc700,len:0x2a3c
00:21:41.265 -> SHA-256 comparison failed:
00:21:41.265 -> Calculated: 74cb8a0835ff948a23b1fa30f5641bd3d3fc50d3e848150763aab2d791fb9d34
00:21:41.312 -> Expected: ffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffff
00:21:41.312 -> Attempting to boot anyway...
00:21:41.312 -> entry 0x403c98d8
00:21:42.359 -> Boot number: 1
00:21:42.359 -> Wakeup was not caused by deep sleep: 0
00:21:42.359 -> Wakeup not by touchpad
00:21:42.407 -> Going to sleep now
00:21:42.502 -> ESP-ROM:esp32s3-20210327
00:21:42.549 -> Build:Mar 27 2021
00:21:42.549 -> rst:0x5 (DSLEEP),boot:0x8 (SPI_FAST_FLASH_BOOT)
00:21:42.549 -> SPIWP:0xee
00:21:42.549 -> mode:DIO, clock div:1
00:21:42.549 -> load:0x3fce3808,len:0x43c
00:21:42.549 -> load:0x403c9700,len:0xbec
00:21:42.549 -> load:0x403cc700,len:0x2a3c
00:21:42.549 -> SHA-256 comparison failed:
00:21:42.549 -> Calculated: 74cb8a0835ff948a23b1fa30f5641bd3d3fc50d3e848150763aab2d791fb9d34
00:21:42.549 -> Expected: ffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffff
00:21:42.549 -> Attempting to boot anyway...
00:21:42.549 -> entry 0x403c98d8
00:21:43.598 -> Boot number: 2
00:21:43.598 -> Wakeup caused by touchpad
00:21:43.598 -> E (1006) TOUCH_SENSOR: touch_pad_get_wakeup_status(222): Touch channel error
00:21:43.598 -> 
00:21:43.598 -> assert failed: esp_sleep_get_touchpad_wakeup_status sleep_modes.c:931 (ret == ESP_OK && "wakeup reason is RTC_TOUCH_TRIG_EN but SENS_TOUCH_MEAS_EN is zero")
00:21:43.598 -> 
00:21:43.598 -> 
00:21:43.598 -> Backtrace:0x403771ea:0x3fce2c700x4037b211:0x3fce2c90 0x403807cd:0x3fce2cb0 0x4200d315:0x3fce2de0 0x4200150b:0x3fce2e10 0x42001603:0x3fce2e30 0x4200219a:0x3fce2e90 
00:21:43.646 -> 
00:21:43.646 -> 
00:21:43.646 -> 
00:21:43.646 -> 
00:21:43.646 -> ELF file SHA256: 0000000000000000
00:21:43.646 -> 
00:21:43.646 -> Rebooting...
00:21:43.646 -> ESP-ROM:esp32s3-20210327
00:21:43.646 -> Build:Mar 27 2021
00:21:43.646 -> rst:0xc (RTC_SW_CPU_RST),boot:0x8 (SPI_FAST_FLASH_BOOT)
00:21:43.646 -> Saved PC:0x42020262
00:21:43.646 -> SPIWP:0xee
00:21:43.646 -> mode:DIO, clock div:1
00:21:43.646 -> load:0x3fce3808,len:0x43c
00:21:43.646 -> load:0x403c9700,len:0xbec
00:21:43.646 -> load:0x403cc700,len:0x2a3c
00:21:43.646 -> SHA-256 comparison failed:
00:21:43.646 -> Calculated: 74cb8a0835ff948a23b1fa30f5641bd3d3fc50d3e848150763aab2d791fb9d34
00:21:43.646 -> Expected: ffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffff
00:21:43.694 -> Attempting to boot anyway...
00:21:43.694 -> entry 0x403c98d8
00:21:44.744 -> Boot number: 1
00:21:44.744 -> Wakeup was not caused by deep sleep: 0
00:21:44.744 -> Wakeup not by touchpad
00:21:44.744 -> Going to sleep now
```

### Other Steps to Reproduce

I tried by modifying the variables below. But they all failed.

Threshold 40
Threshold 1500
T1 ~ T14 ( gpio 1 ~ gpio14)
Disabled all Wakeup_Source except ESP_SLEEP_WAKEUP_TOUCHPAD.

When I used the function below, the program didn't wake up from deep sleep.
esp_sleep_disable_wakeup_source(ESP_SLEEP_WAKEUP_TOUCHPAD);

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

SimGallery, 2022-08-23:
Hello. I needed a touch to wake my esp32-s3 up
I looked at the document and thought as follows. It is assumed that "ESP_PD_OPTION_AUTO" does not work on ESP32-S3. 
  Therefore, I added the following code and it worked. `esp_sleep_pd_config(ESP_PD_DOMAIN_RTC_PERIPH, ESP_PD_OPTION_ON);`
I don't know if this code works normally with low-power deep sleep. My cheap power supply shows 0.00A when deep sleep begins. If you are in urgent need of this function like me, try this method before distributing the official version. :)

```cpp
#define Threshold 1500 /* Greater the value, more the sensitivity */

void callback(){
  //placeholder callback function
}

void setup(){
  Serial.begin(115200);
  pinMode(7, OUTPUT);
  digitalWrite(7, HIGH); 
  delay(1000); //Take some time to open up the Serial Monitor
  
  int ESP_report =esp_sleep_pd_config(ESP_PD_DOMAIN_RTC_PERIPH, ESP_PD_OPTION_ON);
  Serial.println(ESP_report3);
  //T4 -> IO4
  touchAttachInterrupt(T4, callback, Threshold);
  int ESP_report = esp_sleep_enable_touchpad_wakeup();
  Serial.println(ESP_report);
  
  Serial.println("Going to sleep now");
  esp_deep_sleep_start();
}

void loop(){
  //This will never be reached
}
```

closed 2022-08-23 by SimGallery

Gerr63, 2022-09-24:
Hallo

Have the same problem.
When I read Sim Gallery's article, I believed that it would also solve my problem.
I tested the sketch from SimGallery but the power consumption is still 60mA in deep sleep mode.
So this is not a solution yet. #