espressif/arduino-esp32 #7151: Guru mediation error on both cores while trying to read data from pins

MetriciRO, 2022-08-18:
### Board

Olimex ESP32-EVB rev. I

### Device Description

Olimex ESP32-EVB rev. I

### Hardware Configuration

No

### Version

v2.0.3

### IDE Name

PlatformIO

### Operating System

Windows 10

### Flash frequency

40

### PSRAM enabled

no

### Upload speed

460800

### Description

I am trying to read some Wiegand card using the Bosch ARD-AYK12 - RFID Proximity Reader, and the following library: https://github.com/jvondrus/Wiegand26 ,but it doesn't work.

Do I need to use level shifters ?

### Sketch

```cpp
/*
 *  Wiegand26 - https://github.com/jvondrus/Wiegand26
 *  by Jiri Vondrus (https://github.com/jvondrus)
 *  Version 1.1.0 06-2020
 */
#include <Arduino.h>
#include <Wiegand26.h> // Wiegand RFID

// Status report defines
#define INICIALIZE 0
#define DATA_SENDED 1
#define CONNECTION 2
#define LOGIC_FAULT 3
#define RCV_TIMEOUT 4
#define BITS_FAULTS 5
#define PARITY_FRST 6
#define PARITY_SCND 7

// Inicialize Wiegand26
Wiegand26 wiegand;

#define wiegandD0 14 // Pin for Data 0
#define wiegandD1 13 // Pin for Data 1

// Function for reading Wiegand data
void wiegandPinChanged()
{
  wiegand.readData();
}

// Print received Wiegand state
void wiegandState(uint8_t state)
{
  if (bitRead(state, INICIALIZE))
  {
    Serial.println("Wiegand26 - Inicialized");
  }
  if (bitRead(state, DATA_SENDED))
  {
    Serial.println("Wiegand26 - Data sended");
  }
  if (bitRead(state, CONNECTION))
  {
    Serial.println("Wiegand26 - Disconnected");
  }
  if (bitRead(state, LOGIC_FAULT))
  {
    Serial.println("Wiegand26 - Wiring problem");
  }
  if (bitRead(state, RCV_TIMEOUT))
  {
    Serial.println("Wiegand26 - Time Out");
  }
  if (bitRead(state, BITS_FAULTS))
  {
    Serial.println("Wiegand26 - Buffer size fault");
  }
  if (bitRead(state, PARITY_FRST))
  {
    Serial.println("Wiegand26 - First parity fault");
  }
  if (bitRead(state, PARITY_SCND))
  {
    Serial.println("Wiegand26 - Second parity fault");
  }
}

// Print received Wiegand key
void wiegandKey(uint8_t value)
{
  Serial.print("Wiegand26 - Key: 0x");
  Serial.println(value, HEX);

  Serial.print("Wiegand26 - Key: 0b");
  Serial.println(value, BIN);

  Serial.print("Wiegand26 - Key: ");
  Serial.println(value, DEC);

  Serial.println();
}

// Print received Wiegand code
void wiegandCode(unsigned long value)
{
  Serial.print("Wiegand26 - Code: 0x");
  Serial.println(value, HEX);

  Serial.print("Wiegand26 - Code: 0b");
  Serial.println(value, BIN);

  Serial.print("Wiegand26 - Code: ");
  Serial.println(value, DEC);

  Serial.println();
}

// Print received Wiegand data
void wiegandData(unsigned long value)
{
  Serial.print("Wiegand26 - Data: 0x");
  Serial.println(value, HEX);

  Serial.print("Wiegand26 - Data: 0b");
  Serial.println(value, BIN);

  Serial.print("Wiegand26 - Data: ");
  Serial.println(value, DEC);

  Serial.println();
}

// Setup
void setup()
{
  // Serial line
  Serial.begin(115200, SERIAL_8N1);
  Serial.setTimeout(15);
  delay(1000);

  // Wiegand RFID
  wiegand.onKey(wiegandKey);
  wiegand.onCode(wiegandCode);
  wiegand.onData(wiegandData);
  wiegand.onState(wiegandState);
  wiegand.begin(wiegandD0, wiegandD1, true, false);
  //            Data0      Data1      false == Send state only on change
  //                                  true  == Send state on each data reading
  //                                        false == non-swap data (125kH readers)
  //                                        true  == swap data (some 13.56MHz readers)

  // Interrupt for Wiegand data pin
  attachInterrupt(digitalPinToInterrupt(wiegandD0), wiegandPinChanged, FALLING);
  attachInterrupt(digitalPinToInterrupt(wiegandD1), wiegandPinChanged, FALLING);
}

// Main loop
void loop()
{

  // Manually invoke Wiegand state
  // wiegand.readState ();

  delay(250);
}
```

### Debug Message

```plain
rst:0x1 (POWERON_RESET),boot:0x1b (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:1184
load:0x40078000,len:12784
load:0x40080400,len:3032
entry 0x400805e4
[��mum����2-hal-cpu.c:214] setCpuFrequencyMhz(): PLL: 480 / 2 = 240 Mhz, APB: 80000000 Hz
Wiegand26 - Inicialized
Wiegand26 - Data: 0xE28244
Wiegand26 - Data: 0b111000101000001001000100
Wiegand26 - Data: 14844484

Wiegand26 - Inicialized
WieGuru Meditation Error: Core  1 panic'ed (Interrupt wdt timeout on CPU1). 

Core  1 register dump:
PC      : 0x4008c8c8  PS      : 0x00060d35  A0      : 0x8008bb1a  A1      : 0x3ffbec2c
A2      : 0x3ffb9068  A3      : 0x3ffc68c8  A4      : 0x00000004  A5      : 0x00060d23  
A6      : 0x00060d23  A7      : 0x00000001  A8      : 0x3ffc68c8  A9      : 0x00000019
A10     : 0x3ffc68c8  A11     : 0x00000019  A12     : 0x3ffc380c  A13     : 0x00060d23  
A14     : 0x007beeb8  A15     : 0x003fffff  SAR     : 0x00000009  EXCCAUSE: 0x00000006
EXCVADDR: 0x00000000  LBEG    : 0x4008841d  LEND    : 0x4008842d  LCOUNT  : 0xfffffff9  
Core  1 was running in ISR context:
EPC1    : 0x400e842b  EPC2    : 0x00000000  EPC3    : 0x00000000  EPC4    : 0x00000000

Backtrace:0x4008c8c5:0x3ffbec2c |<-CORRUPTED

  #0  0x4008c8c5:0x3ffbec2c in vListInsert at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/list.c:166 (discriminator 1)

Core  0 register dump:
PC      : 0x4008ca49  PS      : 0x00060035  A0      : 0x8008b743  A1      : 0x3ffbe80c
A2      : 0x3ffbeeb8  A3      : 0xb33fffff  A4      : 0x0000abab  A5      : 0x00060023
A6      : 0x00060021  A7      : 0x0000cdcd  A8      : 0x0000abab  A9      : 0xffffffff
A10     : 0x00000000  A11     : 0x00000000  A12     : 0x3ffc28d8  A13     : 0x00000007
A14     : 0x007beeb8  A15     : 0x003fffff  SAR     : 0x0000001a  EXCCAUSE: 0x00000006  
EXCVADDR: 0x00000000  LBEG    : 0x00000000  LEND    : 0x00000000  LCOUNT  : 0x00000000

Backtrace:0x4008ca46:0x3ffbe80c |<-CORRUPTED

  #0  0x4008ca46:0x3ffbe80c in compare_and_set_native at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp_hw_support/include/soc/compare_set.h:25
      (inlined by) spinlock_acquire at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp_hw_support/include/soc/spinlock.h:103
      (inlined by) xPortEnterCriticalTimeout at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/port/xtensa/port.c:288

ELF file SHA256: 0000000000000000

Rebooting...
```
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

SuGlider, 2022-08-21:
It may be an issue related to `ditigalRead()` executed within the ISR. It needs investigation to verify it.

VojtechBartoska, 2022-09-21:
Can you please try to test this under latest version 2.0.5 @MetriciRO? There was a fix to GPIO which should have influence on this.

MetriciRO, 2022-09-26:
> Can you please try to test this under latest version 2.0.5 @MetriciRO? There was a fix to GPIO which should have influence on this.

Hello.
I found that the library https://github.com/jvondrus/Wiegand26 was causing the error. It is maintained anymore so I have since switched to another library and I don't have this error anymore.
I am sorry that I can't test this for you.

closed 2022-09-26 by MetriciRO