espressif/arduino-esp32 #7155: ESP32 Flash params set to 0x0220? 0x022f?

larryyyyy, 2022-08-19:
### Board

ESP32 Dev Module

### Device Description

just simple wemos d1 r32.

### Hardware Configuration

just simple wemos d1 r32.

### Version

v2.0.4

### IDE Name

PlatformIO

### Operating System

macOS 11.6

### Flash frequency

240Mhz

### PSRAM enabled

yes

### Upload speed

921600

### Description

I have been using esp32 to develop projects for a period of time. I am using platformio to handle the upload config stuff. Its smooth and we have developed some libs to increase our development time. However, recently I have upgraded esptool with the platformio to get more board supported, with the command `pio pkg upgrade`. After the upgrade, these are the packages i am using:

PACKAGES: 
 - framework-arduinoespressif32 @ 3.20004.220818 (2.0.4) 
 - tool-esptoolpy @ 1.30300.0 (3.3.0) 
 - tool-mkfatfs @ 2.0.1 
 - tool-mklittlefs @ 1.203.210628 (2.3) 
 - tool-mkspiffs @ 2.230.0 (2.30) 
 - toolchain-xtensa-esp32 @ 8.4.0+2021r2-patch3
 
and some dependency that I am using:

Dependency Graph
|-- ArduinoJson @ 6.19.4
|-- StreamUtils @ 1.6.3
|   |-- EEPROM @ 2.0.0
|-- arduino-timer @ 2.3.1
|-- EEPROM @ 2.0.0

I was using eeprom to save some config values like wifissid and pw. It was all good until the upgrade. After the upgrade, I found that the platformio set the Flash params address to start on 0x0220. I guess originally it was 0x022f, and I use ArduinoIDE it is the value that ArduinoIDE use. This change makes that whole data shifted 16bit and I spend a lot of time finding the problem. 

Currently, the solution I am using is that I init the eepromStream starting from 16 instead of 0. However Im worrying one day it change again and I got to spend time finding this again. 

If anyone knows why, please let me know. Is it somewhere I set the value or any other reason? Thank you so much.

### Sketch

```cpp
esp32 wemos d1
```

### Debug Message

```plain
Linking .pio/build/release/firmware.elf
Retrieving maximum program size .pio/build/release/firmware.elf
Checking size .pio/build/release/firmware.elf
Advanced Memory Usage is available via "PlatformIO Home > Project Inspect"
RAM:   [=         ]  13.5% (used 44092 bytes from 327680 bytes)
Flash: [=======   ]  66.1% (used 866141 bytes from 1310720 bytes)
Building .pio/build/release/firmware.bin
esptool.py v3.3
Creating esp32 image...
Merged 25 ELF sections
Successfully created esp32 image.
Configuring upload protocol...
AVAILABLE: cmsis-dap, esp-bridge, esp-prog, espota, esptool, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa
CURRENT: upload_protocol = esptool
Looking for upload port...
Using manually specified: /dev/tty.SLAB_USBtoUART
Uploading .pio/build/release/firmware.bin
esptool.py v3.3
Serial port /dev/tty.SLAB_USBtoUART
Connecting....
Chip is ESP32-PICO-D4 (revision 1)
Features: WiFi, BT, Dual Core, 240MHz, Embedded Flash, VRef calibration in efuse, Coding Scheme None
Crystal is 40MHz
MAC: 24:a1:60:75:59:28
Uploading stub...
Running stub...
Stub running...
Changing baud rate to 921600
Changed.
Configuring flash size...
Flash will be erased from 0x00001000 to 0x00005fff...
Flash will be erased from 0x00008000 to 0x00008fff...
Flash will be erased from 0x0000e000 to 0x0000ffff...
Flash will be erased from 0x00010000 to 0x000e4fff...
Flash params set to 0x0220
Compressed 17408 bytes to 12093...
Writing at 0x00001000... (100 %)
Wrote 17408 bytes (12093 compressed) at 0x00001000 in 0.4 seconds (effective 327.6 kbit/s)...
Hash of data verified.
Compressed 3072 bytes to 128...
Writing at 0x00008000... (100 %)
Wrote 3072 bytes (128 compressed) at 0x00008000 in 0.1 seconds (effective 378.4 kbit/s)...
Hash of data verified.
Compressed 8192 bytes to 47...
Writing at 0x0000e000... (100 %)
Wrote 8192 bytes (47 compressed) at 0x0000e000 in 0.1 seconds (effective 563.8 kbit/s)...
Hash of data verified.
Compressed 872032 bytes to 558998...
Writing at 0x00010000... (2 %)
Writing at 0x000184d9... (5 %)
Writing at 0x00024b01... (8 %)
Writing at 0x0002da0a... (11 %)
Writing at 0x0003b312... (14 %)
Writing at 0x00040a84... (17 %)
Writing at 0x00046798... (20 %)
Writing at 0x0004c088... (22 %)
Writing at 0x00051f0b... (25 %)
Writing at 0x000574e1... (28 %)
Writing at 0x0005cfbd... (31 %)
Writing at 0x00062425... (34 %)
Writing at 0x00067ecf... (37 %)
Writing at 0x0006d309... (40 %)
Writing at 0x000723f5... (42 %)
Writing at 0x00077748... (45 %)
Writing at 0x0007c9ae... (48 %)
Writing at 0x00081d1b... (51 %)
Writing at 0x00086f9f... (54 %)
Writing at 0x0008c28d... (57 %)
Writing at 0x00091fbd... (60 %)
Writing at 0x000974c3... (62 %)
Writing at 0x0009c777... (65 %)
Writing at 0x000a1b0d... (68 %)
Writing at 0x000a6ef3... (71 %)
Writing at 0x000ac5ac... (74 %)
Writing at 0x000b1e3e... (77 %)
Writing at 0x000b7c09... (80 %)
Writing at 0x000bd512... (82 %)
Writing at 0x000c38c2... (85 %)
Writing at 0x000ce039... (88 %)
Writing at 0x000d3875... (91 %)
Writing at 0x000d90ad... (94 %)
Writing at 0x000de6e8... (97 %)
Writing at 0x000e4299... (100 %)
Wrote 872032 bytes (558998 compressed) at 0x00010000 in 9.1 seconds (effective 768.2 kbit/s)...
Hash of data verified.
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

Jason2866, 2023-04-19:
It is not the address. It tells what bits are set.

closed 2023-04-19 by Jason2866

TD-er, 2023-04-19:
Indeed, these bits set the flash size, flash frequency and flash mode (DIO/QIO/etc)