espressif/arduino-esp32 #7157: EPS32 Flash size read via API is wrong when flash is in modes `qio, dio, qout`

Jason2866, 2022-08-20:
### Board

esp32 devkit 4 MB

### Device Description

Plain Devkit board, nothing connected

### Hardware Configuration

none

### Version

v2.0.4

### IDE Name

Platformio latest release

### Operating System

macOS 12.5.1

### Flash frequency

40 Mhz

### PSRAM enabled

no

### Upload speed

115200

### Description

Get flash size correctly when in flash modes `qio, qout, dio` as it does when in flash mode `dout`.
The standard Arduino function get flash size is useless in most cases, since it just reads the values which are set from the IDE and it does not detect the real flash size

### Sketch

```cpp
Sketch to test https://github.com/Jason2866/ESP32_Show_Info
Can be run with Arduino IDE too.
The test sketch shows too that flash mode is not correct when queried via the API (qio and qout never shown)
```

### Debug Message

```plain
First (correct) output in mode dout 

entry 0x400805c4
can't allocate memory with malloc

ESP32 SDK: v4.4.3
ESP32 CPU FREQ: 240 MHz
ESP32 APB FREQ: 80.0 MHz
ESP32 FLASH CHIP ID: 1458392
ESP32 FLASH REAL SIZE: 4.00 MB
ESP32 FLASH SIZE (MAGIC BYTE): 4.00 MB
ESP32 FLASH MODE: 3, 0=QIO, 1=QOUT, 2=DIO, 3=DOUT
ESP32 RAM SIZE: 366.19 KB
ESP32 FREE RAM: 342.56 KB
ESP32 MAX RAM ALLOC: 119.99 KB
ESP32 FREE PSRAM: 0.00 KB
```
Second (wrong) output in mode qio (shown wrong dio too)
```
entry 0x400805d0
can't allocate memory with malloc

ESP32 SDK: v4.4.3
ESP32 CPU FREQ: 240 MHz
ESP32 APB FREQ: 80.0 MHz
ESP32 FLASH CHIP ID: 729324
ESP32 FLASH REAL SIZE: 0.00 MB
ESP32 FLASH SIZE (MAGIC BYTE): 4.00 MB
ESP32 FLASH MODE: 2, 0=QIO, 1=QOUT, 2=DIO, 3=DOUT
ESP32 RAM SIZE: 366.19 KB
ESP32 FREE RAM: 342.56 KB
ESP32 MAX RAM ALLOC: 119.99 KB
ESP32 FREE PSRAM: 0.00 KB
```
```

### Other Steps to Reproduce

Running the example sketch

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

TD-er, 2022-08-20:
At boot you also can see the clock-div.
Would be nice if we could also get the runtime value to compute the actual flash frequency.

This function is also reading what's set during build:
```c++
uint32_t EspClass::getFlashChipSpeed(void)
{
    esp_image_header_t fhdr;
    if(flashRead(ESP_FLASH_IMAGE_BASE, (uint32_t*)&fhdr, sizeof(esp_image_header_t)) && fhdr.magic != ESP_IMAGE_HEADER_MAGIC) {
        return 0;
    }
    return magicFlashChipSpeed(fhdr.spi_speed);
}
```

Jason2866, 2022-08-21:
@Staars did find a way. Credits for this go to him!

Correct Flash Chip ID can be read in all modes with:
```
uint32_t ESP_getFlashChipId(void)
{
  uint32_t id = g_rom_flashchip.device_id;
  id = ((id & 0xff) << 16) | ((id >> 16) & 0xff) | (id & 0xff00);
  return id;
}
``` 
AND when this is correctly fetched the flash size can be done :-)

PR #7159 (2022-09-07): Get real Flash Chip Size via API call
actually only the value from the IDE which has been set is read from magic header.
Now the real size is read from the flash chip.

Credits got to @Staars

Fixes #7157

### Checklist
1. [x] Please provide specific title of the PR describing the change, including the component name (*eg. „Update of Documentation link on Readme.md“*)
2. [x] Please provide related links (*eg. Issue which will be closed by this Pull Request*)
3. [x] Please **update relevant Documentation** if applicable
4. [x] Please check [Contributing guide](https://docs.espressif.com/project [... cut]
closed 2022-09-07 by P-R-O-C-H-Y