espressif/arduino-esp32 #7160: pins_arduino.h

wmarchewka, 2022-08-21:
### Board

WT32-ETH01

### Device Description

WT32-ETH01

### Hardware Configuration

#define VSPI_MISO -1
#define VSPI_CS 12
#define VSPI_SCLK 14
#define VSPI_MOSI 15

### Version

v2.0.4

### IDE Name

Platformio

### Operating System

Mac OS 11.6

### Flash frequency

40Mhz

### PSRAM enabled

yes

### Upload speed

96100

### Description

Code worked when Platformio Development environment is set to [env:esp32dev] but when changed to [env:wt32-eth01]
 code fails with:

platformio/packages/framework-arduinoespressif32/libraries/SPI/src/SPI.cpp: In member function 'void SPIClass::begin(int8_t, int8_t, int8_t, int8_t)':
/.platformio/packages/framework-arduinoespressif32/libraries/SPI/src/SPI.cpp:64:37: error: 'SCK' was not declared in this scope
         _sck = (_spi_num == VSPI) ? SCK : 14;
                                     
It appears that the:
[espressif/arduino-esp32/blob/master/variants/wt32-eth01/pins_arduino.h](https://github.com/espressif/arduino-esp32/blob/master/variants/wt32-eth01/pins_arduino.h)

Does not contain the same elements as the other library:
static const uint8_t SS    = 5;
static const uint8_t MOSI  = 23;
static const uint8_t MISO  = 19;
static const uint8_t SCK   = 18;

### Sketch

```cpp
#include <Arduino.h>
#include <SPI.h>

#define VSPI_MISO -1
#define VSPI_CS 12
#define VSPI_SCLK 14
#define VSPI_MOSI 15

SPIClass *vspi = NULL;

void setup(){
  vspi = new SPIClass(VSPI);                              //create spi
  vspi->begin(VSPI_SCLK, VSPI_MISO, VSPI_MOSI, VSPI_CS);  //SCLK, MISO, MOSI, SS
  pinMode(VSPI_CS, OUTPUT);                               //VSPI SS
  digitalWrite(VSPI_CS, HIGH);                            //pull ss high t 
}

void loop()
{
  delay(1);
}
```

### Debug Message

```plain
platformio/packages/framework-arduinoespressif32/libraries/SPI/src/SPI.cpp: In member function 'void SPIClass::begin(int8_t, int8_t, int8_t, int8_t)':
/.platformio/packages/framework-arduinoespressif32/libraries/SPI/src/SPI.cpp:64:37: error: 'SCK' was not declared in this scope
         _sck = (_spi_num == VSPI) ? SCK : 14;
```

### Other Steps to Reproduce

[https://community.platformio.org/t/wt32-eth01-spi-issue/28132/2](url)

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

SuGlider, 2022-08-21:
@P-R-O-C-H-Y  - I think  that this  is a case simillar to #6737

P-R-O-C-H-Y, 2022-08-29:
@wmarchewka The author of the board doesn't declare the default SPI pins. Adding the same pins as default ones for ESP32 to the pins_arduino.h can conflict with Ethernet pins.

PR #7161 (2022-09-14): Add default SPI pin definitions if not defined in pins_arduino.h
## Description of Change
Added to some boards missing default pin declaration for SPI in `pins_arduino.h` + some I2C. Fixing SPI error in compilation.

Boards:
metro_esp-32
wt32-eth01

## Tests scenarios
Variant [wt32-eth01](https://github.com/espressif/arduino-esp32/tree/master/variants/wt32-eth01)

## Related links
Closes #7160
closed 2022-09-14 by me-no-dev