espressif/arduino-esp32 #7162: Exception when two attachInterruptArg called, one after another

vovagorodok, 2022-08-22:
### Board

ESP32 Dev

### Device Description

DevKitC

### Hardware Configuration

N/A

### Version

v2.0.3

### IDE Name

PlatformIO

### Operating System

Ubuntu

### Flash frequency

default

### PSRAM enabled

no

### Upload speed

default

### Description

For:
```
attachInterruptArg(digitalPinToInterrupt(33), methodPtr1, objectPtr1, FALLING);
attachInterruptArg(digitalPinToInterrupt(32), methodPtr2, objectPtr2, FALLING);
```
I have following exception:
```
...
E (252) gpio: esp_ipc_call_blocking failed (0x103)
[     3][E][esp32-hal-gpio.c:156] __attachInterruptFunctionalArg(): GPIO ISR Service Failed To Start
E (255) gpio: gpio_install_isr_service(449): GPIO isr service already installed
Guru Meditation Error: Core  0 panic'ed (LoadProhibited). Exception was unhandled.

Core  0 register dump:
...
```
When I'm trying to comment one line, then it works correctly. Any reason why I can't register interrupt method for each pin?

### Sketch

```cpp
N/A
```

### Debug Message

```plain
N/A
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

P-R-O-C-H-Y, 2022-08-31:
Hi @vovagorodok Can you retest please with 2.0.4 version? No error on my side.

vovagorodok, 2022-09-02:
Hi @P-R-O-C-H-Y ,
Thanks! Currently I'm on (2-3 weeks) vacations. I'll check it when when I come back

jwillhoeft, 2022-10-12:
I am getting the same error and stack trace when I am combining some larger libraries in my project (WifiManager, SPIFFS, Encoder, ...)
It also happens when I tried to compile an older project (using NeoPixel, Encoder, Wifi, ..) in ESP 2.0.4 and 2.0.5. This older project still compiles and runs in ESP32 1.0.6.

I tried to reproduce it with the example code "GPIOInterrupt.ino" (which should be similar to the code snippet above from @vovagorodok), but that worked as expected.

My hardware is a NodeMCU ESP32 Dev Kit / ESP-WROOM-32. I am using the Arduino IDE 1.8.9.

Stack trace:
`
15:25:58.494 -> ELF file SHA256: 0000000000000000
15:25:58.527 -> 
15:25:58.527 -> Rebooting...
15:25:58.527 -> ets Jun  8 2016 00:22:57
15:25:58.527 -> 
15:25:58.527 -> rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
15:25:58.527 -> configsip: 0, SPIWP:0xee
15:25:58.527 -> clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
15:25:58.527 -> mode:DIO, clock div:1
15:25:58.527 -> load:0x3fff0030,len:1344
15:25:58.527 -> load:0x40078000,len:13864
15:25:58.527 -> load:0x40080400,len:3608
15:25:58.527 -> entry 0x400805f0
15:25:58.659 -> E (149) gpio: esp_ipc_call_blocking failed (0x103)
15:25:58.659 -> E (149) gpio: gpio_install_isr_service(449): GPIO isr service already installed
15:25:58.659 -> Guru Meditation Error: Core  0 panic'ed (LoadProhibited). Exception was unhandled.
15:25:58.693 -> 
15:25:58.693 -> Core  0 register dump:
15:25:58.693 -> PC      : 0x400f23f7  PS      : 0x00060533  A0      : 0x800d5f1f  A1      : 0x3ffe3b30  
15:25:58.693 -> A2      : 0x00000000  A3      : 0x3ffc1f78  A4      : 0x3ffc1f78  A5      : 0x00060523  
15:25:58.693 -> A6      : 0x00060520  A7      : 0x00000001  A8      : 0x800d5adc  A9      : 0x3ffe3af0  
15:25:58.693 -> A10     : 0x3ffbdc04  A11     : 0x00800000  A12     : 0x00000017  A13     : 0x3ffe3b30  
15:25:58.726 -> A14     : 0x007bdbf0  A15     : 0x003fffff  SAR     : 0x00000009  EXCCAUSE: 0x0000001c  
15:25:58.726 -> EXCVADDR: 0x00000000  LBEG    : 0x40086315  LEND    : 0x40086325  LCOUNT  : 0xfffffff7  
15:25:58.726 -> 
15:25:58.726 -> 
15:25:58.726 -> Backtrace:0x400f23f4:0x3ffe3b300x400d5f1c:0x3ffe3b50 0x400d42b8:0x3ffe3b80 0x400d4339:0x3ffe3bb0 0x400d1641:0x3ffe3bd0 0x400d1ab8:0x3ffe3bf0 0x400dc5c7:0x3ffe3c10 0x40082bdd:0x3ffe3c40 0x400792ca:0x3ffe3c90  |<-CORRUPTED
15:25:58.759 ->
`

mrengineer7777, 2022-10-12:
Is it possible one of your libraries is targeting the older core?  Simplify your project, then add libraries until it fails.

vovagorodok, 2022-10-12:
Retested. Works fine with new version

P-R-O-C-H-Y, 2022-10-13:
Thanks for info :) Closing as there is no issue with newer versions.

closed 2022-10-13 by P-R-O-C-H-Y