espressif/arduino-esp32 #7165: ESP32-S3 Powercycling right after uploading a sketch using Arduino IDE

SuGlider, 2022-08-22:
### Board

ESP32-S3

### Device Description

ESP32-S3-DevkitC-1 1.0 

### Hardware Configuration

None

### Version

v2.0.4

### IDE Name

Arduino IDE 1.8.15

### Operating System

Windows 11

### Flash frequency

QIO 80MHz

### PSRAM enabled

no

### Upload speed

Tried 115200 and 921600

### Description

I have two, apparently equal, ESP32-S3 boards.
For one of them I can upload any sketch and it works fine right after reset.
The other one keeps reseting after uploading and I can't make it work.

BUT.... when I use IDF examples (HelloWolds and Blink RGB LED) both works fine!
So it seems that something may be wrong with Arduino second-stage bootloader or some boot configuration.

No error message even with DEBUG turned on.

### Sketch

```cpp
// 01.Basics --> BareMinimum Arduino Example

void setup() {
  // put your setup code here, to run once:

}

void loop() {
  // put your main code here, to run repeatedly:

}
```

### Debug Message

```plain
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x1 (POWERON),boot:0x8 (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fce3808,len:0x43c
load:0x403c9700,len:0xbec
load:0x403cc700,len:0x2a3c
SHA-256 comparison failed:
Calculated: 3611f330726ad4edf64f62982b0568332a5800bc59425cab90be5379db92c108
Expected: ffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffff
Attempting to boot anyway...
entry 0x403c98d8
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x7 (TG0WDT_SYS_RST),boot:0x8 (SPI_FAST_FLASH_BOOT)
Saved PC:0x403cebce
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fce3808,len:0x43c
load:0x403c9700,len:0xbec
load:0x403cc700,len:0x2a3c
SHA-256 comparison failed:
Calculated: 3611f330726ad4edf64f62982b0568332a5800bc59425cab90be5379db92c108
Expected: ffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffff
Attempting to boot anyway...
entry 0x403c98d8
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x7 (TG0WDT_SYS_RST),boot:0x8 (SPI_FAST_FLASH_BOOT)
Saved PC:0x403cebce
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fce3808,len:0x43c
load:0x403c9700,len:0xbec
load:0x403cc700,len:0x2a3c
SHA-256 comparison failed:
Calculated: 3611f330726ad4edf64f62982b0568332a5800bc59425cab90be5379db92c108
Expected: ffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffff
Attempting to boot anyway...
entry 0x403c98d8
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x10 (RTCWDT_RTC_RST),boot:0x8 (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fce3808,len:0x43c
load:0x403c9700,len:0xbec
load:0x403cc700,len:0x2a3c
SHA-256 comparison failed:
Calculated: 3611f330726ad4edf64f62982b0568332a5800bc59425cab90be5379db92c108
Expected: ffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffff
Attempting to boot anyway...
entry 0x403c98d8
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x7 (TG0WDT_SYS_RST),boot:0x8 (SPI_FAST_FLASH_BOOT)
Saved PC:0x403cebce
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fce3808,len:0x43c
load:0x403c9700,len:0xbec
load:0x403cc700,len:0x2a3c
SHA-256 comparison failed:
Calculated: 3611f330726ad4edf64f62982b0568332a5800bc59425cab90be5379db92c108
Expected: ffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffff
Attempting to boot anyway...
entry 0x403c98d8
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x7 (TG0WDT_SYS_RST),boot:0x8 (SPI_FAST_FLASH_BOOT)
Saved PC:0x403cebce
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fce3808,len:0x43c
load:0x403c9700,len:0xbec
load:0x403cc700,len:0x2a3c
SHA-256 comparison failed:
Calculated: 3611f330726ad4edf64f62982b0568332a5800bc59425cab90be5379db92c108
Expected: ffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffff
Attempting to boot anyway...
entry 0x403c98d8
```

### Other Steps to Reproduce

As simple as that. I get TG0WDT_SYS_RST or RTCWDT_RTC_RST in the second-stage bootloader execution.

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

SuGlider, 2022-08-22:
Can it be related to #7125 ?

SuGlider, 2022-08-22:
The S3 that works fine has this output:

```
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x1 (POWERON),boot:0x8 (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fce3808,len:0x43c
load:0x403c9700,len:0xbec
load:0x403cc700,len:0x2a3c
SHA-256 comparison failed:
Calculated: 3611f330726ad4edf64f62982b0568332a5800bc59425cab90be5379db92c108
Expected: ffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffff
Attempting to boot anyway...
entry 0x403c98d8

```

SuGlider, 2022-08-22:
This is the picture of the S3 that works fine:

[image]

SuGlider, 2022-08-22:
This is the picture of the S3 that fails:

[image]

SuGlider, 2022-08-22:
Both seem to be exactly the same CHIP!

But, when I run IDF HelloWorld example.... it works perfectly well in both boards.

SuGlider, 2022-08-22:
Using IDF 4.4.1, I can run it on Both bards.

#### Log Output for the "Good board" is 
```
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x1 (POWERON),boot:0x8 (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fcd0108,len:0x20cc
load:0x403b6000,len:0xd1c
load:0x403ba000,len:0x3454
entry 0x403b6290
I (26) boot: ESP-IDF v4.4.1 2nd stage bootloader
I (26) boot: compile time 10:37:47
D (26) bootloader_flash: non-XMC chip detected by SFDP Read (C8), skip.
D (31) bootloader_flash: mmu set block paddr=0x00000000 (was 0xffffffff)
I (38) boot: chip revision: 0
D (41) boot.esp32s3: magic e9
D (44) boot.esp32s3: segments 03
D (48) boot.esp32s3: spi_mode 02
D (51) boot.esp32s3: spi_speed 0f
D (54) boot.esp32s3: spi_size 01
I (58) boot.esp32s3: Boot SPI Speed : 80MHz
I (63) boot.esp32s3: SPI Mode       : DIO
I (67) boot.esp32s3: SPI Flash Size : 2MB
D (72) boot: Enabling RTCWDT(9000 ms)
I (76) boot: Enabling RNG early entropy source...
D (81) bootloader_flash: mmu set paddr=00000000 count=1 size=c00 src_addr=8000 src_addr_aligned=0
D (90) boot: mapped partition table 0x8000 at 0x3c008000
D (95) flash_parts: partition table verified, 4 entries
I (101) boot: Partition Table:
I (104) boot: ## Label            Usage          Type ST Offset   Length
D (112) boot: load partition table entry 0x3c008000
D (117) boot: type=1 subtype=2
I (120) boot:  0 nvs              WiFi data        01 02 00009000 00006000
D (127) boot: load partition table entry 0x3c008020
D (132) boot: type=1 subtype=1
I (135) boot:  1 phy_init         RF data          01 01 0000f000 00001000
D (143) boot: load partition table entry 0x3c008040
D (148) boot: type=0 subtype=0
I (151) boot:  2 factory          factory app      00 00 00010000 00100000
I (159) boot: End of partition table
D (163) boot: Trying partition index -1 offs 0x10000 size 0x100000
D (169) esp_image: reading image header @ 0x10000
D (174) bootloader_flash: mmu set block paddr=0x00010000 (was 0xffffffff)
D (181) esp_image: image header: 0xe9 0x06 0x02 0x01 40375164
V (187) esp_image: loading segment header 0 at offset 0x10018
V (192) esp_image: segment data length 0x7ae0 data starts 0x10020
V (199) esp_image: segment 0 map_segment 1 segment_data_offs 0x10020 load_addr 0x3c020020
I (207) esp_image: segment 0: paddr=00010020 vaddr=3c020020 size=07ae0h ( 31456) map
D (215) esp_image: free data page_count 0x0000003f
D (220) bootloader_flash: mmu set paddr=00010000 count=1 size=7ae0 src_addr=10020 src_addr_aligned=10000
V (235) esp_image: loading segment header 1 at offset 0x17b00
D (236) bootloader_flash: mmu set block paddr=0x00010000 (was 0xffffffff)
V (242) esp_image: segment data length 0x26f8 data starts 0x17b08
V (248) esp_image: segment 1 map_segment 0 segment_data_offs 0x17b08 load_addr 0x3fc90630
I (257) esp_image: segment 1: paddr=00017b08 vaddr=3fc90630 size=026f8h (  9976) load
D (265) esp_image: free data page_count 0x0000003f
D (270) bootloader_flash: mmu set paddr=00010000 count=1 size=26f8 src_addr=17b08 src_addr_aligned=10000
V (282) esp_image: loading segment header 2 at offset 0x1a200
D (285) bootloader_flash: mmu set block paddr=0x00010000 (was 0xffffffff)
V (292) esp_image: segment data length 0x5e10 data starts 0x1a208
V (298) esp_image: segment 2 map_segment 0 segment_data_offs 0x1a208 load_addr 0x40374000
I (307) esp_image: segment 2: paddr=0001a208 vaddr=40374000 size=05e10h ( 24080) load
D (315) esp_image: free data page_count 0x0000003f
D (320) bootloader_flash: mmu set paddr=00010000 count=2 size=5e10 src_addr=1a208 src_addr_aligned=10000
V (335) esp_image: loading segment header 3 at offset 0x20018
D (335) bootloader_flash: mmu set block paddr=0x00020000 (was 0xffffffff)
V (342) esp_image: segment data length 0x173a0 data starts 0x20020
[... 29 lines cut ...]
V (545) boot: i mmu set paddr=00020000 vaddr=42000000 size=95136 n=2
V (551) boot: rc=0
D (553) boot: start: 0x40375164
I (567) cpu_start: Pro cpu up.
I (568) cpu_start: Starting app cpu, entry point is 0x403750bc
I (0) cpu_start: App cpu up.
I (581) cpu_start: Pro cpu start user code
I (581) cpu_start: cpu freq: 160000000
I (581) cpu_start: Application information:
I (584) cpu_start: Project name:     hello_world
I (589) cpu_start: App version:      1
I (593) cpu_start: Compile time:     Aug 22 2022 10:37:31
I (600) cpu_start: ELF file SHA256:  71c5a3bff93e6ff3...
I (606) cpu_start: ESP-IDF:          v4.4.1
I (610) heap_init: Initializing. RAM available for dynamic allocation:
I (618) heap_init: At 3FC93718 len 0004C8E8 (306 KiB): D/IRAM
I (624) heap_init: At 3FCE0000 len 0000EE34 (59 KiB): STACK/DRAM
I (631) heap_init: At 3FCF0000 len 00008000 (32 KiB): DRAM
I (637) heap_init: At 600FE000 len 00002000 (8 KiB): RTCRAM
I (644) spi_flash: detected chip: gd
I (648) spi_flash: flash io: dio
W (651) spi_flash: Detected size(8192k) larger than the size in the binary image header(2048k). Using the size in the binary image header.
I (665) sleep: Configure to isolate all GPIO pins in sleep state
I (671) sleep: Enable automatic switching of GPIO sleep configuration[â¸®
```

SuGlider, 2022-08-22:
#### IDF 4.4.1 Log Output for the "BAD board" is

```
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x1 (POWERON),boot:0x8 (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fcd0108,len:0x20cc
load:0x403b6000,len:0xd1c
load:0x403ba000,len:0x3454
entry 0x403b6290
I (26) boot: ESP-IDF v4.4.1 2nd stage bootloader
I (26) boot: compile time 10:37:47
D (26) bootloader_flash: non-XMC chip detected by SFDP Read (C8), skip.
D (31) bootloader_flash: mmu set block paddr=0x00000000 (was 0xffffffff)
I (38) boot: chip revision: 0
D (43) boot.esp32s3: magic e9
D (44) boot.esp32s3: segments 03
D (48) boot.esp32s3: spi_mode 02
D (51) boot.esp32s3: spi_speed 0f
D (54) boot.esp32s3: spi_size 01
I (58) boot.esp32s3: Boot SPI Speed : 80MHz
I (63) boot.esp32s3: SPI Mode       : DIO
I (67) boot.esp32s3: SPI Flash Size : 2MB
D (72) boot: Enabling RTCWDT(9000 ms)
I (76) boot: Enabling RNG early entropy source...
D (81) bootloader_flash: mmu set paddr=00000000 count=1 size=c00 src_addr=8000 src_addr_aligned=0
D (90) boot: mapped partition table 0x8000 at 0x3c008000
D (95) flash_parts: partition table verified, 4 entries
I (101) boot: Partition Table:
I (104) boot: ## Label            Usage          Type ST Offset   Length
D (112) boot: load partition table entry 0x3c008000
D (117) boot: type=1 subtype=2
I (120) boot:  0 nvs              WiFi data        01 02 00009000 00006000
D (127) boot: load partition table entry 0x3c008020
D (132) boot: type=1 subtype=1
I (135) boot:  1 phy_init         RF data          01 01 0000f000 00001000
D (143) boot: load partition table entry 0x3c008040
D (148) boot: type=0 subtype=0
I (151) boot:  2 factory          factory app      00 00 00010000 00100000
I (159) boot: End of partition table
D (163) boot: Trying partition index -1 offs 0x10000 size 0x100000
D (169) esp_image: reading image header @ 0x10000
D (174) bootloader_flash: mmu set block paddr=0x00010000 (was 0xffffffff)
D (181) esp_image: image header: 0xe9 0x06 0x02 0x01 40375164
V (187) esp_image: loading segment header 0 at offset 0x10018
V (192) esp_image: segment data length 0x7ae0 data starts 0x10020
V (199) esp_image: segment 0 map_segment 1 segment_data_offs 0x10020 load_addr 0x3c020020
I (207) esp_image: segment 0: paddr=00010020 vaddr=3c020020 size=07ae0h ( 31456) map
D (215) esp_image: free data page_count 0x0000003f
D (220) bootloader_flash: mmu set paddr=00010000 count=1 size=7ae0 src_addr=10020 src_addr_aligned=10000
V (235) esp_image: loading segment header 1 at offset 0x17b00
D (236) bootloader_flash: mmu set block paddr=0x00010000 (was 0xffffffff)
V (242) esp_image: segment data length 0x26f8 data starts 0x17b08
V (248) esp_image: segment 1 map_segment 0 segment_data_offs 0x17b08 load_addr 0x3fc90630
I (257) esp_image: segment 1: paddr=00017b08 vaddr=3fc90630 size=026f8h (  9976) load
D (265) esp_image: free data page_count 0x0000003f
D (270) bootloader_flash: mmu set paddr=00010000 count=1 size=26f8 src_addr=17b08 src_addr_aligned=10000
V (282) esp_image: loading segment header 2 at offset 0x1a200
D (285) bootloader_flash: mmu set block paddr=0x00010000 (was 0xffffffff)
V (292) esp_image: segment data length 0x5e10 data starts 0x1a208
V (298) esp_image: segment 2 map_segment 0 segment_data_offs 0x1a208 load_addr 0x40374000
I (307) esp_image: segment 2: paddr=0001a208 vaddr=40374000 size=05e10h ( 24080) load
D (315) esp_image: free data page_count 0x0000003f
D (320) bootloader_flash: mmu set paddr=00010000 count=2 size=5e10 src_addr=1a208 src_addr_aligned=10000
V (335) esp_image: loading segment header 3 at offset 0x20018
D (335) bootloader_flash: mmu set block paddr=0x00020000 (was 0xffffffff)
V (342) esp_image: segment data length 0x173a0 data starts 0x20020
V (349) esp_image: segment 3 map_segment 1 segment_data_offs 0x20020 load_addr 0x42000020
[... 28 lines cut ...]
V (545) boot: i mmu set paddr=00020000 vaddr=42000000 size=95136 n=2
V (551) boot: rc=0
D (553) boot: start: 0x40375164
I (567) cpu_start: Pro cpu up.
I (568) cpu_start: Starting app cpu, entry point is 0x403750bc
I (0) cpu_start: App cpu up.
I (581) cpu_start: Pro cpu start user code
I (581) cpu_start: cpu freq: 160000000
I (581) cpu_start: Application information:
I (584) cpu_start: Project name:     hello_world
I (589) cpu_start: App version:      1
I (593) cpu_start: Compile time:     Aug 22 2022 10:37:31
I (600) cpu_start: ELF file SHA256:  71c5a3bff93e6ff3...
I (605) cpu_start: ESP-IDF:          v4.4.1
I (610) heap_init: Initializing. RAM available for dynamic allocation:
I (618) heap_init: At 3FC93718 len 0004C8E8 (306 KiB): D/IRAM
I (624) heap_init: At 3FCE0000 len 0000EE34 (59 KiB): STACK/DRAM
I (631) heap_init: At 3FCF0000 len 00008000 (32 KiB): DRAM
I (637) heap_init: At 600FE000 len 00002000 (8 KiB): RTCRAM
I (644) spi_flash: detected chip: gd
I (648) spi_flash: flash io: dio
W (651) spi_flash: Detected size(8192k) larger than the size in the binary image header(2048k). Using the size in the binary image header.
I (665) sleep: Configure to isolate all GPIO pins in sleep state
I (671) sleep: Enable automatic switching of GPIO sleep configuration[â¸®
```

SuGlider, 2022-08-22:
Testing it with all possible ESP32-S3 Arduino Cores (upload speed  minimum at 115200):

- Core 2.0.3: Works FINE!!!!
- Core 2.0.4: Fails as described in the issue  

@me-no-dev --- This seems to be a REGRESSION.

Jason2866, 2022-08-22:
@SuGlider Have you tried PlatformIO? Install is done in a few minutes if you have not used before. If both devices runs with Platformio and core 2.0.4. it is probably Arduino IDE.

SuGlider, 2022-08-22:
Hi @Jason2866 - Thanks for the suggestion. 
I just created an empty project in PIO using version 5.1.0
I set it to use the default esp32-s3-devkitc-1 board and then just build/upload/monitor the project.

Same issue. Power Cycle - reseting.

#### platformio.ini 
```
[env:esp32-s3-devkitc-1]
platform = espressif32
board = esp32-s3-devkitc-1
framework = arduino
monitor_speed = 115200
```

SuGlider, 2022-08-22:
From the Platform.txt in the PIO environment I can tell that it is using Core 2.0.4 to build it.

Jason2866, 2022-08-22:
Can you please try:
```
[env:esp32-s3-devkitc-1]
platform  = https://github.com/Jason2866/platform-espressif32.git
board = esp32-s3-devkitc-1
framework = arduino
monitor_speed = 115200
```
i did some changes in Platformio to use latest esptool v4.2.1 and some small fixes which are not in the official Platformio espressif32 framework.
The changes can affect the S3

SuGlider, 2022-08-22:
@Jason2866 - Same problem... power cycle.

### PIO Terminal Ouput:
```
Processing esp32-s3-devkitc-1 (platform: https://github.com/Jason2866/platform-espressif32.git; board: esp32-s3-devkitc-1; framework: arduino)
---------------------------------------------------------------------------------------------------------------------------------------------------------------------Platform Manager: Installing git+https://github.com/Jason2866/platform-espressif32.git
git version 2.32.0.windows.2
Cloning into 'C:\Users\esp-user\.platformio\.cache\tmp\pkg-installing-7ifqokh2'...
remote: Enumerating objects: 572, done.
remote: Counting objects: 100% (572/572), done.
remote: Compressing objects: 100% (431/431), done.
remote: Total 572 (delta 253), reused 340 (delta 119), pack-reused 0
Receiving objects: 100% (572/572), 448.82 KiB | 764.00 KiB/s, done.
Resolving deltas: 100% (253/253), done.
Platform Manager: espressif32@2.0.4+1.sha.28cf70e has been installed!
Tool Manager: Installing https://github.com/tasmota/arduino-esp32/releases/download/2.0.4.1/framework-arduinoespressif32.tar.gz
Unpacking  [####################################]  100%
Tool Manager: framework-arduinoespressif32@2.0.4 has been installed!
Tool Manager: Installing espressif/toolchain-xtensa-esp32s3 @ 8.4.0+2021r2-patch3
Unpacking  [####################################]  100%
Tool Manager: toolchain-xtensa-esp32s3@8.4.0+2021r2-patch3 has been installed!
Tool Manager: Installing espressif/toolchain-riscv32-esp @ 8.4.0+2021r2-patch3
Unpacking  [####################################]  100%
Tool Manager: toolchain-riscv32-esp@8.4.0+2021r2-patch3 has been installed!
Tool Manager: Installing https://github.com/tasmota/esptool/releases/download/v4.2.1/esptool-4.2.1.zip
Unpacking  [####################################]  100%
Tool Manager: tool-esptoolpy@1.40201.220809 has been installed!
Tool Manager: Installing platformio/tool-mkspiffs @ ~2.230.0
Unpacking  [####################################]  100%
Tool Manager: tool-mkspiffs@2.230.0 has been installed!
Tool Manager: Installing platformio/tool-mklittlefs @ ~1.203.0
Unpacking  [####################################]  100%
Tool Manager: tool-mklittlefs@1.203.210628 has been installed!
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/espressif32/esp32-s3-devkitc-1.html
PLATFORM: Espressif 32 (2.0.4+1.sha.28cf70e) > Espressif ESP32-S3-DevKitC-1-N8 (8 MB QD, No PSRAM)
HARDWARE: ESP32S3 240MHz, 320KB RAM, 8MB Flash
DEBUG: Current (esp-builtin) On-board (esp-builtin) External (cmsis-dap, esp-bridge, esp-prog, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa)
PACKAGES:
 - framework-arduinoespressif32 @ 2.0.4
 - tool-esptoolpy @ 1.40201.220809 (4.2.1)
 - tool-mklittlefs @ 1.203.210628 (2.3)
 - tool-mkspiffs @ 2.230.0 (2.30)
 - toolchain-riscv32-esp @ 8.4.0+2021r2-patch3 
 - toolchain-xtensa-esp32s3 @ 8.4.0+2021r2-patch3
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 32 compatible libraries
Scanning dependencies...
No dependencies
Building in release mode
Compiling .pio\build\esp32-s3-devkitc-1\src\main.cpp.o
Compiling .pio\build\esp32-s3-devkitc-1\FrameworkArduino\Esp.cpp.o
Compiling .pio\build\esp32-s3-devkitc-1\FrameworkArduino\FirmwareMSC.cpp.o
Compiling .pio\build\esp32-s3-devkitc-1\FrameworkArduino\FunctionalInterrupt.cpp.o
Compiling .pio\build\esp32-s3-devkitc-1\FrameworkArduino\HWCDC.cpp.o
Compiling .pio\build\esp32-s3-devkitc-1\FrameworkArduino\HardwareSerial.cpp.o
Compiling .pio\build\esp32-s3-devkitc-1\FrameworkArduino\IPAddress.cpp.o
Compiling .pio\build\esp32-s3-devkitc-1\FrameworkArduino\IPv6Address.cpp.o
Compiling .pio\build\esp32-s3-devkitc-1\FrameworkArduino\MD5Builder.cpp.o
Compiling .pio\build\esp32-s3-devkitc-1\FrameworkArduino\Print.cpp.o
Compiling .pio\build\esp32-s3-devkitc-1\FrameworkArduino\Stream.cpp.o
Compiling .pio\build\esp32-s3-devkitc-1\FrameworkArduino\StreamString.cpp.o
Compiling .pio\build\esp32-s3-devkitc-1\FrameworkArduino\Tone.cpp.o
Compiling .pio\build\esp32-s3-devkitc-1\FrameworkArduino\USB.cpp.o
Compiling .pio\build\esp32-s3-devkitc-1\FrameworkArduino\USBCDC.cpp.o
Compiling .pio\build\esp32-s3-devkitc-1\FrameworkArduino\USBMSC.cpp.o
Compiling .pio\build\esp32-s3-devkitc-1\FrameworkArduino\WMath.cpp.o
Compiling .pio\build\esp32-s3-devkitc-1\FrameworkArduino\WString.cpp.o
Compiling .pio\build\esp32-s3-devkitc-1\FrameworkArduino\base64.cpp.o
[... 207 lines cut ...]
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fce3808,len:0x118
load:0x403c9700,len:0xba4
load:0x403cc700,len:0x287c
SHA-256 comparison failed:
Calculated: 5915a3eb601e6348eb01a09185a40349448a38ad4ee23325591d0f91b300e34c
Expected: ffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffff
Attempting to boot anyway...
entry 0x403c98c0
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x7 (TG0WDT_SYS_RST),boot:0x8 (SPI_FAST_FLASH_BOOT)
Saved PC:0x403cea0e
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fce3808,len:0x118
load:0x403c9700,len:0xba4
load:0x403cc700,len:0x287c
SHA-256 comparison failed:
Calculated: 5915a3eb601e6348eb01a09185a40349448a38ad4ee23325591d0f91b300e34c
Expected: ffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffff
Attempting to boot anyway...
entry 0x403c98c0
```

Jason2866, 2022-08-22:
Weird, and the other device works here too?

SuGlider, 2022-08-22:
Yes.... That's really weird...
Even Worst: If I use IDF to build and upload a "HelloWorld", it works on both boards.

I think that the issue lies on something related to the second-stage bootloader, because it fails with WDT.

SuGlider, 2022-08-22:
Another important information is that both boards work with Arduino Creo 2.0.3... which has the "wrong" S3 bootloader.

Jason2866, 2022-08-22:
Mhh, another strange thing. I builded a framework where the IDF bootlog is enabled. This does work as it should for the ESP32, ESP32-S2 and the ESP32-C3 and i see messages from the 2nd stage bootlloader.
BUT for the S3 i get NO Bootlogs, nothing!

Jason2866, 2022-08-22:
And why is it Flash Mode `DIO`? The board has `QIO` Flash and you have for sure set QIO.

SuGlider, 2022-08-22:
> And why is it Flash Mode `DIO`? The board has `QIO` Flash and you have for sure set QIO.

Already tried it. No change.

Jason2866, 2022-08-22:
@me-no-dev  I have this error log message during a framework build run with the arduino lib builder
```
[1611/1611] Generating idf_libs
IDF_TARGET: esp32s3, MEMCONF: qio_qspi, PWD: /Users/mac/Git/esp32-arduino-lib-builder, OUT: /Users/mac/Git/esp32-arduino-lib-builder/out/tools/sdk/esp32s3
Renaming 'mbedtls' to 'mbedtls_2': /Users/mac/Git/esp32-arduino-lib-builder/build/esp-idf/mbedtls/mbedtls/library/libmbedtls.a
awk: syntax error at source line 1
 context is
	 >>> /ASFLAGS=\[/{n++}{print>n"pio_start.txt" <<< 
awk: illegal statement at source line 1
awk: syntax error at source line 1
 context is
	 >>> /"ARDUINO_ARCH_ESP32"/{n++}{print>n"pio_end.txt" <<< 
awk: illegal statement at source line 1
cat: pio_start.txt: No such file or directory
rm: pio_end.txt: No such file or directory
rm: 1pio_start.txt: No such file or directory
rm: pio_start.txt: No such file or directory
cat: 1pio_end.txt: No such file or directory
rm: 1pio_end.txt: No such file or directory
awk: syntax error at source line 1
 context is
	 >>> /compiler.cpreprocessor.flags.esp32s3=/{n++}{print>n"platform_start.txt" <<< 
awk: illegal statement at source line 1
gsed: can't read 1platform_start.txt: No such file or directory
awk: syntax error at source line 1
 context is
	 >>> /compiler.ar.flags.esp32s3=/{n++}{print>n"platform_mid.txt" <<< 
awk: illegal statement at source line 1
cat: platform_start.txt: No such file or directory
cat: 1platform_mid.txt: No such file or directory
* Build BootLoader: configs/defconfig.common;configs/defconfig.esp32s3;configs/defconfig.qio;configs/defconfig.120m;configs/defconfig.qio_ram
```
Idk if this is related, anyhow it is not nice to see

me-no-dev, 2022-08-23:
@Jason2866 this is unrelated and probably caused by awk version or file encoding/line endings

VojtechBartoska, 2022-08-24:
This issue will be investigated at the end of September, it's postponed due to HW location.

Jason2866, 2022-08-24:
Any idea how to get logs from the 2nd stage bootloader for the S3 ?
Tried a simple Blink sketch in IDF. For every MCU except the S3 i get log outputs from the 2nd stage bootloader. Is there a option which is not listed in sdkconfig for the S3 ?

Jason2866, 2022-08-25:
I come to the comclusion that OPI flash mode is not working whether in IDF nor in Arduino. Probably OPI device crashes since the init procedure is missing OPI support.
IDF is doing probably a bit less wrong, so the device starts.
My conclusion is based on the starting message which indicates `dio and 2MB flash size` and i missed or do not understand how esptool.py is activating the `opi` mode.
Another indicator that something is just totally off is the fact that it is impossible to active log messages of the 2nd stage bootloader for the S3.

me-no-dev, 2022-08-25:
@Jason2866 no issue with OPI here

Jason2866, 2022-08-25:
> @Jason2866 no issue with OPI here

I do you know it is really! running in mode OPI?

me-no-dev, 2022-08-25:
@Jason2866 OPI can run only in OPI mode. Performance improvement is also visible. What kind of sign do you need? Try to boot OPI chip in another mode and you will see :)

There are no more inconsistencies in flash modes in Arduino. That has all been fixed (in bootloaders and linked libs).

How are you trying to enable boot logs?

Jason2866, 2022-08-25:
In IDF sdkconfig. Used a simple Blink sketch. Got bootlogs for ESP32, ESP32-C3 and ESP32-S2. For ESP32-S3 no way to get something.

Jason2866, 2022-08-25:
@me-no-dev In IDF sdkconfig. Used a simple Blink sketch. Got bootlogs for ESP32, ESP32-C3 and ESP32-S2. For ESP32-S3 no way to get something.

me-no-dev, 2022-08-25:
@Jason2866 you can poke IDF team about that, though I see no reason... you are using the UART port right? Not the USB one?

Jason2866, 2022-08-25:
Yes, Uart and i have connected for tests the USB port too to a terminal.

>There are no more inconsistencies in flash modes in Arduino. That has all been fixed (in bootloaders and linked libs).

I am sarcastic, this issue is not real?

Jason2866, 2022-08-25:
Sorry for double postings. Internet issues

Jason2866, 2022-08-25:
> @Jason2866 issue here is something totally different. We do not know yet what is causing it. It happens so far on only one board that @SuGlider has.

No not a single issue, a team member from us (Tasmota @Staars) has the same issue. His board a S3 maple eye doing the same bootloop.

TD-er, 2022-08-25:
> OPI can run only in OPI mode

That's quite important info.
Can S3 boards also be made using different flash?
If so, then it would be nice to have some kind of autodetect flag in the build env or else we end up doubling the amount of bin types for S3 boards (well, whenever I start supporting these, my own ESP32-S3-BOX is still sealed)

Jason2866, 2022-08-25:
@TD-er  See https://github.com/espressif/arduino-esp32/discussions/6426

VojtechBartoska, 2022-09-21:
@SuGlider Please write what have been investigated and what is current resolution.

SuGlider, 2022-11-05:
The issue was related to a bad SoC.
We were able to track it to a call related to RTC setup, but that has nothig to do with real issue.
Therefore, this issue is closed now.

closed 2022-11-05 by SuGlider