espressif/arduino-esp32 #7168: ESP32 WROOM 32D problem with reseting all the time

domus12, 2022-08-22:
### Board

Custom board using ESP32 WROOM 32D

### Device Description

Schematic:
[https://obrazki.elektroda.pl/6373217700_1660836820.png](url)

### Hardware Configuration

ESP custom board

### Version

v2.0.4

### IDE Name

Arduino IDE

### Operating System

Windows 10

### Flash frequency

40MHz or 80 MHz

### PSRAM enabled

no

### Upload speed

all tested

### Description

Hello, I have a problem with the board based on ESP32-WROOM 32D. Namely, in Arduino I go after uploading the program, the Loading END message pops up, but the board keeps resetting:

`rst:0x7 (TG0WDT_SYS_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0030,len:1344
load:0x40078000,len:13516
load:0x40080400,len:3604
csum err:0x6a!=0xc2
ets_main.c 371 
ets Jun  8 2016 00:22:57

rst:0x7 (TG0WDT_SYS_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0030,len:1344
load:0xd0012000,len:3379
ho 0 tail 11 room 5
load:0xad0008e0,len:-51216124
1162 mmu set 00010000, pos 00010000
1162 mmu set 00020000, pos 00020000
1162 mmu set 00030000, pos 00030000
1162 mmu set 00040000, pos 00040000
1162 mmu set 00050000, pos 00050000
1162 mmu set 00060000, pos 00060000
1162 mmu set 00070000, pos 00070000
1162 mmu set 00080000, pos 00080000
1162 mmu set 00090000, pos 00090000
1162 mmu set 000a0000, pos 000a0000
1162 mmu set 000b0000, pos 000b0000
1162 mmu set 000c0000, pos 000c0000
1162 mmu set 000d0000, pos 000d0000
1162 mmu set 000e0000, pos 000e0000
1162 mmu set 000f0000, pos 000f0000
1162 mmu set 00100000, pos 00100000
1162 mmu set 00110000, pos 00110000
1162 mmu set 00120000, pos 00120000
1162 mmu set 00130000, pos 00130000
1162 mmu set 00140000, pos 00140000
1162 mmu set 00150000, pos 00150000
1162 mmu set 00160000, pos 00160000
1162 mmu set 00170000, pos 00170000
1162 mmu set 00180000, pos 00180000
1162 mmu set 00190000, pos 00190000
1162 mmu set ets Jun  8 2016 00:22:57 `

### Sketch

```cpp
I try with simple sketch Hello World

void setup() {
  // initialize digital pin LED_BUILTIN as an output.
  Serial.begin(115200);
}

// the loop function runs over and over again forever
void loop() {
Serial.println("Hello");              // wait for a second
}
```

### Debug Message

```plain
rst:0x7 (TG0WDT_SYS_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0030,len:1344
load:0x40078000,len:13516
load:0x40080400,len:3604
csum err:0x6a!=0xc2
ets_main.c 371 
ets Jun  8 2016 00:22:57

rst:0x7 (TG0WDT_SYS_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0030,len:1344
load:0xd0012000,len:3379
ho 0 tail 11 room 5
load:0xad0008e0,len:-51216124
1162 mmu set 00010000, pos 00010000
1162 mmu set 00020000, pos 00020000
1162 mmu set 00030000, pos 00030000
1162 mmu set 00040000, pos 00040000
1162 mmu set 00050000, pos 00050000
1162 mmu set 00060000, pos 00060000
1162 mmu set 00070000, pos 00070000
1162 mmu set 00080000, pos 00080000
1162 mmu set 00090000, pos 00090000
1162 mmu set 000a0000, pos 000a0000
1162 mmu set 000b0000, pos 000b0000
1162 mmu set 000c0000, pos 000c0000
1162 mmu set 000d0000, pos 000d0000
1162 mmu set 000e0000, pos 000e0000
1162 mmu set 000f0000, pos 000f0000
1162 mmu set 00100000, pos 00100000
1162 mmu set 00110000, pos 00110000
1162 mmu set 00120000, pos 00120000
1162 mmu set 00130000, pos 00130000
1162 mmu set 00140000, pos 00140000
1162 mmu set 00150000, pos 00150000
1162 mmu set 00160000, pos 00160000
1162 mmu set 00170000, pos 00170000
1162 mmu set 00180000, pos 00180000
1162 mmu set 00190000, pos 00190000
1162 mmu set ets Jun  8 2016 00:22:57
```

### Other Steps to Reproduce

I tried to change:
CPU freq 240 or 160 MHz
Flash Freq
PSRAM on/off

Nothing work.

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

atanisoft, 2022-08-22:
@domus12 Usually this indicates one of the flash pin is being used incorrectly. Since your schematic png omits many details on these pins it is impossible to assist.

domus12, 2022-08-22:
Okay, thank you for reply. I attach all pins:
[url=https://obrazki.elektroda.pl/3705590300_1661192828.png]

Is there a possibility to avoid that without schematic change? It is important to leave the schematic as is

atanisoft, 2022-08-22:
> Is there a possibility to avoid that without schematic change? It is important to leave the schematic as is

No, you are using flash exclusive pins for your TFT device. GPIO 6-11 (module pins 17-22) are connected to the flash embedded inside the module, in the datasheet for the module they are also recommended not to be used.

Per your updated schematic you should be able to adjust the three offending pins (TFT_SCL, TFT_SDO, TFT_SDI) to alternative pins.

domus12, 2022-08-22:
ad.1 So there isn't any option to change those pins 6-11 to different ones in program or use another type of flashing?
ad 2. Which pins are for SCL, SDO and SDI (LCD) in esp32?

atanisoft, 2022-08-22:
> So there isn't any option to change those pins 6-11 to different ones in program or use another type of flashing?

Nope, those are hardwired inside the module itself and can not be altered.

> Which pins are for SCL, SDO and SDI (LCD) in esp32?

That depends on the bus and your code. In my case I use SD-SPI for storage of files using the following pins:
GPIO 2 : MISO
GPIO 15: MOSI
GPIO 14: CLK
GPIO 13: CS

domus12, 2022-08-22:
@atanisoft 
So if I change:

-TFT SCL to GPIO 14
-TFT SDO to GPIO 2 
-TFT SDI to GPIO 15
-TFT CS to GPIO 13 

Everything will be working without resets?

atanisoft, 2022-08-22:
Yes, it should since you won't be using the flash pins.

domus12, 2022-08-22:
@atanisoft Thank You very much!

closed 2022-08-22 by domus12