espressif/arduino-esp32 #7175: ESP32C3 light sleep by UART

eos1d3, 2022-08-24:
### Board

ESP32-C3-DevKitM-1

### Device Description

ESP32-C3-DevKitM-1 connected with STM32-G030 via UART

### Hardware Configuration

Serial1 RX pin: GPIO3
Serial1 TX pin:  GPIO2

### Version

v2.0.4

### IDE Name

PlatformIO  Espressif 32 (5.1.0) > Espressif ESP32-C3-DevKitM-1

### Operating System

macOS 12

### Flash frequency

40MHz

### PSRAM enabled

no

### Upload speed

115200

### Description

There is a [solution](https://github.com/espressif/arduino-esp32/issues/6976) for this issue. This solution helps UART wake up. But there are still issues:

When you keep uploading the same sketch (without power reset),  or pressing reset button, sometimes wake up may cause system reset  with the following message:
```
SP-ROM:esp32c3-api1-20210207
Build:Feb  7 2021
rst:0x8 (TG1WDT_SYS_RST),boot:0xe (SPI_FAST_FLASH_BOOT)
```
When restarting happens, it may keep on restarting for a long time, or may resume normal wake up after some restarting. And it may restart again.

But It seems power reset does not have this restarting problem. Just unplug the USB cable and it mostly works.

Another problem is that after wake up, UART data will become garbage. For example, when I send 14 bytes to wake up it mostly receive 8 bytes and all are garbage. So what is the use of UART wake up?
 

### Sketch

```cpp
void setup()
{
    Serial.begin(115200);
    Serial1.begin(115200, SERIAL_8N1, RXD2, TXD2);
    while (!Serial || !Serial1) {
        delay(500);
    }

    // GPIO 20 is the default UART_0 RX pin
    gpio_sleep_set_direction(GPIO_NUM_3, GPIO_MODE_INPUT);
    gpio_sleep_set_pull_mode(GPIO_NUM_3, GPIO_PULLUP_ONLY);

    int wake_thresh = 3;
    uart_set_wakeup_threshold(UART_NUM_1, wake_thresh);
    esp_sleep_enable_uart_wakeup(UART_NUM_1);
    Serial.flush();
    Serial1.flush();   
}
```

### Debug Message

```plain
ESP-ROM:esp32c3-api1-20210207
Build:Feb  7 2021
rst:0x8 (TG1WDT_SYS_RST),boot:0xe (SPI_FAST_FLASH_BOOT)
Saved PC:0x40380000
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fcd6100,len:0x438
load:0x403ce000,len:0x918
load:0x403d0000,len:0x24e4
Attempting to boot anyway...
entry 0x403ce000
ESP32 Chip model = ESP32-C3 Rev 3
SDK version = v4.4.1-472-gc9140caf8c
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

SuGlider, 2022-08-26:
@eos1d3 - I tried this and I can't reproduce the issue.

Please, make sure you are using Arduino Core 2.0.4, because this output `SDK version = v4.4.1-472-gc9140caf8c` is not for 2.0.4

SuGlider, 2022-08-26:
`TG1WDT_SYS_RST` is usually related to using SPI Flash pins or some task that takes too long to "return".

eos1d3, 2022-08-27:
Today I have more tests. And to simplify the test, I focus on GPIO light sleep first.
The sketch is listed below:
```
#include <Arduino.h>
#include <HardwareSerial.h>
#include <WiFi.h>
#include "time.h"
#include <esp_sleep.h>
#include <driver/uart.h>
#include <esp_task_wdt.h>

#define LED1 0
#define LED_UART 1
#define LED_RUN 10
#define RXD2 3
#define TXD2 2
#define WAKE_PIN GPIO_NUM_9
#define RX_DATA_MAX_SIZE 10

unsigned long ledUARTLastChange;
uint8_t rxData[RX_DATA_MAX_SIZE];

void setup()
{
    Serial.begin(115200);
    Serial1.begin(115200, SERIAL_8N1, RXD2, TXD2);
    while (!Serial || !Serial1)
    {
        delay(500);
    }

    pinMode(LED1, OUTPUT);
    pinMode(LED_UART, OUTPUT);
    pinMode(LED_RUN, OUTPUT);
    pinMode(WAKE_PIN, INPUT_PULLUP);

    Serial.printf("GPIO Wake up ok? %d\n", esp_sleep_is_valid_wakeup_gpio(WAKE_PIN));
    gpio_wakeup_enable(WAKE_PIN, GPIO_INTR_HIGH_LEVEL);
    esp_sleep_enable_gpio_wakeup();
}

void loop()
{
    if (!digitalRead(WAKE_PIN))
    {
        Serial.println("== Sleep ==");
        Serial.flush();
        Serial1.flush();
        esp_light_sleep_start();

        Serial.println("== Wake up ==");
    }
}

```

And `platformio.ini` is

_[env:esp32-c3-devkitm-1]
platform = espressif32@5.1.0
board = esp32-c3-devkitm-1
framework = arduino
monitor_speed = 115200
monitor_port = /dev/cu.usbserial-14140
upload_speed = 1500000
upload_port = /dev/cu.usbserial-14140_

I tried `espressif32@5.1.1,` `espressif32@5.1.0` and also `espressif32@4.4.0`
5.1.0 is said [Added support for Arduino v2.0.4](https://github.com/platformio/platform-espressif32/releases)

And all these versions will lock up the CPU and no console message after just a few press of the test button. Mostly it will not last for 10 times of button press.

Then I tried ESP32 IDF 4.4.2 and the offical `lightsleep` example. I do not modify the code and it also has the same lock up issue.

Finally I tried IDF 5.0 beta1 with the same `lightsleep` example, only this version works normally without any lock up.

SuGlider, 2022-08-27:
> #define LED1 0 

GPIO 0 is used by the board to reset it when the sketch is uploaded.

SuGlider, 2022-08-27:
>#define WAKE_PIN GPIO_NUM_9 
>#define RX_DATA_MAX_SIZE 10

GPIO 9 and 10 are used to access Flash.
It is used in QIO mode, but not in DIO.

eos1d3, 2022-08-28:
@SuGlider Hi, thanks for the information.

GPIO0 can be used and is not reset purpose. While GPIO9 is the boot button and it is used by ESP32 IDF `lightsleep` example.  So GPIO0 and GPIO9 are not the root of cause.

But from your suggestion, I found something:

1. when using GPIO9 as wakeup pin, if I use on board button, there is no lockup issue.
2. when using GPIO9 as wakeup pin, and connect  it via a jumper wire with 0.1uF cap to a breadboard. The filter cap must create the lockup issue in around 5 press for any of the buttons. When I remove the cap, it is much better but it will lockup in 100 times or more.
3. Now use GPIO4 as wake up pin, with the 0.1 cap. it is not easy to get the lock up but it will lock up finally. Removing the cap seems no lock up.

So can this be noise issue of the jumper wire (15cm) with the breadboard? But it is strange that using ESP IDF 5.0 beta1, no lock up occurs.

SuGlider, 2022-08-28:
@eos1d3 - nevermind.... I didn't see that you are using the C3. GPIO 9 and 10 are problem for the ESP32.

SuGlider, 2022-08-28:
>upload_speed = 1500000
>upload_port = /dev/cu.usbserial-14140

I can see that the upload speed is not standard:
Please change it to 921600 
`upload_speed = 921600`

SuGlider, 2022-08-28:
>Then I tried ESP32 IDF 4.4.2 and the offical lightsleep example. I do not modify the code and it also has the same lock up issue.
Finally I tried IDF 5.0 beta1 with the same lightsleep example, only this version works normally without any lock up.

This is interesting.... I'll try it as well.

Can you also send a picture of your board/circuit?
The `STM32-G030` is also configured at 115200 - 8N1?
How is it connected? Common GND is used?

SuGlider, 2022-08-28:
>And all these versions will lock up the CPU and no console message after just a few press of the test button. Mostly it will not last for 10 times of button press.

There is the button bouncing effect... Not sure how it affects your exemple.

eos1d3, 2022-08-28:
Hi,

Many thanks for prompt help!

`upload_speed = 1500000` is ESP32C3 flash upload speed. It will be the same if I use 115200. And I also see `Hash of data verified` after upload. I use this speed for all espressif boards for almost a year as all of them can reach this speed during upload. (not all other boards can reach this)

For STM32G0, TX, RX and GND from the development board are connected directly to ESP32C3. It is very simply with just 3 wires. Both sides are using 115200 baud, 8N1. They are working for months and never have any error until I start testing ESP32C3 light sleep.

I know bouncing effect. And the bouncing effect will cause problem even IDF v5.0b1. So I always have 0.1uF cap added to the pin. With the filter cap, IDF 5.0b1 works without any issue. But with the same wiring, arudino-esp32 and IDF 4.4.2 just do not work.

Please try this:

1. An ESP32C3 board
2. Connect a wire to GPIO9 to a breadboard with a toggle switch
3. Add a 0.1uF between this wire and GND.
4. Pressing on-board button or external toggle switch will lock up immediately
5. Remove the cap, lock up almost gone but still there
6. Removing the wire from GPIO, press only onboard button, no lockup 

Thanks!

VojtechBartoska, 2023-02-16:
Hello,

can this be closed @SuGlider & @eos1d3?

eos1d3, 2023-02-16:
Hi, I have switched to ESP IDF and do not have time to check this issue. If I have new finding in the future, I will update this issue. Thanks!

closed 2023-02-16 by eos1d3