espressif/arduino-esp32 #7176: USB CDC function serial/vendor example not work for the usb device

zhuyeaini9, 2022-08-24:
### Board

ESP32S3

### Device Description

USB interface device used to measure magnet.
this device has one ftdi chip in it.

### Hardware Configuration

GPIO 19 20 connect to the device's usb port.

### Version

v2.0.4

### IDE Name

arduino ide 1.8.19

### Operating System

win10

### Flash frequency

80

### PSRAM enabled

no

### Upload speed

921600

### Description

I used the esp32s3 board's GPIO 19 20 connect to my usb device.
the usb device use one usb to uart chip(ft231xq)
the usb device can not find via the usb example usb serial or usb vendor.
it can only detected the unplug event of my usb device.

in fact,I report this similar issue at esp32-idf which has solved:
https://github.com/espressif/esp-idf/issues/8645
the esp32-idf's usb example has one:
usb\host\cdc\cdc_acm_vcp
which works well with my device.

any advice or help is welcome.

### Sketch

```cpp
example of official usb serial/usb vendor.
```

### Debug Message

```plain
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x1 (POWERON),boot:0x8 (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fce3808,len:0x43c
load:0x403c9700,len:0xbec
load:0x403cc700,len:0x2a3c
SHA-256 comparison failed:
Calculated: 3611f330726ad4edf64f62982b0568332a5800bc59425cab90be5379db92c108
Expected: ffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffff
Attempting to boot anyway...
entry 0x403c98d8
[   110][D][esp32-hal-tinyusb.c:673] tinyusb_enable_interface(): Interface CDC enabled
[   110][D][esp32-hal-tinyusb.c:673] tinyusb_enable_interface(): Interface MSC enabled
[   114][D][esp32-hal-tinyusb.c:673] tinyusb_enable_interface(): Interface VENDOR enabled
[   153][D][FirmwareMSC.cpp:94] msc_update_get_required_disk_sectors(): APP size: 329952 (645 sectors)
[   153][D][FirmwareMSC.cpp:100] msc_update_get_required_disk_sectors(): OTA size: 1310720 (2560 sectors)
[   160][D][FirmwareMSC.cpp:113] msc_update_get_required_disk_sectors(): USING FAT12
[   167][D][FirmwareMSC.cpp:116] msc_update_get_required_disk_sectors(): FAT sector size: 512
[   176][D][FirmwareMSC.cpp:117] msc_update_get_required_disk_sectors(): FAT data sectors: 3221
[   184][D][FirmwareMSC.cpp:118] msc_update_get_required_disk_sectors(): FAT table sectors: 10
[   192][D][FirmwareMSC.cpp:119] msc_update_get_required_disk_sectors(): FAT total sectors: 3233 (1616KB)
[   202][D][esp32-hal-tinyusb.c:673] tinyusb_enable_interface(): Interface DFU enabled
[   210][D][esp32-hal-tinyusb.c:562] tinyusb_load_enabled_interfaces(): Load Done: if_num: 5, descr_len: 139, if_mask: 0x1b
USB UNPLUGGED
SB UNPLUGGED
USB UNPLUGGED
USB UNPLUGGED
USB UNPLUGGED
USB UNPLUGGED
USB UNPLUGGED
USB UNPLUGGED
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

me-no-dev, 2022-08-24:
We do not currently support USB Host functionality. It seems that is what you need, but instead you enabled the USB Device (through TinyUSB)

zhuyeaini9, 2022-08-25:
got it.
hope the usb host function coming soon.
thanks.

closed 2022-08-25 by zhuyeaini9