espressif/arduino-esp32 #7177: I2S mic not working anymore

MikeBailleul, 2022-08-24:
### Board

ESP32 Dev Module

### Device Description

Attached to an INMP441 MEMS microphone

### Hardware Configuration

INMP441 is configured as follow:
WS -> 25
SCK -> 32
SD -> 33
L/R -> GND
GND -> GND
VDD -> 3.3V

### Version

v2.0.4

### IDE Name

Arduino & PlatformIO

### Operating System

macOS 12.5.1

### Flash frequency

40Mhz

### PSRAM enabled

no

### Upload speed

460800

### Description

Microphone does not work anymore with version 2.0.4.
When switching back to 2.0.3, it works normally.

### Sketch

```cpp
https://github.com/atomic14/esp32-i2s-mic-test/blob/main/i2s_mic_test/i2s_mic_test.ino
```

### Debug Message

```plain
None
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

SuGlider, 2022-08-24:
@PilnyTomas - Could you please help to triage this issue?

PilnyTomas, 2022-08-24:
Not much to work with here. There was no change in the top-level library, maybe IDF problem.
@MikeBailleul Please post a complete and minimal sketch + picture of your connection.
The clock pin is by default set to GPIO 14 and data input is default to GPIO 35 but I assume you have changed that since you are connecting to 32 and 33.

Jason2866, 2022-08-24:
@MikeBailleul Can you try with the branch https://github.com/espressif/arduino-esp32/tree/idf-release/v4.4 ? We are using a i2s mic for Tasmota and it does work. See [example](https://github.com/arendst/Tasmota/discussions/16226) For Tasmota we use our own Arduino framework build which is based on a later IDF version than the actual core 2.0.4. Thats why i suggested to try the newest build.

MikeBailleul, 2022-08-24:
@PilnyTomas the sketch and pins in my first post are the one I use.
I have tried this setup by changing only the arduino-esp32 version (from 1.0.0 to 2.0.4) and the only one that does NOT work and produce unexpected output is 2.0.4.
Also, I am not the only having this issue: https://github.com/atomic14/esp32-i2s-mic-test/issues/2#issuecomment-1200308996

@Jason2866 sure I will be happy to test it out, could you explain how to use this specific release into Arduino software?
As of now, my choices are limited to the one already built/released:
[image]
I have used [the doc](https://docs.espressif.com/projects/arduino-esp32/en/latest/installing.html) that has those URLs:
```
https://raw.githubusercontent.com/espressif/arduino-esp32/gh-pages/package_esp32_index.json
https://raw.githubusercontent.com/espressif/arduino-esp32/gh-pages/package_esp32_dev_index.json
```

Jason2866, 2022-08-24:
@MikeBailleul  Do not ask me about ArduinoIDE ;-) I never used it. I do everything with VSC and PlatformIO. With Platformio it is easy, i just reference the branch and if needed the specific commit.

MikeBailleul, 2022-08-24:
The only reason I still use ArduinoIDE sometimes is for the plot feature (I'm not aware of such a feature in PlatformIO, maybe you know?), this is handy when working with sound, for visualisation :)

I had no idea I could reference the branch/commit in platformIO, neat feature. 
Not sure if the syntax I use is correct though:
```
platform = espressif32
platform_packages = platform_packages @ https://github.com/espressif/arduino-esp32.git#idf-release/v4.4
```
If this is correct, the issue persists.
Can you provide the syntax to use a specific commit? I can try to identify from which commit the issue appears.

Jason2866, 2022-08-24:
The syntax is not correct. Use
```
platform = espressif32
platform_packages = framework-arduinoespressif32 @ https://github.com/espressif/arduino-esp32.git#idf-release/v4.4
```

Jason2866, 2022-08-24:
And for a commit
```
....https://github.com/espressif/arduino-esp32.git#d22bcb765b35f8a1869a5ed185886eec77e3fa21
```

Jason2866, 2022-08-24:
If serial plotter is the only reason https://community.platformio.org/t/plotting-using-platformio/6597/8

MikeBailleul, 2022-08-24:
I identified it, from this commit, the microphone stops working: https://github.com/espressif/arduino-esp32/commit/666c66d3d1c7437aaebaaadf2084927b7fc476c4

I also tried the branch you suggested with the more recent IDF but it does not work.

Jason2866, 2022-08-25:
This is the correction of flash modes. Before always the same boot loader was flashed.
So always the same flash mode was used. Try all flash modes for your board.
Can be set in your platformio.ini to override the default
```
[env:esp32]
platform = https://github.com/......
platform_packages = framework-arduinoespressif32 @ https.....
board = esp32dev
board_build.flash_mode = qio
framework = arduino
monitor_speed = 115200
```

PilnyTomas, 2022-08-25:
> ### Sketch
> 
> ```c++
> https://github.com/atomic14/esp32-i2s-mic-test/blob/main/i2s_mic_test/i2s_mic_test.ino
> ```

Sorry, I missed that.
I see that you are using IDF original I2S driver. So that rules out Arduino ... I suggest opening issue in IDF: https://github.com/espressif/esp-idf/issues/new/choose

vernonet, 2022-10-18:
I have a similar problem, I had to add the code:
```
#if ESP_ARDUINO_VERSION >= ESP_ARDUINO_VERSION_VAL(2, 0, 4)
  #define I2S_MIC_CHANNEL            I2S_CHANNEL_FMT_ONLY_RIGHT 
#else
  #define I2S_MIC_CHANNEL            I2S_CHANNEL_FMT_ONLY_LEFT
#endif  
........
i2s_config_t i2s_config = {
........
  .channel_format = I2S_MIC_CHANNEL, 
........
};
```
That is, you need to replace "I2S_CHANNEL_FMT_ONLY_LEFT" with "I2S_CHANNEL_FMT_ONLY_RIGHT"

PilnyTomas, 2022-11-09:
Seeing that the IDF team took responsibility for this issue, I am closing this one.

closed 2022-11-09 by PilnyTomas

BlueAndi, 2024-12-18:
Is this solved meanwhile in the release/v2.0.x branch?