espressif/arduino-esp32 #7179: Support for vTaskList() in Arduino core so we can monitor task status

osamahnini, 2022-08-24:
### Related area

FreeRTOS

### Hardware specification

All esp32 devices

### Is your feature request related to a problem?

No

### Describe the solution you'd like

Current when you use vTaskList() FreeRTOS API and try to compile, you get this error: "undefined reference".
The runtime stats must be enabled in FreeRTOS so you can use this API. I think the solution is to enable runtime stats by default.

### I have checked existing list of Feature requests and the Contribution Guide

- [X] I confirm I have checked existing list of Feature requests and Contribution Guide.

me-no-dev, 2022-08-25:
We need to check if this will cause any performance/memory degradation first.

qnxsgwy, 2022-09-08:
If can view the cpu usage, is very useful for debugging, hope to support this function. 
But performance degradation also needs attention，hopefully as an option that can be on/off at compile time.

fknrdcls, 2022-09-09:
Well, I'm no stranger to squeezing performance out of the ESP32 and I use it by default on my projects.
48kHz stereo realtime MP3 encoder: https://www.esp32.com/viewtopic.php?f=2&t=9186
Using it to gauge performance really didn't notice any issues at all. I think there is certainly more benefit than potential issues. Also Arduino being something that is also used as a learning tool, it would be extremely beneficial to allow users to see tasks usage and CPU affinity. Happy to add an example if it's added.

davepl, 2022-09-16:
I too would very much appreciate this support in the Arudino variant... just hit this omission myself!

szerwi, 2022-11-01:
+1, I would also need that feature in my project

everslick, 2022-11-23:
me 2

yekomS, 2022-12-04:
also need this feature :-)

marnov999, 2022-12-31:
so any new info on this? I would need it too, or maybe some way to recompile the lib ourself with the feature enabled?

davepl, 2022-12-31:
I wound up making my own idle threads that whack the watchdog and track whatever idle cycles they can burn, and that gives me overall CPU, but not per-task…

bobemoe, 2023-02-22:
Yeah defo missing this too, just started using RTOS tasks, and need a way to list and monitor them. This looks like just the ticket.

synologic, 2023-06-24:
All, this has been discussed all over the internet for quite a while now, we should get it :)

VojtechBartoska, 2024-02-07:
Hello all, changing this to 3.1.0 milestone, we are starting investigating of this is doable and let you know. 

@lucasssvaz will do the research and share the updates later here.

lucasssvaz, 2024-02-20:
We'll first need to implement some way to easily reconfigure and recompile the libs for the users. Currently working on that.

me-no-dev, 2024-10-24:
Will something like that work? https://github.com/espressif/arduino-esp32/pull/10515#issue-2610981384

hitecSmartHome, 2024-10-24:
It would be the best if it would return or populate an array or json instead of printing it out. That way we could actually do something with the results

me-no-dev, 2024-10-24:
@hitecSmartHome that is up to you to implement. You can use the code in the PR as starting point. Array is already used there and that is what is printed.

hitecSmartHome, 2024-10-24:
I see.

closed 2024-11-14 by lucasssvaz