espressif/arduino-esp32 #7182: setCpuFrequencyMhz() changes Serial bauds if frequency<80Mhz

savejeff, 2022-08-25:
### Board

ESP32-S3 DevKitC-1

### Device Description

plane ESP32-S3 DevKitC-1 with nothing attached

### Hardware Configuration

.

### Version

v2.0.4

### IDE Name

Arduini IDE

### Operating System

Windows 10

### Flash frequency

all of them

### PSRAM enabled

no

### Upload speed

921600

### Description

the Issued descripbed in this [issue](https://github.com/espressif/arduino-esp32/issues/6032) is still not fixed for (at least) the ESP32-S3

The problem is with every CPU Freq below 80Mhz, the Baudrate must be adjusted up to compensate slower clock speeds. 

### Sketch

```cpp
uint32_t Freq = 0;
const int bauds = 115200;
int my_bauds;
int cpufreqs[6] = {240, 160, 80, 40, 20, 10};
int i = 0;

void setup() {
  Freq = getCpuFrequencyMhz();
  if (Freq < 80) {
    my_bauds = 80 / Freq * bauds;
  }
  else {
    my_bauds = bauds;
  }

  Serial.begin(my_bauds);        // Attention dépend de la frequence CPU si elle est <80Mhz 
  delay(500);
  //
  Freq = getCpuFrequencyMhz();
  Serial.print("CPU Freq = ");
  Serial.print(Freq);
  Serial.println(" MHz");
  Freq = getXtalFrequencyMhz();
  Serial.print("XTAL Freq = ");
  Serial.print(Freq);
  Serial.println(" MHz");
  Freq = getApbFrequency();
  Serial.print("APB Freq = ");
  Serial.print(Freq);
  Serial.println(" Hz");
}
void loop() {
  Serial.print("\nchange CPU freq to ");
  Serial.println(cpufreqs[i]);
  delay(1000);
  setCpuFrequencyMhz(cpufreqs[i]);
  Freq = getCpuFrequencyMhz();
  Serial.print("Send to serial with my_bauds = ");
  Serial.println(my_bauds);
  Serial.print("CPU Freq = ");
  Serial.print(Freq);
  Serial.println(" MHz");
  //  Change bauds rate ?

  if (Freq < 80) {
    my_bauds = 80 / Freq * bauds;
  }
  else {
    my_bauds = bauds;
  }
  Serial.end();
  delay(1000);
  Serial.begin(my_bauds);
  delay(1000);
  Serial.print("\nchange my_bauds to ");
  Serial.println(my_bauds);
  Serial.print("Serial set to ");
  Serial.println(my_bauds);
  Serial.print("CPU Freq = ");
  Serial.print(Freq);
  Serial.println(" MHz");
  Serial.print("my_bauds = ");
  Serial.println(my_bauds);

  delay(1000);
  i++;
  i = i % 6;
}
```

### Debug Message

```plain
change CPU freq to 240
⸮Bŭk⸮⸮⸮
change my_bauds to 115200
Serial set to 115200
CPU Freq = 240 MHz
my_bauds = 115200

change CPU freq to 160
Send to serial with my_bauds = 115200
CPU Freq = 160 MHz

change my_bauds to 115200
Serial set to 115200
CPU Freq = 160 MHz
my_bauds = 115200

change CPU freq to 80
Send to serial with my_bauds = 115200
CPU Freq = 80 MHz

change my_bauds to 115200
Serial set to 115200
CPU Freq = 80 MHz
my_bauds = 115200

change CPU freq to 40
f⸮⸮⸮`~⸮⸮
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

SuGlider, 2022-08-26:
I see that it has not been implemented for C3. That's the issue.

savejeff, 2022-08-26:
What do you mean "for C3" ? 
If you are talking about ESP32-C3: the problem is present with all variants from ESP32, ESP32-C3 and ESP32-S3.

About the I2C and SPI issue, i have only tested this on the C3. S3 and S might also have the same issues

SuGlider, 2022-08-26:
Just a quick question... Are you using any ESP at lower than 80MHz?

Please explain the usage.

We will verify it for all SoC.
But it is not sure that it will be implemented because there is recommendation from IDF regarding it.

C3 is RiskV and has a different internal structure regarding CPU Frequency.

S2 and S3, the same regarding different internal structure.

Therefore, I'm sure if it is possible to solve this issue

savejeff, 2022-08-27:
> Just a quick question... Are you using any ESP at lower than 80MHz?
> 
> Please explain the usage.

In general i don't think there needs to be an specific application when we talk about low level functions like these. they should IMHO work as expected to allow fast and reliable development without pitfalls. 

But I have in fact an actual application: My device is battery-powered to record data. between recordings, the device moves into a standby mode where everything essential is kept running but with reduced power consumption. so i reduce the esp clock as low as possible where still the important measurements can be performed. at the start of the next recording, the clock is increased and everything runs on full steam again.

> We will verify it for all SoC. But it is not sure that it will be implemented because there is recommendation from IDF regarding it.
> 
> C3 is RISC V and has a different internal structure regarding CPU Frequency.
> 
> S2 and S3, the same regarding different internal structure.
> 
> Therefore, I'm **not** sure if it is possible to solve this issue

Why would it not be possible? the code above is a working fix for all architectures. 
The simplest implementation would be to not directly change the baud/clock rates in the setCpuFrequencyMhz but in the updateBaudRate function of the serial class. here is only the simple fix
```
virtual void updateBaudRate(unsigned long baud) {
  int Freq = getCpuFrequencyMhz();
  if (Freq < 80) {
    baud= 80 / Freq * baud;
  }

  [...]
}
```

needs to be used. 

SPI is even simpler. SPI.beginTransaction receives the desired clock rate in the SPISettings parameter. so analog to the Serial fix, a simple correction of the clock speed would be sufficient I think.

So a change below 80Mhz would look like this: 
```

setCpuFrequencyMhz(20);
Serial.updateBaudRate(115200);

```

SuGlider, 2022-08-27:
@savejeff - I use Arduino Core 2.0.4 and tested this sketch with ESP32, ESP32-S2, ESP32-S3 and ESP32-C3:

``` cpp
void setup() {
  int cpufreqs[6] = {240, 160, 80, 40, 20, 10};

  Serial.begin(115200);        // Attention dépend de la frequence CPU si elle est <80Mhz
  delay(1000); // just a second to make sure that we will get all messages in the Serial Terminal
  Serial.println("\n Starting...\n");
  Serial.flush();
  // initial information
  uint32_t Freq = getCpuFrequencyMhz();
  Serial.print("CPU Freq = ");
  Serial.print(Freq);
  Serial.println(" MHz");
  Freq = getXtalFrequencyMhz();
  Serial.print("XTAL Freq = ");
  Serial.print(Freq);
  Serial.println(" MHz");
  Freq = getApbFrequency();
  Serial.print("APB Freq = ");
  Serial.print(Freq);
  Serial.println(" Hz");
  delay(500);
#ifdef CONFIG_IDF_TARGET_ESP32C3
  uint8_t firstFreq = 1;
#else
  uint8_t firstFreq = 0;
#endif
  for (uint8_t i = firstFreq; i < sizeof(cpufreqs) / sizeof(int); i++) {
    Serial.printf("\n------- Trying CPU Freq = %d ---------\n", cpufreqs[i]);
    Serial.flush();  // wait to empty the UART FIFO before changing the CPU Freq.
    setCpuFrequencyMhz(cpufreqs[i]);
    Serial.updateBaudRate(115200);
    //
    Freq = getCpuFrequencyMhz();
    Serial.print("CPU Freq = ");
    Serial.print(Freq);
    Serial.println(" MHz");
    Freq = getXtalFrequencyMhz();
    Serial.print("XTAL Freq = ");
    Serial.print(Freq);
    Serial.println(" MHz");
    Freq = getApbFrequency();
    Serial.print("APB Freq = ");
    Serial.print(Freq);
    Serial.println(" Hz");
    Serial.println("Moving to the fext frequency ... after pause of 2 seconds.");
    delay(2000);
  }
  Serial.println("\n-------------------\n");
  Serial.println("End of testing...");
  Serial.println("\n-------------------\n");
}
void loop() {
}

```

### Results:

It works perfectly fine with ESP32 and ESP32-S2.
It fails with ESP32-S3 and ESP32-C3 under 40MHz. 
Therefore, we must implement it for S3 and C3 only.

SuGlider, 2022-08-27:
It is necessary to check what is the source of Clock for SPI, I2C and others, in order to make sure it works fine under APB Freq (80MHz).

SuGlider, 2022-08-27:
> So a change below 80Mhz would look like this:
```
setCpuFrequencyMhz(20);
Serial.updateBaudRate(115200);
```

As per the example above used for testing, this exactly how it will work for the UART peripheral.

savejeff, 2022-09-02:
> int Freq = getCpuFrequencyMhz();
>   if (Freq < 80) {
>     baud= 80 / Freq * baud;
>   }

I think the most important thing is that Serial works as expected. i2C and SPI has a clock speed determined by the master, so the communication is not broken, just slowed down. A fix here would of course be best.

When i find time I'll check by what factor SPI is slowed down. maybe its as simple as with Serial. 

Thanks for taking up the issue

SuGlider, 2022-11-20:
@savejeff - PR #7496 fixes this UART issue. Please feel free to open a new issue regarding I2C and/or SPI, separately, if necessary.

PR #7496 (2022-12-13): fixes UART at lower than 80MHz | S3 C3
## Description of Change
This PR fixes UART baud rate when ESP32-S3 and ESP32-C3 are running under 80MHz.

## Tests scenarios

Using ESP32, ESP32-S2, ESP32-S3 and ESP32-C3

Testing sketch:

``` cpp
uint32_t Freq = 0;
const int bauds = 115200;
uint8_t cpufreqs[] = {240, 160, 80, 40, 20, 10};
int i = 0;

void setup() {
  Freq = getCpuFrequencyMhz();

  Serial.begin(bauds);
  delay(500);

  Freq = getCpuFrequencyMhz();
  Serial.print("CPU Freq = ");
  Serial.print(Freq);
  Serial.println(" MHz");
  Freq = getXtalFrequencyMhz();
  Serial.print("XTAL Freq = ") [... cut]
closed 2022-12-13 by me-no-dev