espressif/arduino-esp32 #7185: Chip Temperature with ESP.getChipTemp()

savejeff, 2022-08-26:
### Related area

System/Chip

### Hardware specification

ESP32-S3

### Is your feature request related to a problem?

no

### Describe the solution you'd like

i propose a new getter function to the ESP class to get the chip internal temperature. 

here is an example function to read the chip internal temperature sensor:

```

#include "soc/adc_periph.h"

float temprature_sens_read()
{
	*((uint32_t*) SENS_SAR_TSENS_CTRL_REG) |= SENS_TSENS_DUMP_OUT;
	
	ulong time0 = millis();
	while(true)
	{
		if(millis() - time0 > 250)
			return -1;

		bool ready = *((uint32_t*) SENS_SAR_TSENS_CTRL_REG) & SENS_TSENS_READY;

		
		if(ready)
		{
			uint8_t value = *((uint32_t*) SENS_SAR_TSENS_CTRL_REG) & SENS_TSENS_OUT;
			float temp = 0.4386 * value - 20.52;

			*((uint32_t*) SENS_SAR_TSENS_CTRL_REG) &= ~SENS_TSENS_DUMP_OUT;

			return temp;
		}
	}
}

```

Improvements might be to check the offset value as described in [TRM](https://www.espressif.com/sites/default/files/documentation/esp32-s3_technical_reference_manual_en.pdf) on page 1164

### I have checked existing list of Feature requests and the Contribution Guide

- [X] I confirm I have checked existing list of Feature requests and Contribution Guide.

SuGlider, 2022-08-27:
Thanks @savejeff - We will analyze and consider your proposal!
Feel free to add a PR if you wish so.

SuGlider, 2022-08-27:
@P-R-O-C-H-Y, would you like to evaluate it?

savejeff, 2022-08-27:
> Thanks @savejeff - We will analyze and consider your proposal! Feel free to add a PR if you wish so.

Thx ;] 
Im not familiar enough with the code structure to implement this myself good enough. 
I investigated how to read the internal chip temp and only found "this is not implemented yet" so i thought i put out an example implementation. 

There is improvement potential with the timeout. I have not checked what the real "time to ready" is and if it is a fixed time that can just be waited for to avoid deadlock without extra variables and code.

P-R-O-C-H-Y, 2022-09-02:
Hi @savejeff, 

you can take a look into file [esp32-hal-mics.c](https://github.com/espressif/arduino-esp32/blob/master/cores/esp32/esp32-hal-misc.c#L67) where the function `temperatureRead()` is. Maybe what we can improve, is an option to be able to set the range (offset) for the chip temperature reading. Now the default is used an d no option to change that.
`default used:  offset =  0, measure range:-10℃ ~  80℃, error < 1℃ `

P-R-O-C-H-Y, 2022-09-06:
@savejeff What version of ESP32-Arduino core do you use?

savejeff, 2022-09-07:
I originally test this on PlatformIO with the latest core version (equivalent to Arduino 2.0.4)

But i tested this also with Arduino IDE with the latest core version 2.0.4 and I'm getting the same linker error:

```

c:/users/save_/appdata/local/arduino15/packages/esp32/tools/xtensa-esp32s3-elf-gcc/gcc8_4_0-esp-2021r2-patch3/bin/../lib/gcc/xtensa-esp32s3-elf/8.4.0/../../../../xtensa-esp32s3-elf/bin/ld.exe: core\core.a(esp32-hal-misc.c.o):(.literal.temperatureRead+0x4): undefined reference to `temp_sensor_set_config'
c:/users/save_/appdata/local/arduino15/packages/esp32/tools/xtensa-esp32s3-elf-gcc/gcc8_4_0-esp-2021r2-patch3/bin/../lib/gcc/xtensa-esp32s3-elf/8.4.0/../../../../xtensa-esp32s3-elf/bin/ld.exe: core\core.a(esp32-hal-misc.c.o):(.literal.temperatureRead+0x8): undefined reference to `temp_sensor_start'
c:/users/save_/appdata/local/arduino15/packages/esp32/tools/xtensa-esp32s3-elf-gcc/gcc8_4_0-esp-2021r2-patch3/bin/../lib/gcc/xtensa-esp32s3-elf/8.4.0/../../../../xtensa-esp32s3-elf/bin/ld.exe: core\core.a(esp32-hal-misc.c.o):(.literal.temperatureRead+0xc): undefined reference to `temp_sensor_read_celsius'
c:/users/save_/appdata/local/arduino15/packages/esp32/tools/xtensa-esp32s3-elf-gcc/gcc8_4_0-esp-2021r2-patch3/bin/../lib/gcc/xtensa-esp32s3-elf/8.4.0/../../../../xtensa-esp32s3-elf/bin/ld.exe: core\core.a(esp32-hal-misc.c.o):(.literal.temperatureRead+0x10): undefined reference to `temp_sensor_stop'
c:/users/save_/appdata/local/arduino15/packages/esp32/tools/xtensa-esp32s3-elf-gcc/gcc8_4_0-esp-2021r2-patch3/bin/../lib/gcc/xtensa-esp32s3-elf/8.4.0/../../../../xtensa-esp32s3-elf/bin/ld.exe: core\core.a(esp32-hal-misc.c.o): in function `temperatureRead':
C:\Users\Save_\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.4\cores\esp32/esp32-hal-misc.c:71: undefined reference to `temp_sensor_set_config'
c:/users/save_/appdata/local/arduino15/packages/esp32/tools/xtensa-esp32s3-elf-gcc/gcc8_4_0-esp-2021r2-patch3/bin/../lib/gcc/xtensa-esp32s3-elf/8.4.0/../../../../xtensa-esp32s3-elf/bin/ld.exe: C:\Users\Save_\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.4\cores\esp32/esp32-hal-misc.c:72: undefined reference to `temp_sensor_start'
c:/users/save_/appdata/local/arduino15/packages/esp32/tools/xtensa-esp32s3-elf-gcc/gcc8_4_0-esp-2021r2-patch3/bin/../lib/gcc/xtensa-esp32s3-elf/8.4.0/../../../../xtensa-esp32s3-elf/bin/ld.exe: C:\Users\Save_\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.4\cores\esp32/esp32-hal-misc.c:73: undefined reference to `temp_sensor_read_celsius'
c:/users/save_/appdata/local/arduino15/packages/esp32/tools/xtensa-esp32s3-elf-gcc/gcc8_4_0-esp-2021r2-patch3/bin/../lib/gcc/xtensa-esp32s3-elf/8.4.0/../../../../xtensa-esp32s3-elf/bin/ld.exe: C:\Users\Save_\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.4\cores\esp32/esp32-hal-misc.c:74: undefined reference to `temp_sensor_stop'
collect2.exe: error: ld returned 1 exit status
exit status 1

```

P-R-O-C-H-Y, 2022-09-07:
Yes you are right, I see same compilation issue on 2.0.4. However on master branch, its already fixed. So in 2.0.5 it will be available for ESP32S3. But I am getting some weird temperatures read, I will test it more further and find the issue. If you can, please retest it in Arduino IDE with ESP32 core master branch :)

savejeff, 2022-09-07:
What board are you using? temperature readings seem to not work with my devkit boards but with the wroom module i have on a customer layout.

about the master branch, i don't use the Arduino IDE (only to check before i start an issue on github). as far as i know there is no way to switch to a newer version in PlatformIO before a new release

Jason2866, 2022-09-21:
It needs [this IDF commit](https://github.com/espressif/esp-idf/commit/f2ac141ebfb4cfbc1dfb3f06119f46bf1b3f39cc) or later for building the Arduino libs to get correct S3 temp readings

Jason2866, 2022-12-22:
Imho, this can be closed. Internal temp sensor is measuring the Chip temperature.
More code will just add bit changed temperatur "numbers". For chip temperature it is precise enough as it is now!
If the environment temperature should be measured there is ALWAYS an external sensor needed.

savejeff, 2022-12-22:
jup if the code is includede in the latest release. 
Would be cool to have the ESP.getCPUTemperature() member function, but its okay this way

closed 2022-12-22 by savejeff