espressif/arduino-esp32 #7186: Eeprom update to limit the writing cycle on the flash memory

podaen, 2022-08-26:
### Related area

ESP arduino

### Hardware specification

All

### Is your feature request related to a problem?

No problem here

### Describe the solution you'd like

The flash memory is limited in writing to +1000 cycles, so I would suggest to add EEPROM.update to avoid unnecessary use.

### I have checked existing list of Feature requests and the Contribution Guide

- [X] I confirm I have checked existing list of Feature requests and Contribution Guide.

SuGlider, 2022-08-27:
ESP32 Arduino EEPROM uses NVS for storing the data. NVS already does wear levelling and supports flash encryption.

You can find some information about here: 
https://www.esp32.com/viewtopic.php?f=2&t=3380
https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/storage/nvs_flash.html

If Flash wearing is a huge problem for your Application, you may consider using an external EEPROM chip.
https://stackoverflow.com/questions/70826170/in-esp32-esp-idf-when-to-use-eeprom-vs-nvs-vs-spiffs
https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/storage/wear-levelling.html

podaen, 2022-08-27:
What is this wear levelling? Is it that it checks first and if the data is the same, it doesn't write over it?

SuGlider, 2022-08-27:
Wear leveling is a process that is programmed into flash memory devices that ensures that the blocks of memory in the flash device are used as efficiently as possible. The algorithm allows data to be written evenly across the device, which is essential for maximizing the capacity and longevity of the memory.

SuGlider, 2022-08-27:
https://www.delkin.com/blog/ssd-nand-flash-wear-leveling-and-maximizing-ssd-life/

SuGlider, 2022-08-27:
https://www.arduino.cc/reference/en/libraries/eepromwearlevel/

closed 2022-08-27 by podaen

lbernstone, 2022-08-27:
EEPROM is provided for backward compatibility with previous Arduino applications.  New, esp32 specific code should use the Preferences library instead.

SuGlider, 2022-08-27:
ESP32 Arduino Preferences Lib also uses NVS, very similar to EEPROM Lib.