espressif/arduino-esp32 #7187: esp32-hal-log.h macros not usable by default

tossp, 2022-08-27:
### Board

esp32-s3

### Device Description

devkitc-1

### Hardware Configuration

esphome:
  name: ts3
  platformio_options:
    # board_build.variant: esp32-s3-devkitc-1
    platform: https://ghproxy.com/https://github.com/platformio/platform-espressif32/archive/refs/heads/develop.zip
    framework: espidf
    platform_packages:
      - platformio/framework-arduinoespressif32 @ https://ghproxy.com/https://github.com/espressif/arduino-esp32/archive/refs/heads/idf-release/v4.4.zip
    build_flags: 
      - "-DBOARD_HAS_PSRAM"
      - "-mfix-esp32-psram-cache-issue"
    board_build.f_cpu: 240000000L
    board_upload.flash_size: 16MB
    board_upload.maximum_ram_size: 327680
    board_upload.maximum_size: 16777216
    board_build.partitions: "../../../esp32_default_16MB.txt"

esp32:
  board: esp32-s3-devkitc-1
  variant: esp32s3

### Version

other

### IDE Name

Platformio

### Operating System

haos

### Flash frequency

Default values for configuration files

### PSRAM enabled

yes

### Upload speed

Default values for configuration files

### Description

An error occurred during compilation.

### Sketch

```cpp
esphome:
  name: ts3
  platformio_options:
    board_build.variant: esp32-s3-devkitc-1
    platform: https://github.com/platformio/platform-espressif32/archive/refs/heads/develop.zip
    framework: espidf
    platform_packages:
      - platformio/framework-arduinoespressif32 @ https://github.com/espressif/arduino-esp32/archive/refs/heads/idf-release/v4.4.zip
[... 205 lines cut ...]
In file included from src/esphome/components/api/api_frame_helper.cpp:3:
src/esphome/core/log.h:18:10: fatal error: esp32-hal-log.h: No such file or directory

***********************************************************************
* Looking for esp32-hal-log.h dependency? Check our library registry!
*
* CLI  > platformio lib search "header:esp32-hal-log.h"
* Web  > https://registry.platformio.org/search?q=header:esp32-hal-log.h
*
***********************************************************************

 #include <esp32-hal-log.h>
          ^~~~~~~~~~~~~~~~~
compilation terminated.
*** [/data/ts3/.pioenvs/ts3/src/esphome/components/api/api_frame_helper.o] Error 1
========================= [FAILED] Took 366.39 seconds =========================
```

### Other Steps to Reproduce

I tried to manually import the header file in the process, but still failed to compile

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

closed 2022-08-27 by tossp

SuGlider, 2022-08-27:
@tossp - It seems a problem in PlatformIO setup of Arduino Framework + ESP Home project...
Not sure how to help. Maybe @Jason2866 can give you a hand.

You may also try to ask about it in the PlatformIO forum as well.

Jason2866, 2022-08-27:
This can not work
```
platform_packages:
      - platformio/framework-arduinoespressif32 @ https://github.com/espressif/arduino-esp32/archive/refs/heads/idf-release/v4.4.zip
 ``` 
 since the espressif releases do not contain a package.json You have to use a IDF4.4 version released from PlatformIO