espressif/arduino-esp32 #7196: IDF_VER macro is "v4.4.1-472-gc9140caf8c". Shouldn't it be "v4.4.2"?

moose4lord, 2022-08-29:
### Board

DOIT ESP32 DevKit V1

### Device Description

plain development board

### Hardware Configuration

No

### Version

v2.0.4

### IDE Name

Arduino IDE

### Operating System

macOS 10.15.7

### Flash frequency

80 MHz

### PSRAM enabled

no

### Upload speed

115200

### Description

If I print the IDF_VER macro it displays "v4.4.1-472-gc9140caf8c". Shouldn't it be "v4.4.2"?

### Sketch

```cpp
Serial.print("IDF_VER: "); Serial.println(IDF_VER);
```

### Debug Message

```plain
This is also used as a compilation flag:
"-DIDF_VER=\"v4.4.1-472-gc9140caf8c\""
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

rel1ct, 2022-08-29:
22:19:21.468 -> ESP.getSdkVersion: v4.4.2-1-gce1fde4b3b
22:19:21.468 -> IDF_VER:  v4.4.2-1-gce1fde4b3b

moose4lord, 2022-08-29:
Well that's interesting. I had reinstalled the latest version (v2.0.4) of the ESP32 core for the Arduino IDE on my Mac, and I get  IDF_VER: v4.4.1-472-gc9140caf8c

What is your environment @rel1ct?

VojtechBartoska, 2022-08-30:
Release notes updated: https://github.com/espressif/arduino-esp32/releases/tag/2.0.4

Thanks for the catch @moose4lord! I'm closing this as solved.

closed 2022-08-30 by VojtechBartoska