espressif/arduino-esp32 #7198: 关于eeprom不能使用乐鑫烧录工具的问题

casitelwithdean, 2022-08-30:
### Board

ESP32-WROOM-S3

### Device Description

ESP32-WROOM-S3

### Hardware Configuration

ESP32-WROOM-S3

### Version

v2.0.2

### IDE Name

arduino

### Operating System

win10

### Flash frequency

80M

### PSRAM enabled

yes

### Upload speed

921600

### Description

这是一个非常amazing的问题。
这是第一个的代码
void setup() {
Serial.begin(9600);
}
void loop() {
Serial.println("aa");
delay(100);
}
在arduino上烧录运行正常，然后导出编译bin文件，使用乐鑫烧录工具烧录仍然运行正常。
下面加上eeprom的代码
#include "EEPROM.h"
void setup() {
Serial.begin(9600);
if (!EEPROM.begin(1000)) {
Serial.println("Failed to initialise EEPROM Restarting...");
delay(1000);
ESP.restart();
}
}
void loop() {
Serial.println("aa");
delay(100);
}
这个代码在arduino烧录运行也是十分正常的，但是通过乐鑫烧录工具烧录，就会报错重启，显示eeprom初始化失败。反复试了很多次，只要有eeprom并且通过乐鑫的烧录工具烧录就会重启！！
这个问题请问如何解决呢？

### Sketch

```cpp
Sketch
```

### Debug Message

```plain
Debug Message
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

SuGlider, 2022-08-31:
EEPROM uses NVS Flash partition. Are you sure that you are using the right Flash Tool parameters setup?

casitelwithdean, 2022-08-31:
> EEPROM uses NVS Flash partition. Are you sure that you are using the right Flash Tool parameters setup?

谢谢回复，我使用分区表bin的是\Arduino\hardware\espressif\esp32\tools\partitions\default.bin
我查看了default.csv，里面的nvs有着0x5000的size，但是仍然不行。请问是不是还有其他的nvs要配置呢？

VojtechBartoska, 2022-09-05:
@casitelwithdean can you please translate issue title to English?

closed 2022-09-05 by casitelwithdean