espressif/arduino-esp32 #7203: Enable PPP client support APIs in LwIP for GSM connectivity

avillacis, 2022-08-30:
### Related area

Networking

### Hardware specification

Support for GSM/GPRS modules via serial connection

### Is your feature request related to a problem?

Currently, the Arduino IDE support for ESP32 (as of version 2.0.4) does not enable support for APIs that allow integration of PPPoS connections through GSM/GPRS cellular modules (or any other PPP source/sink), with the lwIP stack available from Espressif:

```sh
$ grep -r PPP_SUPPORT ~/.arduino15/packages/esp32/
...
/home/alex/.arduino15/packages/esp32/hardware/esp32/2.0.4/tools/sdk/esp32/sdkconfig:# CONFIG_LWIP_PPP_SUPPORT is not set
/home/alex/.arduino15/packages/esp32/hardware/esp32/2.0.4/tools/sdk/esp32/sdkconfig:# CONFIG_PPP_SUPPORT is not set
...
/home/alex/.arduino15/packages/esp32/hardware/esp32/2.0.4/tools/sdk/esp32c3/sdkconfig:# CONFIG_LWIP_PPP_SUPPORT is not set
/home/alex/.arduino15/packages/esp32/hardware/esp32/2.0.4/tools/sdk/esp32c3/sdkconfig:# CONFIG_PPP_SUPPORT is not set
...
/home/alex/.arduino15/packages/esp32/hardware/esp32/2.0.4/tools/sdk/esp32s2/sdkconfig:# CONFIG_LWIP_PPP_SUPPORT is not set
/home/alex/.arduino15/packages/esp32/hardware/esp32/2.0.4/tools/sdk/esp32s2/sdkconfig:# CONFIG_PPP_SUPPORT is not set
...
/home/alex/.arduino15/packages/esp32/hardware/esp32/2.0.4/tools/sdk/esp32s3/sdkconfig:# CONFIG_LWIP_PPP_SUPPORT is not set
/home/alex/.arduino15/packages/esp32/hardware/esp32/2.0.4/tools/sdk/esp32s3/sdkconfig:# CONFIG_PPP_SUPPORT is not set
...
```

This forces anyone who wants to integrate PPPoS into the lwIP stack to recompile Arduino-ESP32 on their own. Otherwise errors like these appear (example with https://github.com/vshymanskyy/ESP32_PPPoS):
```
...

/home/alex/Arduino/libraries/ESP32_PPPoS/src/PPPoS.cpp:226:5: error: 'pppapi_connect' was not declared in this scope
     pppapi_connect(ppp, 0);
     ^~~~~~~~~~~~~~
/home/alex/Arduino/libraries/ESP32_PPPoS/src/PPPoS.cpp:226:5: note: suggested alternative: 'lwip_connect'
     pppapi_connect(ppp, 0);
     ^~~~~~~~~~~~~~
     lwip_connect
/home/alex/Arduino/libraries/ESP32_PPPoS/src/PPPoS.cpp: In member function 'void PPPoS::end()':
/home/alex/Arduino/libraries/ESP32_PPPoS/src/PPPoS.cpp:236:18: error: 'ppp' was not declared in this scope
     pppapi_close(ppp, 0);
                  ^~~
/home/alex/Arduino/libraries/ESP32_PPPoS/src/PPPoS.cpp:236:18: note: suggested alternative: 'pipe'
     pppapi_close(ppp, 0);
                  ^~~
                  pipe
/home/alex/Arduino/libraries/ESP32_PPPoS/src/PPPoS.cpp:236:5: error: 'pppapi_close' was not declared in this scope
     pppapi_close(ppp, 0);
     ^~~~~~~~~~~~
/home/alex/Arduino/libraries/ESP32_PPPoS/src/PPPoS.cpp:236:5: note: suggested alternative: 'lwip_close'
     pppapi_close(ppp, 0);
     ^~~~~~~~~~~~
     lwip_close
...
```

### Describe the solution you'd like

Oficially enable relevant PPP support in the networking support for Arduino-ESP32, so that code linking into it can work correctly. No new code appears to be necessary, only enabling conditional compilation of existing source code.

### Describe alternatives you've considered

There are some external libraries, such as [TinyGSM](https://github.com/vshymanskyy/TinyGSM) that allow opening of Arduino-compatible Client objects usable for some libraries. However, if a project is already commited to socket-based networking, it is awkward to duplicate network support through WiFi and through GSM.

### Additional context

This support has already been requested by some other reports, without apparent success: https://github.com/espressif/arduino-esp32/issues/1858 https://github.com/espressif/arduino-esp32/issues/5345

### I have checked existing list of Feature requests and the Contribution Guide

- [X] I confirm I have checked existing list of Feature requests and Contribution Guide.

SuGlider, 2022-08-31:
PPP has been removed from Arduino Core 2.0.0 to 2.0.1, if I remember.
The reason is that it consumes a lot of Flash and users were complaining about the size of the final binary.

My suggestion for solving this specific case would be to use Arduino as Component in an IDF project and then enable PPP in the LWIP menuconfig manually. Another option would be use Arduino Core 2.0.0 which has PPP enabled.

avillacis, 2022-09-01:
Yes, I remember that. I was one of the users who complained about the excessive size of the binary.

I will look into alternate ways to use the code, as suggested.

RafaelLindarte, 2022-09-05:
hey guys, I find it frustrating not being able to implement the ppp_client and resolve it using third-party libraries. TinyGsm is a great library, but there's a simpler solution compatible with native esp APIs. please add support for this feature in platformio.

kshitijgarg2609, 2023-07-17:
Hello, can we have macros to enable pppos or any other method?

closed 2025-01-16 by Parsaabasi