espressif/arduino-esp32 #7211: noInterrupts() not implemented - can not turn off interrupts

caternuson, 2022-09-02:
### Board

ESP32-PICO-MINI-02

### Device Description

Adafruit Feather ESP32 V2
https://www.adafruit.com/product/5400

### Hardware Configuration

No additional hardware attached. However, to run the example sketch to demonstrate the issue, two digital pins need to be connected together somehow. This was done on a breadboard using a jumper wire.

### Version

v2.0.4

### IDE Name

Arduino IDE

### Operating System

Ubuntu

### Flash frequency

n/a

### PSRAM enabled

yes

### Upload speed

921600

### Description

The Arduino methods for disabling/enabling interrupts are no implemented:
https://www.arduino.cc/reference/en/language/functions/interrupts/nointerrupts/
https://www.arduino.cc/reference/en/language/functions/interrupts/interrupts/

Interrupts are "always on".

### Sketch

```cpp
#define OUT_PIN 14
#define IN_PIN 32

int count = 0;

void counter() {
  count++;
}

void setup() {
  Serial.begin(9600);

  pinMode(OUT_PIN, OUTPUT);
  digitalWrite(OUT_PIN, LOW);
  pinMode(IN_PIN, INPUT);

  attachInterrupt(digitalPinToInterrupt(IN_PIN), counter, RISING);

  for (int i=0; i<50; i++) {
    digitalWrite(OUT_PIN, HIGH);
    digitalWrite(OUT_PIN, LOW);
  }

  noInterrupts();

  for (int i=0; i<50; i++) {
    digitalWrite(OUT_PIN, HIGH);
    digitalWrite(OUT_PIN, LOW);
  }

  Serial.print("count = "); Serial.println(count);
}

void loop() {
}
```

### Debug Message

```plain
No debug messages. Sketch runs without issue. However, the expected output is 50, not 100.

count = 100
```
```

### Other Steps to Reproduce

This is the same issue reported in this older thread:
https://github.com/espressif/arduino-esp32/issues/832
that was closed claiming it's not a BSP issue.

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

SuGlider, 2022-09-03:
@caternuson - thanks for reporting this issue.

I see there is some simple code for it, but it may not be working for all cases... 
https://github.com/espressif/arduino-esp32/blob/master/cores/esp32/Arduino.h#L84-L85

I'll test it further and try to find a fix for it.

SuGlider, 2022-09-05:
`portDISABLE_INTERRUPTS();` and `portENABLE_INTERRUPTS();` are the equivalent to `noInterrupts()` and `interrupts()`

ESP32 runs FreeRTOS, thus, the sketch must be VERY careful using `noInterrups()` ...
Disabling Interrupts will lead to potential problems for most Arduino sketches such as WDT panic.

The example of this issue results in WDT Panic with a Guru Mediation  Error.

```
count = 50
Guru Meditation Error: Core  1 panic'ed (Interrupt wdt timeout on CPU1).
```

This code does what you want:
``` cpp
#define OUT_PIN 14
#define IN_PIN 32

int count = 0;

void counter() {
  count++;
}

void setup() {
  Serial.begin(115200);

  pinMode(OUT_PIN, OUTPUT);
  digitalWrite(OUT_PIN, LOW);
  pinMode(IN_PIN, INPUT);

  attachInterrupt(digitalPinToInterrupt(IN_PIN), counter, RISING);

  for (int i=0; i<50; i++) {
    digitalWrite(OUT_PIN, HIGH);
    digitalWrite(OUT_PIN, LOW);
  }

  noInterrupts();
  portDISABLE_INTERRUPTS();

  for (int i=0; i<50; i++) {
    digitalWrite(OUT_PIN, HIGH);
    digitalWrite(OUT_PIN, LOW);
  }

  Serial.print("count = "); Serial.println(count);

  interrupts();
  portENABLE_INTERRUPTS();
}

void loop() {
}
```