espressif/arduino-esp32 #7216: Library logs doesn't work on board selection "ESP32 dev module"

podaen, 2022-09-03:
### Board

ESP32 dev module

### Device Description

ESP32 wrover devkit

### Hardware Configuration

No

### Version

v2.0.4

### IDE Name

Arduino IDE

### Operating System

Windows 7

### Flash frequency

40MHz

### PSRAM enabled

yes

### Upload speed

115200

### Description

The log of the my shetch and hal files work with the "ESP32 dev module" board selected in the menu. But after an update I don't see to get any logs for example WiFi library and also external libraries for that specific board. For board "MH ET live ESP32 minikit" it still works.

### Sketch

```cpp
no
```

### Debug Message

```plain
?
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

SuGlider, 2022-09-03:
>The log of the my shetch and hal files work with the "ESP32 dev module" board selected in the menu. But after an update I don't see to get any logs for example WiFi library and also external libraries for that specific board. For board "MH ET live ESP32 minikit" it still works.

Are you using the Arduino IDE?
In affirmative case, is the `IDE->Menu->Tools->Core Debug Level` set to `Verbose` or some other desired option?

podaen, 2022-09-03:
I m using it with an extension on top... Visual Micro in the visual studio environment 2017-19. I tried to use the Clang compiler but it diden't make any difference.

I don't get any new updates and still hanging o  the c17++.

SuGlider, 2022-09-03:
I see, so you are not using the Arduino IDE, right?

It may be necessary to set manually the `ARDUHAL_LOG_LEVEL` by changing `CONFIG_ARDUHAL_LOG_DEFAULT_LEVEL` in the `sdkconfig.h` of your Arduino installation and board configuration.

You can check it in the `esp32-hal-log.h` file. You may, optionally, want to just change it manually there.

podaen, 2022-09-03:
The Arduino IDE is 1.18

SuGlider, 2022-09-03:
> The Arduino IDE is 1.18

1.8.18?
Thus, you can set the log level in the IDE menu.

podaen, 2022-09-03:
> > The Arduino IDE is 1.18
> 
> 1.8.18? Thus, you can set the log level in the IDE menu.

Yeah, but it doesn't work. I will try it do it manualy.

SuGlider, 2022-09-03:
>I did set that level to diffrent options, but no positive outcomes.

It was working fine before you changed the board in the IDE board manager?
Please give me more details about the changes that leaded to the issue.

podaen, 2022-09-03:
> > I did set that level to diffrent options, but no positive outcomes.
> 
> It was working fine before you changed the board in the IDE board manager? Please give me more details about the changes that leaded to the issue.

Actually it was invers. I first tried the "ESP32 dev module" board and than selected the "mini" that worked.

SuGlider, 2022-09-03:
I just tested it using: Arduino Core 2.0.4 | ESP32  Dev Module | WiFiScan.ino example | "Core Debug Level: Verbose"
I got all the debug messages out to the UART0 (Serial) port.

Sometime, when we install or update the Arduino Core Version in the IDE Menu, the IDE, for some reason, leaves the board options empty. To fix it, I use to select any other board in the IDE Menu, and then select back the board I was using in the IDE Menu. 

Let me know if everything is working fine at your end.

podaen, 2022-09-03:
> I just tested it using: Arduino Core 2.0.4 | ESP32 Dev Module | WiFiScan.ino example | "Core Debug Level: Verbose" I got all the debug messages out to the UART0 (Serial) port.

Itresting thought, I have to think about that what I can do more.

podaen, 2022-09-04:
I looked for the file sdkconfig file and I found multiple files that name. Only for the ESP32 I have 3 files called that name.

[image]

In the Visual Micro menu for the "ESP32 Dev Module" I have an option FLASH MODE which I don't have if I select the "MH ET live ESP32 minikit". I can select the following modes, but I don't know what the mean and what I need to select...

> FLASH MODE
> ---------------
> QIO
> DIO
> QOUT
> DOUT

[image]

I did change it before to an other level and back, because only than the settings are reload. But it didn't help this time.

podaen, 2022-09-04:
I did a lookup [here](https://docs.espressif.com/projects/esptool/en/latest/esp32/advanced-topics/spi-flash-modes.html) and It seems that the flash mode only is a difference in speed.

> qio | Quad I/O | 4 pins used for address & data | Fastest.

These are for the QIO in the sdkconfig

> #define CONFIG_ARDUHAL_LOG_DEFAULT_LEVEL_ERROR 1
> #define CONFIG_ARDUHAL_LOG_DEFAULT_LEVEL 1
> #define CONFIG_ARDUHAL_ESP_LOG 1

I set

> #define CONFIG_ARDUHAL_LOG_DEFAULT_LEVEL 5

in

> G:\Documents\Visual Studio 2019\Arduino\hardware\espressif\esp32\tools\sdk\esp32\qio_qspi\include\sdkconfig.h

And than set in the visual micro menu  Core debug level to None, but I don't have any log so far after setting the default to 5.

podaen, 2022-09-04:
> It was working fine before you changed the board in the IDE board manager?
> Please give me more details about the changes that leaded to the issue.

The debug level was working before I toke the latest ESP Arduino, Visual Studio and Visual Micro.

> I just tested it using: Arduino Core 2.0.4 | ESP32 Dev Module | WiFiScan.ino example | "Core Debug Level: Verbose"
> I got all the debug messages out to the UART0 (Serial) port.

@SuGlider I think something is wrong about my configuration of that project I made in a previous version. When I start a new with the WiFiScan example I get this log.

> Opening port
> Port open
> [   450][I][esp32-hal-psram.c:96] psramInit(): PSRAM enabled
> VMDPR_[  4523][D][WiFiGeneric.cpp:929] _eventCallback(): Arduino Event: 0 - WIFI_READY
> [  4610][V][WiFiGeneric.cpp:338] _arduino_event_cb(): STA Started
> [  4610][D][WiFiGeneric.cpp:929] _eventCallback(): Arduino Event: 2 - STA_START
> Setup done
> scan start
> [ 10916][V][WiFiGeneric.cpp:381] _arduino_event_cb(): SCAN Done: ID: 128, Status: 0, Results: 5
> [ 10916][D][WiFiGeneric.cpp:929] _eventCallback(): Arduino Event: 1 - SCAN_DONE
> scan done
> 5 networks found
> 1: PODAEN (-59)*

When I do the test with my old project and the core debug level to verbose I get them too, but not the external library logs. I m looking if I did enough to enable them.

podaen, 2022-09-04:
The WiFi secure was not in use and there for didn't get any messages of that. The external library was using a callback to display the log. I forgot that. Now it's working and even crashes anymore.

> > It was working fine before you changed the board in the IDE board manager?
> > Please give me more details about the changes that leaded to the issue.
> 
> The debug level was working before I toke the latest ESP Arduino, Visual Studio and Visual Micro.

I was on the wrong pad in searching what the issue was. It had nothing to do by updates I have made in using the latest ESP Arduino, Visual Studio and Visual Micro. 

Trying the example WiFiScan helped me a lot. And maybe changing the compiler in Visual Studio->Project->Properties->Platfrom to Clang did help. I don't know. But that was the only thing I did change as far as I remember. Everything is working now. Thanks for helping me out Su!

closed 2022-09-04 by podaen