espressif/arduino-esp32 #7224: Average BLE Indicate response time 3400%(1020mSec) more than Notification(30mSec)!

SuperTankMan, 2022-09-05:
### Board

ESP32 Devboard

### Device Description

Standard ESP32 development board

### Hardware Configuration

Using Arduino BLE nothing else required.

### Version

v2.0.2

### IDE Name

ArduinoIDE and PlatformIO

### Operating System

Windows 8.1

### Flash frequency

240MHz

### PSRAM enabled

no

### Upload speed

921600

### Description

**The time taken for receiving an Indication to a write to BLE characteristic is so so much longer than a notify.** On notify taking 30milli Seconds as opposed to **1020milli Seconds for Indication.** While I understand Indicate requires Acknowledgment and hence a little longer than Notification it seams Indication is taking far too long and causing issue in our project. 
In the attached sketch I have tried to simplify the code as much as I can. It consists of two separate code for **Server on one board** and a **Client on another board.**

The Client code basically starts a connection to a server then after registering for notification writes to a BLE characteristic and take a time stamp then waits 2500milli second before it repeats the test. While waiting if a Notification or Indication when the Server Notifies or Indicate to the Client the notifyCallback(*) on the Client prints the details and restart the time stamp after printing the return time.

On the Server as soon as a callback (onWrite(*)) to either notification or indication is called the code in the callback increments a local value and either notifies or indicates back to the Client depending.

**I am not sure if extra code is required for Indication on both Client and Server hence it is not there?**

In both Client and Server code you can select by **uncommenting either NOTIFICATION_ON or INDICATION_ON.**

Link to source code and log files: (https://github.com/SuperTankMan/Issue-code/tree/main/BLEIndicateIssue)

### Sketch

```cpp
/*
    Based on Neil Kolban example for IDF: https://github.com/nkolban/esp32-snippets/blob/master/cpp_utils/tests/BLE%20Tests/SampleServer.cpp
    Ported to Arduino ESP32 by Evandro Copercini
    updates by chegewara
*/

#include <BLEDevice.h>
#include <BLEUtils.h>
#include <BLEServer.h>
#include <BLE2902.h>

// See the following for generating UUIDs:
// https://www.uuidgenerator.net/

#define SERVICE_UUID        "4fafc201-1fb5-459e-8fcc-c5c9c331914b"
#define CHARACTERISTIC_UUID "beb5483e-36e1-4688-b7f5-ea07361b26a8"
[... 580 lines cut ...]
This took =1038
In loop......
Setting new characteristic value to "Time since boot: 46"
Sending after  =1464
47489:Notify callback for characteristic beb5483e-36e1-4688-b7f5-ea07361b26a8 of data length 4
data: 19
This took =1026
In loop......
Setting new characteristic value to "Time since boot: 48"
Sending after  =1474
49994:Notify callback for characteristic beb5483e-36e1-4688-b7f5-ea07361b26a8 of data length 4
data: 20
This took =1031
In loop......
Setting new characteristic value to "Time since boot: 51"
Sending after  =1470
```

### Other Steps to Reproduce

Load code in Server and in Client on separate development boards then Reset both at the same time, Client will connect to server and start the testing and timing for response to writing to a BLE characteristic.

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

SuGlider, 2022-09-06:
@SuperTankMan 
> On the Server as soon as a callback (onWrite(*)) to either notification or indication is called the code in the callback increments a local value and either notifies or indicates back to the Client [...]

The problem seems to be in the sketch.
`indicate()` shall never be called from a BLE callback because it will always timeout waiting for the Client Indication ACK.

Use this code in your Server sketch, it works as desired:

``` cpp
/*
    Based on Neil Kolban example for IDF: https://github.com/nkolban/esp32-snippets/blob/master/cpp_utils/tests/BLE%20Tests/SampleServer.cpp
    Ported to Arduino ESP32 by Evandro Copercini
    updates by chegewara
*/

#include <BLEDevice.h>
#include <BLEUtils.h>
#include <BLEServer.h>
#include <BLE2902.h>

// See the following for generating UUIDs:
// https://www.uuidgenerator.net/

#define SERVICE_UUID        "4fafc201-1fb5-459e-8fcc-c5c9c331914b"
#define CHARACTERISTIC_UUID "beb5483e-36e1-4688-b7f5-ea07361b26a8"

//Uncomment one
//#define NOTIFY_ON
#define INDICATE_ON

int count = 0;
bool shallNotify_Indicate = false;

class CharCallbacks : public BLECharacteristicCallbacks
{
  public:
    void onWrite(BLECharacteristic *pChar)
    {
      pChar->setValue(++count);
      shallNotify_Indicate = true;
      
      //      delay(5000);
      //      pChar->setValue(++count);
      //      pChar->indicate();
#if 0
      pChar->setValue(++count);
      pChar->indicate();
      pChar->setValue(++count);
      pChar->indicate();
#endif
    }
};

class CCCDCallbacks : public BLEDescriptorCallbacks
{
  public:
    void onWrite(BLEDescriptor *pDesc)
    {
      Serial.printf("[%ld]:Notification registered\n", millis());
    }
};

BLECharacteristic *pCharacteristic;

BLEDescriptor *myBLECCCD = new BLE2902();

void setup() {
  Serial.begin(115200);
  Serial.println("Starting BLE work!");

  BLEDevice::init("Indicate issue Server");
  BLEServer *pServer = BLEDevice::createServer();
  BLEService *pService = pServer->createService(SERVICE_UUID);
  pCharacteristic = pService->createCharacteristic(
                      CHARACTERISTIC_UUID,
                      BLECharacteristic::PROPERTY_READ |
                      BLECharacteristic::PROPERTY_WRITE |
#ifdef NOTIFY_ON
                      BLECharacteristic::PROPERTY_NOTIFY //|
#endif
#ifdef INDICATE_ON
                      BLECharacteristic::PROPERTY_INDICATE
#endif
                    );

  pCharacteristic->setCallbacks(new CharCallbacks);

  myBLECCCD->setCallbacks(new CCCDCallbacks());
  pCharacteristic->addDescriptor(myBLECCCD);  //Add this line only if the characteristic has the Notify property

  pCharacteristic->setValue(count);
  pService->start();
  // BLEAdvertising *pAdvertising = pServer->getAdvertising();  // this still is working for backward compatibility
  BLEAdvertising *pAdvertising = BLEDevice::getAdvertising();
  pAdvertising->addServiceUUID(SERVICE_UUID);
  pAdvertising->setScanResponse(true);
  pAdvertising->setMinPreferred(0x06);  // functions that help with iPhone connections issue
  pAdvertising->setMinPreferred(0x12);
  BLEDevice::startAdvertising();
  Serial.println("Characteristic defined! Now you can read it in your phone!");
}

void loop()
{
#if 0
  // put your main code here, to run repeatedly:
  if (((BLE2902*)myBLECCCD)->getNotifications())
    if (((BLE2902*)myBLECCCD)->getIndications())
    {
      //    pCharacteristic->setValue(++count);
      //    pCharacteristic->notify();
      //    pCharacteristic->indicate();
    }
#endif
  //  delay(50);
      if (shallNotify_Indicate) {
        shallNotify_Indicate = false;
#ifdef NOTIFY_ON
        pCharacteristic->notify();
#endif
#ifdef INDICATE_ON
        pCharacteristic->indicate();
#endif
      }
}
```

SuGlider, 2022-09-06:
I just added a flag `shallNotify_Indicate` that is set in the `onWrite()` callback and then processed in `loop()`, sending the actual Indicate signal.

SuperTankMan, 2022-09-07:
Hi Rodrigo,

Thanks for you feedback.

It seam that **notify()** within **onWrite()** in the **callback** works fine and it seam to be more elagant. Is there anyway this can still be done within the **onWrite()** function like **notify()** or is notifying within the **onWrite()** callback  also not correct place to call it?

SuGlider, 2022-09-07:
`notify()` can be called inside `onWrite()`.
The problem is only with `indicate()`.

This happens because the same Event Loop that process the `Indicate ACK` also process `onWrite()`.

Thus, when the user `onWrite()` callback  calls `inicate()`, it gets blocked waiting for the `ACK ` or it gets timed out (about 1 second). 

`indicate ACK` will be processed after the timeout and that's the reason for such time that you have seen.

SuperTankMan, 2022-09-08:
Thank you very much, Rodrigo, for all your help.

closed 2022-09-08 by SuperTankMan