espressif/arduino-esp32 #7227: WiFi Hostname is only set when changing mode to WIFI_MODE_STA

tbnobody, 2022-09-06:
### Board

generic

### Device Description

-

### Hardware Configuration

No

### Version

v2.0.4

### IDE Name

VSCode with PlatformIO

### Operating System

Windows 11

### Flash frequency

-

### PSRAM enabled

no

### Upload speed

115200

### Description

There are several issues in issue trackers out there in the internet which are stating that the hostname is not set correctly. When diving deeper into the library, it seems that the hostname is only set on one location: https://github.com/espressif/arduino-esp32/blob/77065bffb2c57704ee46ad0a43270e50860814f5/libraries/WiFi/src/WiFiGeneric.cpp#L1122

That means set a call of `WiFi.setHostname()` does nothing in the first step. It's not even enough to call `WiFi.mode(WIFI_MODE_STA);` if the mode was already WIFI_MODE_STA. Because if current mode equals new mode: https://github.com/espressif/arduino-esp32/blob/77065bffb2c57704ee46ad0a43270e50860814f5/libraries/WiFi/src/WiFiGeneric.cpp#L1109
no changes are applied.

To set the hostname correctly the following snippet has to be used:
```
WiFi.mode(WIFI_MODE_APSTA); // or any other mode
WiFi.setHostname("asdf");
WiFi.mode(WIFI_MODE_STA);
```

If the behavior is known to the user of the class it's  simple to use but it's a little bit unintuitive. Does it make sense to at least document this behavior?

### Sketch

```cpp
-
```

### Debug Message

```plain
-
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

domdfcoding, 2022-12-13:
I've run into the same issue of the hostname being set, with internet unhelpfully guides suggesting to do it they way I was already. 
I've tried the approach listed here (with the exception of `WIFI_STA` instead of `WIFI_MODE_STA`, not sure what the difference is) and it works as I would expect. 

Could this be fixed, or at least the limitation documented, please?

I'm running on an ESP32-C3, if it's relevant.

pedrominatel, 2022-12-14:
Hi, @domdfcoding,

This will be added to the documentation. On the code side, I'll discuss with the team if there is something to do.

closed 2025-01-16 by Parsaabasi