espressif/arduino-esp32 #7228: USBMSC reports repeated "USB UNPLUGGED" and can cause stack overflow

savejeff, 2022-09-06:
### Board

ESP32-S3 DevKitC-1

### Device Description

Plane ESP32-S3 DevKitC-1

### Hardware Configuration

no peripheral attached

### Version

v2.0.4

### IDE Name

Arduini IDE

### Operating System

Windows 10

### Flash frequency

80Mhz

### PSRAM enabled

yes

### Upload speed

921600

### Description

There seems to be a bug in the USB MSC implementation. When the native USB Port is not connected to a USB Host (aka a Laptop/PC) it repeatedly fires the USB UNPLUGGED event. as the event queue is only 5 long, this can cause the code to crash with stack canary when there is enough other code running.

This can be easily reproduced by powering an ESP32-S3 (for example over the UART USB port on the devkit and not connecting or disconnecting the "USB" USB Port.

I found a mention of a similar problem with the ESP32-S2 [here](https://bytemeta.vip/repo/hathach/tinyusb/issues/668) and [here](https://github.com/hathach/tinyusb/issues/564)

### Sketch

```cpp
unchanged USBMSC.ino example:

https://github.com/espressif/arduino-esp32/blob/master/libraries/USB/examples/USBMSC/USBMSC.ino
```

### Debug Message

```plain
USB UNPLUGGED <- this one is fired immediacy
USB UNPLUGGED <- after a second the repeated event start
USB UNPLUGGED
USB UNPLUGGED
USB UNPLUGGED
USB UNPLUGGED
USB UNPLUGGED
USB UNPLUGGED
USB UNPLUGGED
USB UNPLUGGED
USB UNPLUGGED
USB UNPLUGGED
USB UNPLUGGED
USB UNPLUGGED
USB UNPLUGGED
USB UNPLUGGED
USB UNPLUGGED
USB UNPLUGGED
USB UNPLUGGED
USB UNPLUGGED
USB UNPLUGGED
USB UNPLUGGED
USB UNPLUGGED
USB UNPLUGGED
USB UNPLUGGED
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

savejeff, 2022-09-07:
@hathach Could you please look at this issue. I've seen you have investigated a similar issue before. Do you have an idea what could cause the USB UNPLUGGED event to be fired so rapidly? 
Help would be very appreciated!

VojtechBartoska, 2024-10-23:
@txf- on which Arduino version?

txf-, 2024-10-23:
> @txf- on which Arduino version?

Sorry. It is version 2.0.9

savejeff, 2024-10-23:
Since then I have "resolved"/avoided this by rebooting after the first UNPLUG event. 
I have not tested this on newer versions but it would be cool to remove the need of a reboot

VojtechBartoska, 2024-10-24:
> > @txf- on which Arduino version?
> 
> Sorry. It is version 2.0.9

Can you retest please on latest 2.X version which is https://github.com/espressif/arduino-esp32/releases/tag/2.0.17? Thanks

savejeff, 2024-11-01:
I can confirm that this problem still persists
i tested it with PlatformIO and this Version: 
 - framework-arduinoespressif32 @ 3.20017.0 (2.0.17) 
 
 I also tested it with Arduino IDE with Core Version 2.0.18. 
 
 you can test this easily with the Example USB->USBSerial. When unplugging the USB Port "USB UBPLUG" is spammed.
An interesting Fact is, it seems to be influenced by if the Connector is Grounded. The Rate reduces and sometimes stops when the connector touches my laptop housing. This might be a Problem with a floating Signal Line.

dhebbeker, 2025-02-23:
This problem still persists (testing with framework-arduinoespressif32@3.20017.241212+sha.dcc1105b). 

I can also confirm @savejeff's observation about grounding: When the shield of the USB is grounded, the triggers are reduced and sometime stop (testing with [ESP32-S3-DevKitC-1 v1.0](https://docs.espressif.com/projects/esp-dev-kits/en/latest/esp32s3/esp32-s3-devkitc-1/user_guide_v1.0.html#esp32-s3-devkitc-1)).

I also tried @chegewara's workaround (https://github.com/hathach/tinyusb/issues/668#issuecomment-783080207) by calling `USBMSC::mediaPresent(false);` when `ARDUINO_USB_STOPPED_EVENT` event is received by the callback for `ARDUINO_USB_EVENTS`. But it did not help.