espressif/arduino-esp32 #7231: WificlientSecure library

asvith, 2022-09-07:
### Related area

for TLS old wificlientsecure library is supporting but not supporting for esp-mail-client library 

### Hardware specification

esp32s2devkit

### Is your feature request related to a problem?

esp 2.0.2 library wificlientsecure is not supporting for TLS, i changed old wificlientsecure then TLS is working fine but esp mail client library was getting error. kindly support to overcome this issue. Please suggest which library should be used for supporting both TLS and Esp-mail-client .

### Describe the solution you'd like

Please suggest which library should be used for supporting both TLS and Esp-mail-client .

### I have checked existing list of Feature requests and the Contribution Guide

- [X] I confirm I have checked existing list of Feature requests and Contribution Guide.

Raghu41189, 2022-09-20:
I too faced same issue :(

mobizt, 2022-09-21:
Do you refer to this [ESP-Mail-Client library](https://github.com/mobizt/ESP-Mail-Client)? 

The library [ESP-Mail-Client library](https://github.com/mobizt/ESP-Mail-Client) has built-in TCP clients and it's not relied on the core SSL client.

If you use WiFiClientSecure as external Client, please make sure you understand the port and its protocol. 

You can't use SSL client to connect to SMTP ports 25 and 587 and IMAP port 143.  You can use SSL client for SMTP port 465 and IMAP port 993.

closed 2025-01-16 by Parsaabasi