espressif/arduino-esp32 #7232: ESP32 C3 softap fails to start with no error condition

tablatronix, 2022-09-07:
### Board

esp32 c3 devkit

### Device Description

espressif devkit tested
G3v01 tested
Possible also an ESP-Sx issue

### Hardware Configuration

nothing special

### Version

v2.0.4

### IDE Name

pio

### Operating System

macos

### Flash frequency

40

### PSRAM enabled

no

### Upload speed

921600

### Description

For some reason if you toggle sta on and off, softap fails to start an AP, but everything in code appears to start successfully..

I imagine there is some event missing in arduino or a state in IDF that is not returning properly.

I am not sure exactly how many times once or twice toggling sta back or the sta state when begin is called..

### Sketch

```cpp
++
// wifi_basic.ino

#include <arduino.h>
#include <WiFi.h>
#include <esp_wifi.h>  

void setup(){
	Serial.begin(115200);
	delay(2000);
	Serial.println("Startup....");
	Serial.setDebugOutput(true);

	/**
	 * This seems to trigger this issue
	 */
	WiFi.enableSTA(true);
	delay(2000);
	// WiFi.disconnect();
	// delay(2000);
	WiFi.enableSTA(false);
	delay(2000);
	// END
	
	WiFi.begin();
	
    while (WiFi.status() != WL_CONNECTED && millis()<10000) {
        delay(500);
        Serial.print(".");
    }

	if(WiFi.status() == WL_CONNECTED){
		Serial.println("");
		Serial.println("WiFi connected.");
		Serial.println("IP address: ");
		Serial.println(WiFi.localIP());
	}
	else {
		Serial.println("WiFi NOT CONNECTED, starting ap");
		delay(2000);
		WiFi.disconnect(); 
		delay(2000);
		WiFi.enableSTA(false);
		delay(2000);		

	    WiFi.softAP("espsoftap","12345678"); // NO AP ACTUALLY FOUND
	}
}

void loop(){

}
```

### Debug Message

```plain
Startup....
[  2160][D][WiFiGeneric.cpp:929] _eventCallback(): Arduino Event: 0 - WIFI_READY
[  2191][V][WiFiGeneric.cpp:338] _arduino_event_cb(): STA Started
[  2191][D][WiFiGeneric.cpp:929] _eventCallback(): Arduino Event: 2 - STA_START
[  4192][V][WiFiGeneric.cpp:341] _arduino_event_cb(): STA Stopped
[  4193][D][WiFiGeneric.cpp:929] _eventCallback(): Arduino Event: 3 - STA_STOP
[  6200][D][WiFiGeneric.cpp:929] _eventCallback(): Arduino Event: 0 - WIFI_READY
[  6207][V][WiFiGeneric.cpp:338] _arduino_event_cb(): STA Started
[  6208][D][WiFiGeneric.cpp:929] _eventCallback(): Arduino Event: 2 - STA_START
[  6209][V][WiFiGeneric.cpp:97] set_esp_interface_ip(): Configuring Station static IP: 0.0.0.0, MASK: 0.0.0.0, GW: 0.0.0.0
[  6212][E][WiFiSTA.cpp:317] begin(): connect failed! 0x300a
........WiFi NOT CONNECTED, starting ap
[ 14212][V][WiFiGeneric.cpp:341] _arduino_event_cb(): STA Stopped
[ 14213][D][WiFiGeneric.cpp:929] _eventCallback(): Arduino Event: 3 - STA_STOP
[ 16220][D][WiFiGeneric.cpp:929] _eventCallback(): Arduino Event: 0 - WIFI_READY
[ 16231][V][WiFiGeneric.cpp:390] _arduino_event_cb(): AP Started
[ 16232][D][WiFiGeneric.cpp:929] _eventCallback(): Arduino Event: 10 - AP_START
```

### Other Steps to Reproduce

Cannot test in 2.0.3 due to bootloader crashes

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

cniedzi, 2022-09-07:
AP doesn't fire up on my S2.

tablatronix, 2022-09-07:
Minimalized test example even more

cniedzi, 2022-09-07:
> ```c++
> // wifi_basic.ino
> 
> #include <arduino.h>
> #include <WiFi.h>
> #include <esp_wifi.h>  
> 
> void setup(){
> 	Serial.begin(115200);
> 	delay(2000);
> 	Serial.println("Startup....");
> 	Serial.setDebugOutput(true);
> 
> 	WiFi.begin();
> 	
>     while (WiFi.status() != WL_CONNECTED && millis()<10000) {
>         delay(500);
>         Serial.print(".");
>     }
> 
> 	if(WiFi.status() == WL_CONNECTED){
> 		Serial.println("");
> 		Serial.println("WiFi connected.");
> 		Serial.println("IP address: ");
> 		Serial.println(WiFi.localIP());
> 	}
> 	else {
> 		Serial.println("WiFi NOT CONNECTED, starting ap");
> 		WiFi.enableSTA(false); // BREAKING
> 		delay(2000);		
> 	    WiFi.softAP("espsoftap","12345678");
> 	}
> }
> 
> void loop(){
> 
> }
> ```

I had to increase the delay to 5s and decrease millis()<5000 because my S2 was always successfuly connecting to WiFi. After that no AP started...

Miceuz, 2022-09-07:
I can't get AP on ESP32-C3

tablatronix, 2022-09-07:
lol @cniedzi erase your credentials or put junk in there

tablatronix, 2022-09-07:
Confirmed same issue on S2, I might escalate this to espressif

tablatronix, 2022-09-10:
I have been trying to fix this and nothing I do fixes it, no workaround yet other that the one mentioned before
`    bool          _disableSTAConn         = true;  // disable sta when starting ap, if sta is not connected ( stability )`
seems to help avoid whatever gets stuck in the IDF

cniedzi, 2022-09-10:
> I have been trying to fix this and nothing I do fixes it, no workaround yet other that the one mentioned before ` bool _disableSTAConn = false; // disable sta when starting ap, if sta is not connected ( stability )` seems to help avoid whatever gets stuck in the IDF

Did you report this issue to Espressif? Did they say something?

tablatronix, 2022-09-10:
I will post this to IDF as an issue, as I do not think it is an arduino issue

tablatronix, 2022-09-12:
UPDATE: added a workaround to the test sketch, no idea whats going on

Jason2866, 2022-09-15:
@tablatronix Try this. I get a working AP with your test sketch (without the workaround)
```
[env:esp32-c3-devkitm-1]
platform                = https://github.com/jason2866/platform-espressif32.git#Tasmota/205
platform_packages       = framework-arduinoespressif32 @ https://github.com/Jason2866/esp32-arduino-lib-builder/releases/download/996/framework-arduinoespressif32-IDF_Arduino-cd58a68f7.zip
board                   = esp32-c3-devkitm-1
framework               = arduino
```

me-no-dev, 2022-09-15:
Adding `WiFi.setAutoReconnect(false);` before `WiFi.begin();` also helps. I think the problem is caused by the constant attempts to reconnect, which interfere with the SoftAP and the software is not able to handle everything. With the original sketch on S3 I could see the AP, but could rarely be able to connect to it, or get IP. As soon as I made sure that before starting AP the station is no longer trying to connect, everything started working normally. This is with ESP-IDF 4.4.2. I will later check with te latest 4.4 branch.

Here is the log:
```
Startup....
[  2180][D][WiFiGeneric.cpp:929] _eventCallback(): Arduino Event: 0 - WIFI_READY
[  2216][V][WiFiGeneric.cpp:338] _arduino_event_cb(): STA Started
[  2217][V][WiFiGeneric.cpp:97] set_esp_interface_ip(): Configuring Station static IP: 0.0.0.0, MASK: 0.0.0.0, GW: 0.0.0.0
[  2218][D][WiFiGeneric.cpp:929] _eventCallback(): Arduino Event: 2 - STA_START
....[  4639][V][WiFiGeneric.cpp:360] _arduino_event_cb(): STA Disconnected: SSID: hellowifi, BSSID: 00:00:00:00:00:00, Reason: 201
[  4640][D][WiFiGeneric.cpp:929] _eventCallback(): Arduino Event: 5 - STA_DISCONNECTED
[  4647][W][WiFiGeneric.cpp:950] _eventCallback(): Reason: 201 - NO_AP_FOUND
[  4654][D][WiFiGeneric.cpp:966] _eventCallback(): WiFi Reconnect Running
[  4662][V][WiFiGeneric.cpp:97] set_esp_interface_ip(): Configuring Station static IP: 0.0.0.0, MASK: 0.0.0.0, GW: 0.0.0.0
.....[  7081][V][WiFiGeneric.cpp:360] _arduino_event_cb(): STA Disconnected: SSID: hellowifi, BSSID: 00:00:00:00:00:00, Reason: 201
[  7082][D][WiFiGeneric.cpp:929] _eventCallback(): Arduino Event: 5 - STA_DISCONNECTED
[  7089][W][WiFiGeneric.cpp:950] _eventCallback(): Reason: 201 - NO_AP_FOUND
.......WiFi NOT CONNECTED, starting ap
[ 12237][V][WiFiGeneric.cpp:390] _arduino_event_cb(): AP Started
[ 12238][D][WiFiGeneric.cpp:929] _eventCallback(): Arduino Event: 10 - AP_START
[ 41170][V][WiFiGeneric.cpp:405] _arduino_event_cb(): AP Station Connected: MAC: f4:d4:88:61:42:78, AID: 1
[ 41171][D][WiFiGeneric.cpp:929] _eventCallback(): Arduino Event: 12 - AP_STACONNECTED
[ 42313][V][WiFiGeneric.cpp:419] _arduino_event_cb(): AP Station IP Assigned:192.168.4.2
[ 42314][D][WiFiGeneric.cpp:929] _eventCallback(): Arduino Event: 14 - AP_STAIPASSIGNED
[ 51607][V][WiFiGeneric.cpp:412] _arduino_event_cb(): AP Station Disconnected: MAC: f4:d4:88:61:42:78, AID: 1
[ 51608][D][WiFiGeneric.cpp:929] _eventCallback(): Arduino Event: 13 - AP_STADISCONNECTED
```

tablatronix, 2022-09-15:
@me-no-dev that is a different issue, `AP instability with STA in disconnected mode` I have that also if I leave mode as STA+AP.

This is specifically NO softap at all, untested on S3 ( i dont have one)

me-no-dev, 2022-09-16:
Just tested on C3 with what will be in 2.0.5 and here is your sketch with all options to make AP work commented and explained. While STA is trying to reconnect, it switches channels while scanning and that causes AP to lose communication (it works only on one channel).

```cpp
#include <WiFi.h>
#include <esp_wifi.h>  

void setup(){
  Serial.begin(115200);
  Serial.setDebugOutput(true);
  Serial.setDebugOutput(true);
  delay(2000);
  Serial.println("Startup....");

  //WiFi.setAutoReconnect(false); // < ------  WORKS BUT YOU SHOULD KNOW THE IMPLICATIONS
  WiFi.begin("hellowifi","noonehere");
  
  while (WiFi.status() != WL_CONNECTED && millis()<10000) {
    delay(500);
    Serial.print(".");
  }

  if(WiFi.status() == WL_CONNECTED){
    Serial.println("");
    Serial.println("WiFi connected.");
    Serial.println("IP address: ");
    Serial.println(WiFi.localIP());
  }
  else {
    Serial.println("WiFi NOT CONNECTED, starting ap");
    //WiFi.disconnect();      // < ------  WORKS AND JUST STOPS STA RECONNECT ATTEMPTS
    //WiFi.enableSTA(false);  // < ------  WORKS AND TURNS WIFI OFF (AP NOT YET ENABLED)
    delay(2000);    
    WiFi.softAP("espsoftap","12345678"); // no errors that I can see, but ap never shows up
    WiFi.enableSTA(false);  // < ------  WORKS AND TURNS STA OFF
  }
}

void loop(){

}
```

me-no-dev, 2022-09-16:
Tested working on all chips. To not see AP at all, I would guess was an issue in IDF that was since fixed. We will release any day now, so following the above notes, you should be fine.

VojtechBartoska, 2022-09-19:
@tablatronix Can you please help testing this under 2.0.5? Thanks!

tablatronix, 2022-09-20:
Will do, I am on platformio or i would have already

roysG, 2023-01-07:
Is there a

> Just tested on C3 with what will be in 2.0.5 and here is your sketch with all options to make AP work commented and explained. While STA is trying to reconnect, it switches channels while scanning and that causes AP to lose communication (it works only on one channel).
> 
> ```c++
> #include <WiFi.h>
> #include <esp_wifi.h>  
> 
> void setup(){
>   Serial.begin(115200);
>   Serial.setDebugOutput(true);
>   Serial.setDebugOutput(true);
>   delay(2000);
>   Serial.println("Startup....");
> 
>   //WiFi.setAutoReconnect(false); // < ------  WORKS BUT YOU SHOULD KNOW THE IMPLICATIONS
>   WiFi.begin("hellowifi","noonehere");
>   
>   while (WiFi.status() != WL_CONNECTED && millis()<10000) {
>     delay(500);
>     Serial.print(".");
>   }
> 
>   if(WiFi.status() == WL_CONNECTED){
>     Serial.println("");
>     Serial.println("WiFi connected.");
>     Serial.println("IP address: ");
>     Serial.println(WiFi.localIP());
>   }
>   else {
>     Serial.println("WiFi NOT CONNECTED, starting ap");
>     //WiFi.disconnect();      // < ------  WORKS AND JUST STOPS STA RECONNECT ATTEMPTS
>     //WiFi.enableSTA(false);  // < ------  WORKS AND TURNS WIFI OFF (AP NOT YET ENABLED)
>     delay(2000);    
>     WiFi.softAP("espsoftap","12345678"); // no errors that I can see, but ap never shows up
>     //WiFi.enableSTA(false);  // < ------  WORKS AND TURNS STA OFF
>   }
> }
> 
> void loop(){
> 
> }
> ```

Does the two delay(2000) importants?

AndreiKorchunov, 2023-01-17:
Board
esp32 c3 devkit-02
Hardware Configuration
nothing special
Arduino IDE 1.8.19 (Windows Store 1.8.57.0)
Library esp32 v 2.0.6 - softAp not work by standart example for FTM. The access point does not connect to the second esp 32, nor the Samsung Note.  There are no errors in any esp32.
Library esp32 v 2.0.3 - softAp work by standart example for FTM.

LoopOnTech, 2023-01-21:
Take a look at reducing the transmit power after WiFi has started.

The transmit power mappings can be found [here](https://github.com/espressif/arduino-esp32/blob/23d715af1b185fac21558c907af4ff5b86526aa4/libraries/WiFi/src/WiFiGeneric.h#L111).

```
    Serial.println(WiFi.softAP("espsoftap", "12345678"));
    WiFi.setTxPower(WIFI_POWER_15dBm);
    int a = WiFi.getTxPower();
    Serial.print("TX power: ");
    Serial.println(a);
```
Check the getTxPower() value to confirm that the setting is being applied.

LouDou, 2023-04-26:
> Take a look at reducing the transmit power after WiFi has started.
> 
> The transmit power mappings can be found [here](https://github.com/espressif/arduino-esp32/blob/23d715af1b185fac21558c907af4ff5b86526aa4/libraries/WiFi/src/WiFiGeneric.h#L111).
> 
> ```
>     WiFi.softAP("espsoftap", "12345678");
>     WiFi.setTxPower(WIFI_POWER_15dBm);
>     int txPower = WiFi.getTxPower();
>     Serial.print("TX power: ");
>     Serial.println(txPower);
> ```
> 
> Check the getTxPower() value to confirm that the setting is being applied. I could not get anything to work until the power was reduced. Try different values and see what works for you.

Setting the power like this makes it work on ESP32 Lolin S2 mini.

Thanks for the tip!

closed 2025-01-16 by Parsaabasi