espressif/arduino-esp32 #7236: [esp32S3] USBMSC example crash

chegewara, 2022-09-09:
### Board

esp32 s3 devkit

### Device Description

devkit

### Hardware Configuration

-

### Version

v2.0.4

### IDE Name

arduino IDE

### Operating System

linux

### Flash frequency

40

### PSRAM enabled

no

### Upload speed

115200

### Description

When i am trying to perform benchmark on firmware MSC disk app crash (stack canary)

### Sketch

```cpp
https://github.com/espressif/arduino-esp32/tree/master/libraries/USB/examples/USBMSC
```

### Debug Message

```plain
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x1 (POWERON),boot:0x8 (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
Octal Flash Mode Enabled
For OPI Flash, Use Default Flash Boot Mode
mode:SLOW_RD, clock div:1
load:0x3fcd0108,len:0x43c
load:0x403b6000,len:0xbd0
load:0x403ba000,len:0x29c8
entry 0x403b61d8
Guru Meditation Error: Core  0 panic'ed (Unhandled debug exception). 
Debug exception reason: Stack canary watchpoint triggered (usbd)
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

Jason2866, 2022-09-09:
Your flash mode for OPI flash is not (wrong?) set.
```
For OPI Flash, Use Default Flash Boot Mode
```
So writing to flash is not working correctly. Typical issue when flash mode is false.

chegewara, 2022-09-09:
The problem is not with flash, OPI type etc. 
It is `Stack canary watchpoint triggered (usbd)` which requires to be increased.
Other than that example works fine with both OPI and QIO mode, depending on flash supported.

SuGlider, 2022-09-19:
@chegewara - Could you please describe better how to reproduce the issue?
I'm using Win11 and I get no error so far.

chegewara, 2022-09-19:
I have no idea how to describe it better. I am using vanilla example, with just delay() in loop to avoid WDT.
The only difference between my test and yours test is the fact i am using Ubuntu 20.04 and disk's benchmark test.

Example screen:
[image]

SuGlider, 2022-09-19:
The [vanilla example sketch](https://github.com/espressif/arduino-esp32/tree/master/libraries/USB/examples/USBMSC) creates a Mass Storage Drive with just 6KB available space. In Win11, this drive is labled `ESP32S2 MSC` and it has a single file `README.TXT`. I'm (over)writing a new file of another 2KB to the drive several times and I get no error.

@chegewara - if possible, please turn on Debug messages and post here this output when the issue occurs.

closed 2022-10-05 by SuGlider

chegewara, 2022-10-05:
Sorry, ive been sick and i had no time to confirm it, but it is still not solved to me.