espressif/arduino-esp32 #7238: ESP32-S3 DFU Mode creates new COM port every time it connects

jhsa, 2022-09-09:
### Board

ESP32-S3 

### Device Description

ESP32-S3 WROOM board 8Mb flash and 8Mb RAM (I think).

[image]

### Hardware Configuration

TFT display connected via SPI pins, but not in use, I am using the Adafruit TinyUSB library for USB Midi, BLE-Midi libraryand the Arduino Midi Library. I need all running at the same time. For Serial Midi to work I need to have CDC turned off as far as my testing says, and must use DFU to flash the board using the USB OTG connector.  IDE configuration is attached.

[image]

### Version

v2.0.4

### IDE Name

Arduino IDE

### Operating System

Windows 10

### Flash frequency

QIO 80Mhz

### PSRAM enabled

no

### Upload speed

921600

### Description

Reset the Board while pressing the Boot button. 
A Com port is created as shown. In this case COM26, as you can see in the picture. If I disconnect the board and do the same procedure again, it will create another COM port on COM27. So, it is filling my Windows machine with lots of COM ports. I think that is not a good thing, and it should not happen?
Thanks.

[image]

### Sketch

```cpp
Not relevant, as it happens with every single sketch?
```

### Debug Message

```plain
None..
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

jhsa, 2022-09-09:
Further information, if I follow the tutorial on this link to the letter:

https://docs.espressif.com/projects/arduino-esp32/en/latest/tutorials/cdc_dfu_flash.html

I get the following error when flashing.

```
Traceback (most recent call last):
  File "esptool.py", line 5387, in <module>
  File "esptool.py", line 5380, in _main
  File "esptool.py", line 4687, in main
  File "esptool.py", line 114, in get_default_connected_device
  File "esptool.py", line 320, in __init__
  File "serial\__init__.py", line 90, in serial_for_url
  File "serial\serialwin32.py", line 80, in open
esptool.py v3.3
  File "serial\serialwin32.py", line 222, in _reconfigure_port
Serial port COM27
serial.serialutil.SerialException: Cannot configure port, something went wrong. Original message: PermissionError(13, 'A device attached to the system is not functioning.', None, 31)
[1600] Failed to execute script 'esptool' due to unhandled exception!
the selected serial port [1600] Failed to execute script 'esptool' due to unhandled exception!
 does not exist or your board is not connected
```
If I press BOOT at power ON, it will flash but it will increase the COM port number every time I do this.

me-no-dev, 2022-09-14:
> I am using the Adafruit TinyUSB library and must use DFU to flash the board using the USB OTG connector.

1. You are using external library, look for issues there.
2. Since you do not use the internal library, you should disable DFU in the menu.
3. DFU flashing has not been verified on S3 (for Arduino)
4. Operating systems will change the port based on different reasons, but generally different USB descriptors and vid/pid/serial

jhsa, 2022-09-14:
> 1. You are using external library, look for issues there.
> 2. Since you do not use the internal library, you should disable DFU in the menu.
> 3. DFU flashing has not been verified on S3 (for Arduino) [...]

Hi, thank you for your reply.
1. I asked here because I thought this problem is caused by the S3 core.
2.  I cannot disable DFU because otherwise USB MIDI does not work, Don't ask me why, I am a beginner ;)
3.  I can tell you it does work, but as I reported above, with it produces a new COM port on the PC every time I put the Microcontroller in BOOT mode, which is the reason of this issue. It flashes the code OK.
4. Ok, then something might be generating different USB descriptors and vid/pid/serial every time I put the board in BOOT mode. And this is what we would like to find out, I think?

rftafas, 2025-01-10:
The issue seems to be related to external lib. In case of updates in the subject pointing to issues on ESP32-S3 hardware or Espressif portions of the software, re-open this.

closed 2025-01-10 by rftafas