espressif/arduino-esp32 #7240: The WiFi disconnects as soon as the CPU frequency is modified with setCpuFrequencyMhz()

cicciocb, 2022-09-11:
### Board

ESP32 Dev Module

### Device Description

A bare ESP32 Dev Module

### Hardware Configuration

N/A

### Version

v2.0.4

### IDE Name

Arduino IDE

### Operating System

Windows 10

### Flash frequency

80 Mhz

### PSRAM enabled

no

### Upload speed

115200

### Description

The WiFi connection is lost as soon as the setCpuFrequencyMhz() is used with a value or 80 or 160.
The connection comes back when the frequency is set back to 240MHz.

### Sketch

```cpp
#include <WiFi.h>

void onGotIP(WiFiEvent_t event, WiFiEventInfo_t info) {
    Serial.print(F("\nConnected to "));
    Serial.println(WiFi.SSID());
    Serial.print(F("IP address: "));
    Serial.println(WiFi.localIP());
    WiFi.setAutoReconnect(true);
}

void onConnected(WiFiEvent_t event, WiFiEventInfo_t info) {
    Serial.println("CONNECTED");
}

void onDisconnected(WiFiEvent_t event, WiFiEventInfo_t info) {
    Serial.println("DISCONNECTED");
}

void setup()
{
    Serial.begin(115200);
    WiFi.mode(WIFI_STA);
    WiFi.onEvent(onGotIP, ARDUINO_EVENT_WIFI_STA_GOT_IP);
    WiFi.onEvent(onConnected, ARDUINO_EVENT_WIFI_STA_CONNECTED);
    WiFi.onEvent(onDisconnected, ARDUINO_EVENT_WIFI_STA_DISCONNECTED);

    WiFi.setSleep(false); // enable the wifi all the time
    WiFi.begin("Bbox-AED5AE73", "mypass");
    for (int i = 0; i < 50; i++)
    {
        if (WiFi.status() == WL_CONNECTED) {
            Serial.printf("Connected\n");
            break;
        }
        Serial.print(".");
        delay(500);
    }
    delay(10000);
    Serial.print("Freq 160");
    setCpuFrequencyMhz(160);
    delay(10000);
    Serial.print("Freq 80");
    setCpuFrequencyMhz(80);
    delay(10000);
    Serial.print("Freq 240");
    setCpuFrequencyMhz(240);

}

// Add the main program code into the continuous loop() function
void loop()
{

}
```

### Debug Message

```plain
ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0030,len:1344
load:0x40078000,len:13864
load:0x40080400,len:3608
entry 0x400805f0
[    57][D][WiFiGeneric.cpp:929] _eventCallback(): Arduino Event: 0 - WIFI_READY
[   141][V][WiFiGeneric.cpp:338] _arduino_event_cb(): STA Started
[   143][V][WiFiGeneric.cpp:97] set_esp_interface_ip(): Configuring Station static IP: 0.0.0.0, MASK: 0.0.0.0, GW: 0.0.0.0
[   143][D][WiFiGeneric.cpp:929] _eventCallback(): Arduino Event: 2 - STA_START
.[   212][V][WiFiGeneric.cpp:353] _arduino_event_cb(): STA Connected: SSID: Bbox-AED5AE73, BSSID: d8:d7:75:d6:44:80, Channel: 1, Auth: WPA_WPA2_PSK
[   214][D][WiFiGeneric.cpp:929] _eventCallback(): Arduino Event: 4 - STA_CONNECTED
CONNECTED
[   558][V][WiFiGeneric.cpp:367] _arduino_event_cb(): STA Got New IP:192.168.1.85
[   558][D][WiFiGeneric.cpp:929] _eventCallback(): Arduino Event: 7 - STA_GOT_IP
[   561][D][WiFiGeneric.cpp:991] _eventCallback(): STA IP: 192.168.1.85, MASK: 255.255.255.0, GW: 192.168.1.254

Connected to Bbox-AED5AE73
IP address: 192.168.1.85
Connected
��00662][D][esp32-hal-cpu.c:214] setCpuFrequencyMhz(): PLL: 320 / 2 = 160 Mhz, APB: 80000000 Hz
Freq 160
[ 19143][V][WiFiGeneric.cpp:360] _arduino_event_cb(): STA Disconnected: SSID: Bbox-AED5AE73, BSSID: d8:d7:75:d6:44:80, Reason: 200
[ 19144][D][WiFiGeneric.cpp:929] _eventCallback(): Arduino Event: 5 - STA_DISCONNECTED
[ 19152][W][WiFiGeneric.cpp:950] _eventCallback(): Reason: 200 - BEACON_TIMEOUT
[ 19159][D][WiFiGeneric.cpp:966] _eventCallback(): WiFi Reconnect Running
[ 19167][V][WiFiGeneric.cpp:97] set_esp_interface_ip(): Configuring Station static IP: 0.0.0.0, MASK: 0.0.0.0, GW: 0.0.0.0
DISCONNECTED
[ 20663][D][esp32-hal-cpu.c:214] setCpuFrequencyMhz(): PLL: 320 / 4 = 80 Mhz, APB: 80000000 Hz
Freq 80
[ 21241][V][WiFiGeneric.cpp:360] _arduino_event_cb(): STA Disconnected: SSID: Bbox-AED5AE73, BSSID: 00:00:00:00:00:00, Reason: 201
[ 21242][D][WiFiGeneric.cpp:929] _eventCallback(): Arduino Event: 5 - STA_DISCONNECTED
[ 21249][W][WiFiGeneric.cpp:950] _eventCallback(): Reason: 201 - NO_AP_FOUND
[ 21256][D][WiFiGeneric.cpp:975] _eventCallback(): WiFi AutoReconnect Running
[ 21266][V][WiFiGeneric.cpp:97] set_esp_interface_ip(): Configuring Station static IP: 0.0.0.0, MASK: 0.0.0.0, GW: 0.0.0.0
DISCONNECTED
[ 23349][V][WiFiGeneric.cpp:360] _arduino_event_cb(): STA Disconnected: SSID: Bbox-AED5AE73, BSSID: 00:00:00:00:00:00, Reason: 201
[ 23350][D][WiFiGeneric.cpp:929] _eventCallback(): Arduino Event: 5 - STA_DISCONNECTED
[ 23357][W][WiFiGeneric.cpp:950] _eventCallback(): Reason: 201 - NO_AP_FOUND
[ 23364][D][WiFiGeneric.cpp:975] _eventCallback(): WiFi AutoReconnect Running
[ 23374][V][WiFiGeneric.cpp:97] set_esp_interface_ip(): Configuring Station static IP: 0.0.0.0, MASK: 0.0.0.0, GW: 0.0.0.0
DISCONNECTED
[ 25456][V][WiFiGeneric.cpp:360] _arduino_event_cb(): STA Disconnected: SSID: Bbox-AED5AE73, BSSID: 00:00:00:00:00:00, Reason: 201
[ 25457][D][WiFiGeneric.cpp:929] _eventCallback(): Arduino Event: 5 - STA_DISCONNECTED
[ 25464][W][WiFiGeneric.cpp:950] _eventCallback(): Reason: 201 - NO_AP_FOUND
[ 25471][D][WiFiGeneric.cpp:975] _eventCallback(): WiFi AutoReconnect Running
[ 25481][V][WiFiGeneric.cpp:97] set_esp_interface_ip(): Configuring Station static IP: 0.0.0.0, MASK: 0.0.0.0, GW: 0.0.0.0
DISCONNECTED
[ 27563][V][WiFiGeneric.cpp:360] _arduino_event_cb(): STA Disconnected: SSID: Bbox-AED5AE73, BSSID: 00:00:00:00:00:00, Reason: 201
[ 27564][D][WiFiGeneric.cpp:929] _eventCallback(): Arduino Event: 5 - STA_DISCONNECTED
[ 27571][W][WiFiGeneric.cpp:950] _eventCallback(): Reason: 201 - NO_AP_FOUND
[ 27578][D][WiFiGeneric.cpp:975] _eventCallback(): WiFi AutoReconnect Running
[ 27588][V][WiFiGeneric.cpp:97] set_esp_interface_ip(): Configuring Station static IP: 0.0.0.0, MASK: 0.0.0.0, GW: 0.0.0.0
DISCONNECTED
[ 29669][V][WiFiGeneric.cpp:360] _arduino_event_cb(): STA Disconnected: SSID: Bbox-AED5AE73, BSSID: 00:00:00:00:00:00, Reason: 201
[ 29670][D][WiFiGeneric.cpp:929] _eventCallback(): Arduino Event: 5 - STA_DISCONNECTED
[ 29678][W][WiFiGeneric.cpp:950] _eventCallback(): Reason: 201 - NO_AP_FOUND
[ 29684][D][WiFiGeneric.cpp:975] _eventCallback(): WiFi AutoReconnect Running
[ 29695][V][WiFiGeneric.cpp:97] set_esp_interface_ip(): Configuring Station static IP: 0.0.0.0, MASK: 0.0.0.0, GW: 0.0.0.0
DISCONNECTED
[�����mum����2-hal-cpu.c:214] setCpuFrequencyMhz(): PLL: 480 / 2 = 240 Mhz, APB: 80000000 Hz
Freq 240
[ 30853][V][WiFiGeneric.cpp:353] _arduino_event_cb(): STA Connected: SSID: Bbox-AED5AE73, BSSID: d8:d7:75:d6:44:80, Channel: 1, Auth: WPA_WPA2_PSK
[ 30856][D][WiFiGeneric.cpp:929] _eventCallback(): Arduino Event: 4 - STA_CONNECTED
CONNECTED
[ 30879][V][WiFiGeneric.cpp:367] _arduino_event_cb(): STA Got Same IP:192.168.1.85
[ 30879][D][WiFiGeneric.cpp:929] _eventCallback(): Arduino Event: 7 - STA_GOT_IP
[ 30882][D][WiFiGeneric.cpp:991] _eventCallback(): STA IP: 192.168.1.85, MASK: 255.255.255.0, GW: 192.168.1.254

Connected to Bbox-AED5AE73
IP address: 192.168.1.85
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

SuGlider, 2022-09-11:
@cicciocb - I can't reproduce this issue using Arduino Core 2.0.4. 
It works fine here. I can also access any WebSite using `WiFiClient` at any CPU Frequency.

cicciocb, 2022-09-12:
Very strange, I tried to download again the core 2.0.4 and I still have the same problem.
I tried also with Wemos D1 mini and I confirm the same problem.
The same code on an ESP32-S2 works well.

SuGlider, 2022-09-12:
I have used this sketch (similar to the one posted here + WiFiClient test):

``` cpp
#include <WiFi.h>

void onGotIP(WiFiEvent_t event, WiFiEventInfo_t info) {
  Serial.print(F("\nConnected to "));
  Serial.println(WiFi.SSID());
  Serial.print(F("IP address: "));
  Serial.println(WiFi.localIP());
  WiFi.setAutoReconnect(true);
}

void onConnected(WiFiEvent_t event, WiFiEventInfo_t info) {
  Serial.println("CONNECTED");
}

void onDisconnected(WiFiEvent_t event, WiFiEventInfo_t info) {
  Serial.println("DISCONNECTED");
}

void readRemoteSite() {
  const char* host     = "espressif.com";
  const char* url      = "/index.html";
  Serial.print("connecting to ");
  Serial.println(host);

  // Use WiFiClient class to create TCP connections
  WiFiClient client;
  const int httpPort = 80;
  WiFi.mode(WIFI_AP_STA);
  WiFi.enableSTA(true);

  if (!client.connect(host, httpPort)) {
    Serial.println("connection failed");
    return;
  }

  Serial.print("Requesting URL: ");
  Serial.println(url);

  // This will send the request to the server
  client.print(String("GET ") + url + " HTTP/1.1\r\n" +
               "Host: " + host + "\r\n" +
               "Connection: close\r\n\r\n");
  unsigned long timeout = millis();
  while (client.available() == 0) {
    if (millis() - timeout > 5000) {
      Serial.println(">>> Client Timeout !");
      client.stop();
      return;
    }
  }

  // Read all the lines of the reply from server and print them to Serial
  while (client.available()) {
    String line = client.readStringUntil('\r');
    Serial.print(line);
  }

  Serial.println();
  Serial.printf("\nRunning at %d MHz\n", getCpuFrequencyMhz());
  Serial.println("closing connection");
}

void setup()
{
  Serial.begin(115200);
  WiFi.mode(WIFI_STA);
  WiFi.onEvent(onGotIP, ARDUINO_EVENT_WIFI_STA_GOT_IP);
  WiFi.onEvent(onConnected, ARDUINO_EVENT_WIFI_STA_CONNECTED);
  WiFi.onEvent(onDisconnected, ARDUINO_EVENT_WIFI_STA_DISCONNECTED);

  WiFi.setSleep(false); // enable the wifi all the time
  WiFi.begin("SSID", "wifi_Password");
  for (int i = 0; i < 50; i++)
  {
    if (WiFi.status() == WL_CONNECTED) {
      Serial.printf("Connected\n");
      break;
    }
    Serial.print(".");
    delay(500);
  }
  readRemoteSite();
  delay(10000);
  Serial.print("Freq 160");
  setCpuFrequencyMhz(160);
  readRemoteSite();
  delay(10000);

  Serial.print("Freq 80");
  setCpuFrequencyMhz(80);
  readRemoteSite();
  delay(10000);
  readRemoteSite();

  Serial.print("Freq 240");
  setCpuFrequencyMhz(240);
  readRemoteSite();

}

// Add the main program code into the continuous loop() function
void loop()
{
  Serial.println("End of Testing....");
  delay(10000);
}
```

felmue, 2022-09-12:
Hi @cicciocb 

just a thought - have you tried a different AP?

Thanks
Felix

cicciocb, 2022-09-12:
As you can see in the latest log, I'm using another AP (MYSPOT) in another site.

felmue, 2022-09-12:
Hi @cicciocb 

ok, sorry I missed that.

Thanks
Felix

cicciocb, 2022-09-14:
@felmue You are welcome.
In the while I tested with another kind of module, an ESP32-CAM and I confirm this issue.
The the same problem exists also with the SDK 1.0.6
Rolling back the SDK to the version 1.0.4 the problem disappears, so it looks that something happened from the 1.0.4 to the 1.0.6

cicciocb, 2022-09-14:
Just to conclude this post, I found a workaround, maybe can be useful for others.
In fact using the following sequence
````
    WiFi.setSleep(true); delay(100);
    setCpuFrequencyMhz(160);
    WiFi.setSleep(false);
````
instead of the simple 
````
setCpuFrequencyMhz(160);
````
enables to change the CPU frequency without  WiFi disconnections

SuGlider, 2022-09-19:
>Rolling back the SDK to the version 1.0.4 the problem disappears, so it looks that something happened from the 1.0.4 to the 1.0.6

It may be related to IDF version as well... WiFi is based on IDF llibraries.
I think it may have relationship with Authentication and WiFi.
Do you know what is the Authentication method used in the AP (MYSPOT)?
Is it WPA, WAP2, WAP_WAP2, WAP3, None...
Is TKIP enabled?

closed 2022-09-19 by SuGlider

srcastor, 2023-01-09:
Same problem here, the workaround from @cicciocb works but if you often play with the values ends up disconnecting, as example going back from 160Mhz modem sleep true to 240Mhz no modem sleep will force a disconnection, then reconnect and it is stable (until you change the frequency again).

scottchiefbaker, 2023-03-13:
I'm working on a [debugging library](https://github.com/scottchiefbaker/ESP-RIX) and one of the features I'd like to implement is CPU Speed switching on the fly. Whenever I use `setCpuFreqMhx()` **after** WiFi is initialized I lose connection. I'm using ESP 2.0.7 if it matters.

Is CPU speed switching not supported while using WiFi?

fredlcore, 2023-03-19:
Switching CPU speed works for me generally also with WIFI whether using the workaround above or not, but what I experience here is that my sketch runs most of the time with 80 MHz and only for the purpose of outputting a data log, I increase the speed to 240 MHz. With LAN, this brings an output improvement of around 20%, but with WIFI it's actually getting worse! Instead of around 300 kB/s with 80 MHz, I get less than 100 kB/s, sometimes even much less or even a stalled connection. This is not the case when setting the CPU speed once and for all at the beginning of the script, both with 80 or with 240 MHz.

JooJooBee666, 2023-08-09:
This is still an issue and should be reopened.  The workaround that @cicciocb posted does not work for me.  I still lose connectivity about 5 seconds after lowing the CPU speed and it does not reconnect until I bump the speed back up to 240.  Can we reopen this so the actual source of the problem is identified?

JooJooBee666, 2023-08-10:
Should I just open a new issue?

srcastor, 2023-11-09:
Any news on this?