espressif/arduino-esp32 #7243: Custom Board not recognize Nand Flash

hdpklm, 2022-09-12:
### Board

ESP32 D0WD

### Device Description

custom pcb with esp32 and nand flash

### Hardware Configuration

I tested the VDD_SDIO  and its 3,14V so the problem its not in the power supply

| ESP32  | Nand Flash |
| ------------- | ------------- |
| VDD_SDIO  | VCC |
| GPIO6   | SCLK |
| GPIO7   | SI |
| GPIO8   | SO |
| GPIO9   | WP# |
| GPIO10  | HOLD# |
| GPIO11  | CE# |

### Version

latest master (checkout manually)

### IDE Name

vscode

### Operating System

windows 10

### Flash frequency

40Mhz

### PSRAM enabled

no

### Upload speed

19200

### Description

Hello

I build my own pcb with esp32 and nand flash. I mount a lot of pcb with esp32-d2wd without any problem, because it has its own internal flash.
but now i design the pcb with the recommended connection between esp32 and nand flash but it doesn't recognize the flash.

in the datasheet it is mentioned that the maximum size of flash can map its 16 MB, but it is not mentioned if more than this size can cause a problem.

`Up to 16 MB of external flash can be mapped into CPU instruction memory space and read-only memory
space simultaneously`

So I don't know if the problem is because I have to configure the efuse esp32 or something else to read the external flash or the problem is that I connect it to a flash larger than 16MB

thank you

### Sketch

```cpp
no have a code
```

### Debug Message

```plain
C:\hassan\device>esptool.exe --chip esp32 --port COM4 --baud 19200 --before default_reset --after hard_reset write_flash -z --flash_mode dio --flash_freq 80m --flash_size detect 0x10000 ./build/main.ino.bin
esptool.py v3.0-dev
Serial port COM4
Connecting......
Chip is ESP32D0WDQ5 (revision 3)
Features: WiFi, BT, Dual Core, 240MHz, VRef calibration in efuse, Coding Scheme None
Crystal is 40MHz
MAC: 30:c6:f7:56:21:b4
Uploading stub...
Running stub...
Stub running...
Configuring flash size...
Warning: Could not auto-detect Flash size (FlashID=0xffffff, SizeID=0xff), defaulting to 4MB
Compressed 853712 bytes to 480559...

A fatal error occurred: Timed out waiting for packet content
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

me-no-dev, 2022-09-14:
you did not mention what that flash chip is, but I suspect that it is incompatible.

hdpklm, 2022-09-14:
Hello @me-no-dev 

its a 1GB Flash `DS35Q1GA-IB`
Are there any restrictions to mount an external Flash?

thank you

me-no-dev, 2022-09-14:
We do not support such flash chips. You need to use Serial Flash chips capable of at least DIO mode, like [W25Q64](https://www.pjrc.com/store/w25q64fv.pdf) or similar

closed 2022-09-14 by me-no-dev

hdpklm, 2022-09-14:
@me-no-dev 
ok thank you

ParameswaranAP, 2022-12-06:
Can I use SPI-based NAND flash for external memory in ESP32-S3 for my custom board?
is SPIFFS API support this with An flat file system?

lbernstone, 2022-12-06:
I don't know of any NAND that is supported. The maximum SPI speed is 80MHz.  You should use QIO for best performance.
SPIFFS is a poor choice for a large file system.  Use LittleFS or FAT.

ParameswaranAP, 2022-12-06:
Thankyou @lbernstone , we are using external SPI NAND Flash memory for our project with esp32s3 and need wear leveling for increase memory cycle. If we are using LittleFS or FAT is the wearleveling is supported or not?

lbernstone, 2022-12-06:
LittleFS has wear levelling in the code.  FAT can be layered with wear levelling as is done with FFat, but this is not going to be a turnkey integration in arduino.  Note that using wear levelling will also alter the FAT filesystem such that it will not be directly usable outside of the esp32 device.  As this conversation is not actually an issue in arduino-esp32, I think you should take it over to esp32.com

ParameswaranAP, 2022-12-14:
After further R&D and from your suggestion. we are moving for the FAT filesystem with 16MB NOR flash with esp32S3. We aim to store sensor data in a particular .txt file in flash memory based on time intervals which is obtained from the RTC module. The data should last up to max 30 days. Is this the correct method that we are following or kindly suggest if I am wrong?