espressif/arduino-esp32 #7244: setSocketOption doesn't respect inheritance

bertmelis, 2022-09-12:
### Board

devkitv1

### Device Description

n/a

### Hardware Configuration

n/a

### Version

latest master (checkout manually)

### IDE Name

Platformio

### Operating System

Windows 11

### Flash frequency

80Mhz

### PSRAM enabled

no

### Upload speed

115200

### Description

https://github.com/espressif/arduino-esp32/blob/master/libraries/WiFi/src/WiFiClient.h#L89

`setSocketOption` isn't virtual so when calling for a `WiFiClientSecure` object through a `WiFiClient` pointer, you'll get an error.

### Sketch

```cpp
n/a
```

### Debug Message

```plain
[  8002][V][ssl_client.cpp:62] start_ssl_client(): Free internal heap before TLS 241696
[  8010][V][ssl_client.cpp:68] start_ssl_client(): Starting socket
[  8058][V][ssl_client.cpp:149] start_ssl_client(): Seeding the random number generator
[  8062][V][ssl_client.cpp:158] start_ssl_client(): Setting up the SSL/TLS structure...
[  8066][V][ssl_client.cpp:181] start_ssl_client(): Loading CA cert
[  8083][V][ssl_client.cpp:257] start_ssl_client(): Setting hostname for TLS session...
[  8085][V][ssl_client.cpp:272] start_ssl_client(): Performing the SSL/TLS handshake...
[  9556][V][ssl_client.cpp:293] start_ssl_client(): Verifying peer X.509 certificate...
[  9558][V][ssl_client.cpp:301] start_ssl_client(): Certificate verified.
[  9561][V][ssl_client.cpp:316] start_ssl_client(): Free internal heap after TLS 198284
[  9569][E][WiFiClient.cpp:313] setSocketOption(): fail on -1, errno: 9, "Bad file number"
```
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

noorhaq, 2022-09-14:
@bertmelis I am facing the same issue. Can you suggest any workaround or fix for it? It would be really helpful

bertmelis, 2022-09-14:
Make it virtual (in WiFiClient) or cast your pointer to WiFiClientSecure before calling setSocketOption.

The former is a fix, the latter unpractical.

noorhaq, 2022-09-14:
Hi @bertmelis,
I am a newbie on this side and tried by adding virtual keyword just now. No change
Can you add the code example if possible or guide me to a repo where you might have incorporated these changes

bertmelis, 2022-09-15:
Extra: `getNoDelay` doesn't work at all because `getOption` calls raw `getsockopt` directly. There's no `getSocketOption` to make virtual.

bertmelis, 2022-09-15:
As an alternative, one could implement `_fd()` in WiFiClientSecure and make that virtual.

bertmelis, 2022-09-19:
@VojtechBartoska Unless you find this not an issue, I'm happy to create a PR. But there are many roads to Rome and I can only pick one. So guide me in the desired direction.

Imho, we should remove the `setSocketOption` and `getSocketOption` from WiFiClientSecure and implement a `virtual int fd()`. `setSocketOption` and `getSocketOption` is duplicate but working on a different fd. By doing that, also `getOption` will work for WiFiclientSecure.

VojtechBartoska, 2022-09-19:
Hi @bertmelis, thanks for the PR. We will review it and we will see if this is a good option :)