espressif/arduino-esp32 #7246: Can the SD library support Chinese paths or file names?

zhushengji, 2022-09-13:
### Related area

SD

### Hardware specification

ESP32

### Is your feature request related to a problem?

It is common for file names to contain Chinese, so if UTF-8 can be supported, it will be very convenient for development.

### Describe the solution you'd like

SD library support for UTF-8

### I have checked existing list of Feature requests and the Contribution Guide

- [X] I confirm I have checked existing list of Feature requests and Contribution Guide.

VojtechBartoska, 2022-09-13:
what do you think @P-R-O-C-H-Y?

OceanZhouX, 2022-10-16:
Any updates on this?

justdomyself, 2022-11-04:
now  Can it support chinese filename ?

P-R-O-C-H-Y, 2022-11-15:
Hi @zhushengji @OceanZhouX @justdomyself,
where is the issue for you guys? I posted a simple working sketch where Chinese file names and dirs work perfectly (its simplified SD example). Can you retest it and post your debug output from Serial Monitor? Thanks

Tested on ESP32 Wrover kit + Arduino IDE 1.8.19

Sketch:
```
#include "SD.h"

//SD Pins definiton for ESP32-wroker kit
#define SD_SCK  14
#define SD_MISO  2
#define SD_MOSI  15
#define SD_CS  13

void listDir(fs::FS &fs, const char * dirname, uint8_t levels){
    Serial.printf("Listing directory: %s\n", dirname);

    File root = fs.open(dirname);
    if(!root){
        Serial.println("Failed to open directory");
        return;
    }
    if(!root.isDirectory()){
        Serial.println("Not a directory");
        return;
    }

    File file = root.openNextFile();
    while(file){
        if(file.isDirectory()){
            Serial.print("  DIR : ");
            Serial.println(file.name());
            if(levels){
                listDir(fs, file.path(), levels -1);
            }
        } else {
            Serial.print("  FILE: ");
            Serial.print(file.name());
            Serial.print("  SIZE: ");
            Serial.println(file.size());
        }
        file = root.openNextFile();
    }
}

void setup() {

    Serial.begin(115200);
    delay(200);

    SPI.begin(SD_SCK, SD_MISO, SD_MOSI, SD_CS);
    if(!SD.begin(SD_CS,SPI,40000000)){
        Serial.println("Card Mount Failed");
        return;
    }

    //Open file if exists or create new file and create subdir if not exist (chineese symbols are random)
    File file = SD.open("/获取磁盘文件列表/数量.txt", "a", true);
    if (file) {
      Serial.printf("File openned - %s \n", file.name());
      file.println("foo");
      file.close();
    }
    else Serial.println("File not found or cant be created");

    listDir(SD,"/获取磁盘文件列表",1);
  
    SD.end();
}

void loop() {
  // put your main code here, to run repeatedly:
     delay(100);
}
```

Debug output:
```
rst:0x1 (POWERON_RESET),boot:0x1e (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0030,len:1344
load:0x40078000,len:13864
load:0x40080400,len:3608
entry 0x400805f0
File opened - 数量.txt 
Listing directory: /获取磁盘文件列表
  FILE: 数量.txt  SIZE: 5
```

schreibfaul1, 2022-11-15:
The example works. If I want to look at the newly created folder '获取磁盘文件列表' on a Windows or Linux PC, I find:
[image]
What am I doing wrong?

justdomyself, 2022-11-21:
> 
I use my esp32-s borad test it, code below, can't find chinese filename:
`#include "FS.h"
#include "SD_MMC.h"

//最大文件数量
#define MAX_ROWS  300
//文件名最大长度
#define MAX_COLS  50

//文件信息结构体定义
typedef struct 
{
    char listSongs[100][50];
    int index = 0;
}list_t;

list_t listInst;

//插入文件名到列表
int InserSong(const char *name)
{
    int idx = listInst.index;

    if( strstr(name, ".mp3") == NULL ) return -1;

    memset( listInst.listSongs[idx], 0, MAX_COLS );
    strcpy(listInst.listSongs[idx], "/");
    strcat( listInst.listSongs[idx], name);
    listInst.index++;

    return 0;
}

//获取可用曲名数量
int GetSongCounts()
{
   return listInst.index;
}

//获取磁盘文件列表
void listDir(fs::FS &fs, const char * dirname, uint8_t levels)
{
    Serial.printf("Listing directory: %s\n", dirname);

    File root = fs.open(dirname);
    if(!root)
    {
        Serial.println("Failed to open directory");
        return;
    }
    if(!root.isDirectory())
    {
        Serial.println("Not a directory");
        return;
    }

    File file = root.openNextFile();
    while(file)
    {
        if(file.isDirectory())
        {
            Serial.print("  DIR : ");
            Serial.println(file.name());
            if(levels)
            {
                listDir(fs, file.path(), levels -1);
            }
        } 
        else 
        {
            Serial.print("  FILE: ");
            Serial.print(file.name());
            Serial.print("  SIZE: ");
            Serial.println(file.size());

            InserSong(file.name());
        }

        file = root.openNextFile();
    }
}
`

justdomyself, 2022-11-21:
> 
Then we can play chinese filename music by using your audio-iis  library?

P-R-O-C-H-Y, 2022-11-21:
@schreibfaul1 @justdomyself The UTF-8 support will be added in next release 2.0.6, which will be released soon. This [PR](https://github.com/espressif/esp32-arduino-lib-builder/pull/95) in Espressif's Arduino lib-builder repo adds it.

Jason2866, 2022-12-22:
Should be fixed already in master, since libs are builded with the mentioned PR.

schreibfaul1, 2022-12-26:
I can confirm this, it works now.

closed 2023-01-03 by VojtechBartoska