espressif/arduino-esp32 #7248: ESP32C3 HighZ on all GPIOs when deep sleep with RTC

0x0fe, 2022-09-13:
### Board

ESP32-C3 custom board

### Device Description

custom board with ESP32C3F4H and ultra lo Iq LDO (500nA).

### Hardware Configuration

external chips / peripherals have been remove or disconnected for the test, 
some pullups are left :
 0 ,1 ,2, 8 : 10K
10 : 100K

### Version

v2.0.4

### IDE Name

arduino IDE

### Operating System

windows 10

### Flash frequency

80M

### PSRAM enabled

no

### Upload speed

921600 (not relevant for CDC)

### Description

The curent drawn in deep sleep is 2500uA, while it should be 5uA, obviously there is a leak somehwere, i suspect the GPIOs are not going into high impedance when i set them to ANALOG, or somethign else is wrong. I use the RTC timer mode and disable other sources of wakeup. 
If i use the loop to set gpio_reset_pin then it cannot enters deep sleep at all. Since everything else is removed from the board i am not sure it comes from the ESP32C3, besides, teh quiescent current of the board when ESP32C3 is removed have been tested valid (under 0.5uA).

### Sketch

```cpp
void gpios_hiz(void){
  
  pinMode(0, ANALOG);
  pinMode(1, ANALOG);
  pinMode(2, ANALOG);
  pinMode(3, ANALOG);
  pinMode(4, ANALOG);
  pinMode(5, ANALOG);
  pinMode(6, ANALOG);
  pinMode(7, ANALOG);
  pinMode(8, ANALOG);
  pinMode(9, ANALOG);
  pinMode(10, ANALOG);
  
  pinMode(11, ANALOG); 
//  pinMode(12, ANALOG);
//  pinMode(13, ANALOG);
//  pinMode(14, ANALOG);
//  pinMode(15, ANALOG);
//  pinMode(16, ANALOG);
//  pinMode(17, ANALOG);
 
  pinMode(18, ANALOG);
  pinMode(19, ANALOG);
  pinMode(20, ANALOG);
  pinMode(21, ANALOG);

// this does not work and prevents entering deep sleep
//  for(int i = 0; i < SOC_GPIO_PIN_COUNT; ++i) {
//    if(GPIO_IS_VALID_GPIO(i) && !GPIO_IS_INTERNAL_FLASH_PIN(i)) {
//      gpio_reset_pin((gpio_num_t)i);
//    }
//  }
}
void deep_sleep_enter(void){
   
  esp_sleep_disable_wakeup_source(ESP_SLEEP_WAKEUP_ALL);
  esp_sleep_config_gpio_isolate();
  esp_deep_sleep(15*1000000);
}
void setup(){

  Serial.begin(115200);
  //while(!Serial)delay(1);
  printf("System ready");
  delay(8000);
  gpios_hiz();  
  deep_sleep_enter();  
}
void loop(){

}
```

### Debug Message

```plain
none
```

### Other Steps to Reproduce

none

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

0x0fe, 2022-09-13:
i disconnected 3 pullups and it is now down to 1975uA, this confirms the GPIOs are not really in high impedance state.

0x0fe, 2022-09-13:
nb: no rtc crystal is installed, since GPIO 0 and 1 are used

hdpklm, 2022-09-13:
`some pullups are left :
0 ,1 ,2, 8 : 10K
10 : 100K`

3.3V / 10K are 0.00033A or 0.33mA
0.33mA * 4 are about 1.32mA
so when you are remove ESP32, the internal inpedance or maybe its mode output. I not sure but the current is so similar 🤷‍♀️

hdpklm, 2022-09-13:
I have one cuestion: arduino IDE can program one core ESP32-C3 ???
I know the [arduino-esp32](https://github.com/espressif/arduino-esp32) only work with dual-core, any news ?

0x0fe, 2022-09-13:
not sure wht "internal impedance (of what) you are talking about. Besides, adding the pullup resistors is not going to give you the current, itis much more complex than that, some of these pullups are conenccted to MOS, etc. There is also relation between gpios. Bottom line is, it show the GPIOs are not high impedance, which is the problem

0x0fe, 2022-09-13:
yeah i think you dont have much clue, and thus not much business replying in such technical issue.
if your GPIOs are high impedance then the value external pullups have zero incidence, that is quite basic concept to understand. 

If you had to change your external pullups to reduce the current then it means you did not set the GPIOs to high impedance state.

lbernstone, 2022-09-13:
Are you able to get to the power level you are looking for if you use [esp_sleep_pd_config](https://docs.espressif.com/projects/esp-idf/en/latest/esp32c3/api-reference/system/sleep_modes.html#power-down-of-rtc-peripherals-and-memories) to turn off all the power domains?

0x0fe, 2022-09-14:
> Are you able to get to the power level you are looking for if you use [esp_sleep_pd_config](https://docs.espressif.com/projects/esp-idf/en/latest/esp32c3/api-reference/system/sleep_modes.html#power-down-of-rtc-peripherals-and-memories) to turn off all the power domains?

i did not tried to use the sleep pd config to manually turn everything off, good idea let me try this

0x0fe, 2022-09-14:
well, adding these actually increased the current to 3750uA

  ```
// disable all RTC features except the timer
  esp_sleep_pd_config(ESP_PD_DOMAIN_MAX, ESP_PD_OPTION_OFF);
  esp_sleep_pd_config(ESP_PD_DOMAIN_RTC_PERIPH, ESP_PD_OPTION_OFF);
  esp_sleep_pd_config(ESP_PD_DOMAIN_RTC_SLOW_MEM, ESP_PD_OPTION_OFF);
  esp_sleep_pd_config(ESP_PD_DOMAIN_RTC_FAST_MEM, ESP_PD_OPTION_OFF);
```

0x0fe, 2022-09-14:
tried this too, with not much success (around 3650uA)

```
void hibernate(void){

  esp_sleep_pd_config(ESP_PD_DOMAIN_RTC_PERIPH, ESP_PD_OPTION_OFF);
  esp_sleep_pd_config(ESP_PD_DOMAIN_RTC_FAST_MEM, ESP_PD_OPTION_OFF);
  esp_sleep_pd_config(ESP_PD_DOMAIN_XTAL, ESP_PD_OPTION_OFF);
  esp_sleep_pd_config(ESP_PD_DOMAIN_VDDSDIO, ESP_PD_OPTION_OFF);
  esp_sleep_pd_config(ESP_PD_DOMAIN_RTC8M, ESP_PD_OPTION_OFF);
  esp_sleep_pd_config(ESP_PD_DOMAIN_RTC_SLOW_MEM, ESP_PD_OPTION_OFF);
  esp_sleep_enable_timer_wakeup(15 * 1000000);
  printf("going to sleep. woken up %d\n", timesWokenUp++);
  esp_deep_sleep_start();
}
```

0x0fe, 2022-09-14:
for reference the schematic of the second board used for tests, S1 to S6 are going to a disconnected header, D3_P and D3_N is the USB-C receptacle, the only thing not shown on this schematic is the LDO which is the same as the first board, 50nA quiescent current, verified.

[image]

0x0fe, 2022-09-16:
i notice : when the C3 goes out of deep sleep it seems to reset and it does pass the bootloader again, which seems wrong, is that the expeted behaviour? anyway the problem is still not solve, with over 3250uA when it should be in deep sleep under 5uA, now tested after the LDO, so the only consumer here is the C3.

[image]

0x0fe, 2022-09-16:
about VDD SPI, here i feed it with teh LDO, shoudl i disconnect it from the 3V3 rail? I use teh H4 which has internal flash, it always seems a little strange to me that we have to feed VDD_SPI when using internal flash, decoupling is expected, but external feeding not so much.

0x0fe, 2022-09-16:
alright, so i went ahead and cut the traec, letting only decoupling, and miracle, the sleep current is not 5uA, well, what to say other than the issue system is not very efficient

lbernstone, 2022-09-16:
This forum is for issues with the arduino-esp32 software housed and developed here.  It is not for general hardware issues.  Those will receive more attention at [esp-idf](https://github.com/espressif/esp-idf/issues) or [esp32.com](https://esp32.com/viewforum.php?f=12&sid=a07786461de11ac05249cf9b99a47da6).

0x0fe, 2022-09-16:
that is true, however, there was no obvious clue that it was hardware related until i found the root cause, and the platform this project is tied to (by specs) is arduino ESP32 core. 

Regardless i think this is quite a major point to highlight in the hardware design reference and datasheet, the datasheet does not provide any warning regarding deep_sleep current leakage if  VDD_SPI is connected to the 3V3 rail, and the diagram below can easily lead to think it should or could be connected to the 3V3 rail, being directly powered by it. In fact it must NOT be connected to the 3V3 rail (unless we want a >1900uA leakage in deepsleep and halt mode). 
Not sure why they precise it is an input as well. In the case of H4 version (internal flash) it surely cannot be an input in any case since it powers the flash thus cannot be in GPIO mode, and leads to massive leakage if connected to the 3V3 rail.

[image]

[image]
 
So i let the issue open so that Espressif can hopefully read it and add some warning in the next datasheet/hardware referernce revisions.

0x0fe, 2023-02-05:
i close the issue, but i think it should be highlighted in the datasheet that VDD_SPI is not an input.

closed 2023-02-05 by 0x0fe