espressif/arduino-esp32 #7252: I2S  DAC    Board Manager > 2.2.0

JoseEnriqueFA, 2022-09-14:
### Board

Adafruit ESP32 Feather       Adafruit ESP32 Feather    V2

### Device Description

Nothing

### Hardware Configuration

Osciloscope attached to DAC2   pin 26 (both  Adafruit ESP32 Feather       Adafruit ESP32 Feather    V2)

### Version

v2.0.4

### IDE Name

Arduino IDE

### Operating System

Windows 10

### Flash frequency

40 Mhz

### PSRAM enabled

no

### Upload speed

115200

### Description

I have two sketches to generate sine waves (included bellow).
The first one writes directly to the DAC, the other one uses I2S and the internal converter of the ESP32.
With Arduino IDE and the ESP32 Feather board and the ESP32 board-Manager version 2.0.0 
Both sketches work perfectly.

If I update the ESP32 board-Manager to version 2.0.4, I can now work with the Adafruit ESP32 Feather V2.
The Sketch that writes directly to the DAC 2 works perfectly on both the Adafruit ESP32 Feather and the Adafruit ESP32 Feather V2.
But the sketch with I2S, compiles without problems but does NOT work on either the Adafruit ESP32 Feather or Adafruit ESP32 Feather V2. Pin 26 remains at 0v without generating the expected signal.
Evidently the problem is in the Arduino IDE, but there may be a work-around, you could share

### Sketch

```cpp
/*
 * Firsy SKETCH     Writing directly in the DAC
 * Tested on Adafruit  ESP32  Feather V2 y     Adafruit  ESP32 Feather          11-09-2022
 */

#define Num_Samples  112
 

int i = 0;

 // Sin wave
static byte WaveFormTable[Num_Samples] = 

   { 
    0x80, 0x83, 0x87, 0x8A, 0x8E, 0x91, 0x95, 0x98, 0x9B, 0x9E, 0xA2, 0xA5, 0xA7, 0xAA, 0xAD, 0xAF,
    0xB2, 0xB4, 0xB6, 0xB8, 0xB9, 0xBB, 0xBC, 0xBD, 0xBE, 0xBF, 0xBF, 0xBF, 0xC0, 0xBF, 0xBF, 0xBF,
    0xBE, 0xBD, 0xBC, 0xBB, 0xB9, 0xB8, 0xB6, 0xB4, 0xB2, 0xAF, 0xAD, 0xAA, 0xA7, 0xA5, 0xA2, 0x9E,
    0x9B, 0x98, 0x95, 0x91, 0x8E, 0x8A, 0x87, 0x83, 0x80, 0x7C, 0x78, 0x75, 0x71, 0x6E, 0x6A, 0x67,
    0x64, 0x61, 0x5D, 0x5A, 0x58, 0x55, 0x52, 0x50, 0x4D, 0x4B, 0x49, 0x47, 0x46, 0x44, 0x43, 0x42,
    0x41, 0x40, 0x40, 0x40, 0x40, 0x40, 0x40, 0x40, 0x41, 0x42, 0x43, 0x44, 0x46, 0x47, 0x49, 0x4B,
    0x4D, 0x50, 0x52, 0x55, 0x58, 0x5A, 0x5D, 0x61, 0x64, 0x67, 0x6A, 0x6E, 0x71, 0x75, 0x78, 0x7C
   };
   

void setup() 
{
   
}

void loop() 
{
 
  byte wave_type = 0; // Sine
  
  dacWrite(26, WaveFormTable[i]); //25 ou 26
  i++;
  
  if (i >= Num_Samples)
  i = 0; 
    
}

==============================================================================

/*
*  Second SKETCH
 * This is an example to write DAC at high frequency using the I2S peripheral
 */

#include <stdio.h>
#include <string.h>
#include <stdlib.h>

#include "soc/rtc_io_reg.h"
#include "soc/rtc_cntl_reg.h"
#include "soc/sens_reg.h"
#include "soc/rtc.h"

#include <Arduino.h>

#include <driver/i2s.h>
#include <driver/dac.h>   // libreria necesaria para definir DAC_CHANNEL_1 no es necesaria  para DMA DAC 
#include "freertos/queue.h"                           

#define I2S_SAMPLE_RATE     426595   
#define Num_Samples  256       

int i = 0;

static byte WaveFormTable[Num_Samples] = 
   // Sin wave con 256 samples.  
 
 {
   0x80, 0x83, 0x86, 0x89, 0x8C, 0x8F, 0x92, 0x95, 0x98, 0x9B, 0x9E, 0xA1, 0xA4, 0xA7, 0xAA, 0xAD, 0xB0, 0xB3, 0xB6, 0xB9,
   0xBB, 0xBE, 0xC1, 0xC3, 0xC6, 0xC9, 0xCB, 0xCE, 0xD0, 0xD2, 0xD5, 0xD7, 0xD9, 0xDB, 0xDE, 0xE0, 0xE2, 0xE4, 0xE6, 0xE7,
   0xE9, 0xEB, 0xEC, 0xEE, 0xF0, 0xF1, 0xF2, 0xF4, 0xF5, 0xF6, 0xF7, 0xF8, 0xF9, 0xFA, 0xFB, 0xFB, 0xFC, 0xFD, 0xFD, 0xFE,
   0xFE, 0xFE, 0xFE, 0xFE, 0xFF, 0xFE, 0xFE, 0xFE, 0xFE, 0xFE, 0xFD, 0xFD, 0xFC, 0xFB, 0xFB, 0xFA, 0xF9, 0xF8, 0xF7, 0xF6,
   0xF5, 0xF4, 0xF2, 0xF1, 0xF0, 0xEE, 0xEC, 0xEB, 0xE9, 0xE7, 0xE6, 0xE4, 0xE2, 0xE0, 0xDE, 0xDB, 0xD9, 0xD7, 0xD5, 0xD2,
   0xD0, 0xCE, 0xCB, 0xC9, 0xC6, 0xC3, 0xC1, 0xBE, 0xBB, 0xB9, 0xB6, 0xB3, 0xB0, 0xAD, 0xAA, 0xA7, 0xA4, 0xA1, 0x9E, 0x9B,
   0x98, 0x95, 0x92, 0x8F, 0x8C, 0x89, 0x86, 0x83, 0x80, 0x7C, 0x79, 0x76, 0x73, 0x70, 0x6D, 0x6A, 0x67, 0x64, 0x61, 0x5E, 0x5B,
   0x58, 0x55, 0x52, 0x4F, 0x4C, 0x49, 0x46, 0x44, 0x41, 0x3E, 0x3C, 0x39, 0x36, 0x34, 0x31, 0x2F, 0x2D, 0x2A, 0x28, 0x26, 0x24,
   0x21, 0x1F, 0x1D, 0x1B, 0x19, 0x18, 0x16, 0x14, 0x13, 0x11, 0xF, 0xE, 0xD, 0xB, 0xA, 0x9, 0x8, 0x7, 0x6, 0x5, 0x4, 0x4, 0x3,
   0x2, 0x2, 0x1, 0x1, 0x1, 0x1, 0x1, 0x1, 0x1, 0x1, 0x1, 0x1, 0x1, 0x2, 0x2, 0x3, 0x4, 0x4, 0x5, 0x6, 0x7, 0x8, 0x9, 0xA, 0xB,
   0xD, 0xE, 0xF, 0x11, 0x13, 0x14, 0x16, 0x18, 0x19, 0x1B, 0x1D, 0x1F, 0x21, 0x24, 0x26, 0x28, 0x2A, 0x2D, 0x2F, 0x31, 0x34,
   0x36, 0x39, 0x3C, 0x3E, 0x41, 0x44, 0x46, 0x49, 0x4C, 0x4F, 0x52, 0x55, 0x58, 0x5B, 0x5E, 0x61, 0x64, 0x67, 0x6A, 0x6D,
   0x70, 0x73, 0x76, 0x79, 0x7C
 };

void i2sInit()
{
   i2s_config_t i2s_config = {
    .mode = (i2s_mode_t)(I2S_MODE_MASTER | I2S_MODE_TX | I2S_MODE_DAC_BUILT_IN),                   
    .sample_rate =  I2S_SAMPLE_RATE,              // The format of the signal using ADC_BUILT_IN
    .bits_per_sample = I2S_BITS_PER_SAMPLE_16BIT, // is fixed at 12bit, stereo, MSB
    .channel_format = I2S_CHANNEL_FMT_RIGHT_LEFT,      //  I2S_CHANNEL_FMT_ONLY_LEFT     I2S_CHANNEL_FMT_RIGHT_LEFT  ****************************************
    .communication_format = (i2s_comm_format_t)(I2S_COMM_FORMAT_STAND_I2S | 0x01),       
    .intr_alloc_flags = ESP_INTR_FLAG_LEVEL1,               
    .dma_buf_count = 8,
    .dma_buf_len = 64,
    .use_apll = false,
    .tx_desc_auto_clear = false,                            
    .fixed_mclk = 0                                       
   };
   
   i2s_driver_install(I2S_NUM_0, &i2s_config, 0, NULL); //install and start i2s driver
   i2s_set_pin(I2S_NUM_0,NULL);    // for internal DAC                                                  

   
   i2s_set_dac_mode(I2S_DAC_CHANNEL_DISABLE);  // disable ambos canales este si funciona
   //i2s_set_dac_mode(I2S_DAC_CHANNEL_RIGHT_EN); // DAC_1 RIGHT es el DAC_1
   i2s_set_dac_mode(I2S_DAC_CHANNEL_LEFT_EN); // de esta forma tengo solo salida en el DAC_2 
 

}

void setup() {

  // Initialize the I2S peripheral
  i2sInit();

}

void loop() {

    // i2s bytes-written variable
    size_t bytesOut;

    // output the buffer
    while( true )
    {
         i2s_write( I2S_NUM_0, WaveFormTable, Num_Samples, &bytesOut, portMAX_DELAY );

    }

}
```

### Debug Message

```plain
Nothing reported
See Description
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

VojtechBartoska, 2022-09-16:
@PilnyTomas can you please help with triage?

JoseEnriqueFA, 2022-09-26:
> Enviado el: viernes, 16 de septiembre de 2022 13:32

Dear Sirs,

 

I would be grateful if you could share the status of this escalation.

 

It is very important to me to find a solution.

If you can advise me :

  a) With a work around to be able to work with the arduino IDE (board-manager 2.0.0 )and the Adafruit ESP32 feather V2 module 

 

or

 

b) When a new Board Manager to correct the I2S bug could be expected. 

 

thanks in advance for any information

Best wishes

PilnyTomas, 2022-10-04:
Hi, sorry for the delay. It might be a bug in IDF, I am currently investigating it. The simple workaround is to use an older version.

JoseEnriqueFA, 2022-10-04:
Dear Tomás,

 

Best news that an investigation is on going .... I'd appreciate if you could share with me the results .....

 

FYI:

         1- 

                About the last two years, I have been working with the Arduino ID  1.8.13   and the board  ADAFRUIT Feather ESP32 

                To be able to choose your board:    in Tools you have   the Board Manager  (gestor de tarjetas in spanish)  you choose ESP32  and you install the version you need.

                I was working with version 1.0.6  No problem at all.

 

                Then some weeks ago,  I was aware about the new ADAFRUIT Feather ESP32 V2. 

                I took the decision to test this new board ...

                I needed to upgrade the ESP32 Board Manager to 2.0.4 to have the possibility to choose the board ADAFRUIT Feather ESP32 V2. 

                Everything seems to work fine in the ADAFRUIT Feather ESP32 V2, but I discover a problem with the internal ESP32 - DAC  driven by  I2S what is critical for me.  

                Then I tested the same feature in the ADAFRUIT Feather ESP32 and same result "not working". I have shared the sketch

 

                Then I began to down-grade the ESP32 Board Manager, testing the different versions one by one. 

                Only when I reached the ESP32 Board Manager to 2.0.0 (see attached file) the internal ESP32 - DAC  driven by  I2S works again.

                But with this version of Board Manager I cannot choose as target board the  ADAFRUIT Feather ESP32 V2.

 

                Last week, I have realized that there is a new  ESP32 Board Manager the 2.0.5 , I have just tested and  internal ESP32 - DAC  driven by  I2S is not working

 

 

        2-  I2C   Adafruit  feather ESP32 V2       bug  

  

              The I2C  works with the board Adafruit  feather ESP32 V2        if you use the arduino library wire.h    but it does not work with the IDF library driver/i2c.h

             

              While the I2C works with the board Adafruit  feather ESP32    with both drivers     the arduino library wire.h     or the     IDF library driver/i2c.h

 

             I was using a  I2C scanner sketch

 

             I have no idea about the root cause

Enviado el: martes, 04 de octubre de 2022 13:43

PilnyTomas, 2022-10-05:
Hi, can you please try the second sketch with my modifications? https://github.com/PilnyTomas/issues/blob/main/issue_7252B/issue_7252B.ino
In this configuration, the DAC is working for me.
If it works, try to figure out where is the breaking point - you might be trying to use too high sampling frequency? Perhaps too small DMA buffer?
I will check the first sketch later.

JoseEnriqueFA, 2022-10-07:
> Enviado el: miércoles, 05 de octubre de 2022 21:10
>
>
>
> Congratulations !!!
>
>
>
> I was able to compile the received Sketch with Arduino Board Manager 2.0.5 and I can see a sinusoid in the oscilloscope.
>
>
>
> To be done from my side:
>
>    1- I have to study and understand the changes
>
>
>
>    2- sinusoid frequency range, with the introduced changes
>
>
>
>    2- Sporadically, I see some spikes on the oscilloscope.
>
>
>
>
>
> I will share more information in the following days.  It has been a great help.
>
>
>
> Best wishes,
>
> José
>
>
>
>
>
> Enviado el: miércoles, 05 de octubre de 2022 16:37

Hello,

 

Once again:   Thanks a lot for your support

 

 

After some investigation hours with the Adafruit Feather ESP32, I can share:

 

What's make the different between your proposal and my Sketch is the parameter

        .communication_format = I2S_COMM_FORMAT_STAND_MSB,     (  0x03)

 

This is key working with the Arduino  Board Manager  2.0.5  

In previous Board Manager versions, it was possible to use different initiations  for .communication_format  .. 

                                                                                                                                                 I2S_COMM_FORMAT_I2S_MSB  for the Board Manager 1.0.6

                                                                                                                                                 I2S_COMM_FORMAT_STAND_I2S for the Board Manager 2.0.0

 

One remark form my side    

        .use_apll = 1

       produces sporadic glitches with the Arduino  Board Manager  2.0.5   .   See:    https://github.com/espressif/esp-idf/issues/9321

 

      Alternatives:  

                      .use_apll = 0        (For my application  is fine)

      or

 

                     Arduino  Board Manager  1.0.6    .use_apll = 1    does not produce glitches

 

     Probably you can investigate or escalate.

 

 

Next steps from my side:

I will follow this test with the Adafruit Feather ESP32  V2

 

 

Best whishes

José

PilnyTomas, 2022-10-26:
Hi, If you need further assistance, please let me know. If the issue is solved, please close it.
If there is no response in 2 weeks I will consider this as solved close myself.
Thanks for understanding.

JoseEnriqueFA, 2022-10-26:
Hi,

 

Issue closed

 

Thanks a lot for the support

Enviado el: miércoles, 26 de octubre de 2022 8:58

closed 2022-10-26 by JoseEnriqueFA

laoadam, 2023-05-16:
> try the second sketch with my modifications?

Hi Tomas, 
sorry to raise up here,  I tested all sketches and got just straight line on my Oscilloscope hooked to Pin26, any thing wrong with that? 
BTW. can you show how to use your Library to make a ESP32 oscilloscope? 
Best regards.

VojtechBartoska, 2023-05-17:
cc @PilnyTomas