espressif/arduino-esp32 #7253: ESP32 T-BEAM static (precompiled) library in arduino-cli

oscarbaselga, 2022-09-14:
### Board

ESP32 T-BEAM

### Device Description

I am working with the ESP32 T-BEAM board.

### Hardware Configuration

-

### Version

v2.0.4

### IDE Name

Arduino IDE

### Operating System

Ubuntu 22.04

### Flash frequency

-

### PSRAM enabled

no

### Upload speed

-

### Description

Hi there!

I am attempting to build a sketch using a static library `.a`.
I followed this post: [Arduino Tutorials - How to Use Static (Pre-compiled) Libraries in Arduino - Seeed Wiki](https://wiki.seeedstudio.com/Software-Static-Library/) but I got an error during compilation:
```
Library PrebuiltLib has been declared precompiled:
Using precompiled library in /home/oscar/Arduino/libraries/PrebuiltLib/src/esp32
/tmp/arduino-sketch-9595712C3FF3F4D274F59DD186AB57D9/sketch/src/sensors/SensorA.cpp.o:(.literal._ZN11SensorA3runEb+0x18): undefined reference to `suma'
/tmp/arduino-sketch-9595712C3FF3F4D274F59DD186AB57D9/sketch/src/sensors/SensorA.cpp.o: In function `SensorA::run(bool)':
/home/oscar/App/src/sensors/Sensor.hpp:31: undefined reference to `suma'
collect2: error: ld returned 1 exit status

Error during build: exit status 1
```
My static library is called PrebuiltLib and has been built for the `esp32:esp32:t-beam` board. Its source code is:

- `Prebuilt.h`
```
#ifndef PrebuiltLib_H
#define PrebuiltLib_H

#ifdef __cplusplus
extern "C" {
#endif

int suma (int a, int b);

#ifdef __cplusplus
}
#endif

#endif // PrebuiltLib_H
```
- `Prebuilt.c`
```
#include "PrebuiltLib.h"

int suma (int a, int b) {
	return a +  b;
}
```
My arduino-cli version is `0.18.3 commit: d710b642`.

Oscar

### Sketch

```cpp
#include <PrebuiltLib.h>

void setup() {
  // put your setup code here, to run once:

}

void loop() {
  // put your main code here, to run repeatedly:

}
```

### Debug Message

```plain
Library PrebuiltLib has been declared precompiled:
Using precompiled library in /home/oscar/Arduino/libraries/PrebuiltLib/src/esp32
/tmp/arduino-sketch-9595712C3FF3F4D274F59DD186AB57D9/sketch/src/sensors/SensorA.cpp.o:(.literal._ZN11SensorA3runEb+0x18): undefined reference to `suma'
/tmp/arduino-sketch-9595712C3FF3F4D274F59DD186AB57D9/sketch/src/sensors/SensorA.cpp.o: In function `SensorA::run(bool)':
/home/oscar/App/src/sensors/Sensor.hpp:31: undefined reference to `suma'
collect2: error: ld returned 1 exit status

Error during build: exit status 1
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

VojtechBartoska, 2022-09-19:
@me-no-dev Can you please take a look on this?

me-no-dev, 2022-09-19:
Can this be tested with 2.0.5? I think I added the necessary link flag for such external libs at some point. If still does not work, I'll check what could be wrong again. Just FYI, such libs can work only against specific target, meaning if the lib is compiled for ESP32, it will not work on S2, S3 or C3.

me-no-dev, 2022-09-19:
Also the logs from verbose compilation are needed.

oscarbaselga, 2022-09-19:
Hi @me-no-dev, I get an error when compiling with 2.0.5:
```
In file included from /home/oscar/Arduino/hardware/espressif/esp32/tools/sdk/esp32/include/hal/include/hal/dac_types.h:4:0,
                 from /home/oscar/Arduino/hardware/espressif/esp32/tools/sdk/esp32/include/driver/include/driver/dac_common.h:16,
                 from /home/oscar/Arduino/hardware/espressif/esp32/cores/esp32/esp32-hal-dac.c:22:
/home/oscar/Arduino/hardware/espressif/esp32/tools/sdk/esp32/include/hal/include/hal/adc_types.h:90:27: error: expected ',' or '}' before '__attribute__'
     ADC_DIGI_FORMAT_12BIT __attribute__((deprecated)),  /*!<ADC to DMA data format,                [15:12]-channel, [11: 0]-12 bits ADC data (`adc_digi_output_data_t`). Note: For single convert mode. */
                           ^
In file included from /home/oscar/Arduino/hardware/espressif/esp32/tools/sdk/esp32/include/driver/include/driver/adc.h:14:0,
                 from /home/oscar/Arduino/hardware/espressif/esp32/cores/esp32/esp32-hal-adc.c:16:
/home/oscar/Arduino/hardware/espressif/esp32/tools/sdk/esp32/include/hal/include/hal/adc_types.h:90:27: error: expected ',' or '}' before '__attribute__'
     ADC_DIGI_FORMAT_12BIT __attribute__((deprecated)),  /*!<ADC to DMA data format,                [15:12]-channel, [11: 0]-12 bits ADC data (`adc_digi_output_data_t`). Note: For single convert mode. */
                           ^
exit status 1
Error compiling for board T-Beam.
```

oscarbaselga, 2022-09-20:
It worked with v2.0.5, thank you!