espressif/arduino-esp32 #7262: ESP32 save crash to LittleFs in a file?

zekageri, 2022-09-16:
### Related area

Crash

### Hardware specification

ESP32

### Is your feature request related to a problem?

On esp8266 we have a saveCrash library. [Github link](https://github.com/krzychb/EspSaveCrash)

On ESP32 we can not see where and why it crashed if it is on a remote location.

### Describe the solution you'd like

I would like a function which would take a callback which would be called with the crash log.
Something like this:

```C
void setup(){
    esp_addCrashCallback( []( const char* crashLog ){
         // Do whatever i like with crashLog, for example save to LittleFs.
    });
}
```

### Describe alternatives you've considered

There is a guy who tried this with custom partition table and things like that.
But hes implementation only works with 1.0.6 core.

[https://github.com/MathieuDeprez/ESP32_CoreDump_Arduino_1.0.6](https://github.com/MathieuDeprez/ESP32_CoreDump_Arduino_1.0.6)

### I have checked existing list of Feature requests and the Contribution Guide

- [X] I confirm I have checked existing list of Feature requests and Contribution Guide.

trullock, 2023-09-18:
Has there been any progress on this? Are there any external libraries that facilitate this?

me-no-dev, 2023-09-18:
yes. We are adding support for custom panic handler for Arduino 3.0.0, but it will not be able to write to flash or any other storage. Option would be to save it to RTC RAM, reboot and then do what you want with the crash (given you do not crash on boot)

trullock, 2023-09-18:
is there any even beta code in the works I can look at ahead of the release? Thanks

trullock, 2023-09-18:
@zekageri how are you writing the core dump? thanks

zekageri, 2023-09-18:
Iam not sure but i think the core writes to that partition automatically if it finds it. I read that we have to enable it in the menu config so it might be a false info. What i read it could be garbage.

trullock, 2023-09-18:
OK thanks I'll go on a hunt and report back

Parsaabasi, 2025-01-14:
Hello, 

This issue seems resolved. In case it persists, please feel free to reopen it. 

Thanks

closed 2025-01-14 by Parsaabasi

saiteja0990, 2025-01-23:
Hi, how can we read only crash data in arduino coredump... 
in espressif ide it has "esp_err_t esp_core_dump_image_get(size_t* out_addr, size_t *out_size);" this function to read only crash data instead of reading full partition, but this function wont support in arduino  do we have any alternative method to read only crash data and store it in littlefs file 
any one......