espressif/arduino-esp32 #7265: ESP32-C3FH Wifi Tx (STA and SoftAp) fail to connect or generate network

axibo-reiner, 2022-09-17:
### Board

ESP32-C3FH

### Device Description

Custom hardware

### Hardware Configuration

- using USB CDC on the ESP32C3 device ( have also tested with the device disabled)
- noise on 3.3v during TX ~40mVpp
- hardware matches what is required in the design guidelines for capacitance  (see schematic), decoupling caps are on a different sheet)
- oscillator tested and working
- Antenna was verified by looking at RSSI scans of networks located ~50ft away. 
[mdr2.pdf](https://github.com/espressif/arduino-esp32/files/9592147/mdr2.pdf)

### Version

v2.0.4

### IDE Name

platformio and Arduino IDE 2.0 (Both tested)

### Operating System

Ubuntu 22

### Flash frequency

160Mhz

### PSRAM enabled

no

### Upload speed

460800

### Description

The ESP32C3 seemingly unable to do anything related to TXing. The device cannot host an AP or connect to any network in station mode. 

When running a scan program it is capable of scanning networks and seeing them clearly with strong RSSI. However when connecting to those networks the connection fails. 

### Sketch

```cpp
#include <Arduino.h>
#include "esp32-hal-cpu.h"
#include "WiFi.h"

#define Serial USBSerial

const char* ssid = "secret";
const char* password = "secret";

void initWiFi() {
 delay(4000);
  WiFi.mode(WIFI_STA);
    WiFi.disconnect();
  WiFi.begin(ssid, password);
  Serial.print("Connecting to WiFi ..");
  while (WiFi.status() != WL_CONNECTED) {
    Serial.print('.');
    delay(1000);
  }
  Serial.println(WiFi.localIP());
}

void setup() {
  Serial.begin(115200);
  initWiFi();
  Serial.print("RRSI: ");
  Serial.println(WiFi.RSSI());
}

void loop(){

}
```

### Debug Message

```plain
In station mode:

[ 37680][D][WiFiGeneric.cpp:929] _eventCallback(): Arduino Event: 5 - STA_DISCONNECTED
[ 37681][W][WiFiGeneric.cpp:950] _eventCallback(): Reason: 2 - AUTH_EXPIRE
[ 37682][D][WiFiGeneric.cpp:975] _eventCallback(): WiFi AutoReconnect Running
```

In AP Mode:
[  4142][D][WiFiGeneric.cpp:929] _eventCallback(): Arduino Event: 0 - WIFI_READY
[  4176][V][WiFiGeneric.cpp:390] _arduino_event_cb(): AP Started
[  4177][D][WiFiGeneric.cpp:929] _eventCallback(): Arduino Event: 10 - AP_START
```

### Other Steps to Reproduce

2.0.5 and 2.0.3 where also tested. 

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

Jason2866, 2022-09-18:
Have you tried a known good bought c3 device? Cant reproduce your issue.
If the antenna matching is not good, the result is what you have.
The Lolin C3 is a example of a commercial device for this.

Jason2866, 2022-09-22:
[Here](https://github.com/espressif/arduino-esp32/issues/7232#issuecomment-1249120593) is a know working sketch

LoopOnTech, 2023-01-21:
Take a look at reducing the transmit power after WiFi has started.

The transmit power mappings can be found [here](https://github.com/espressif/arduino-esp32/blob/23d715af1b185fac21558c907af4ff5b86526aa4/libraries/WiFi/src/WiFiGeneric.h#L111).

`    Serial.println(WiFi.softAP("espsoftap", "12345678")); // no errors that I can see, but ap never shows up
    WiFi.setTxPower(WIFI_POWER_15dBm);
    int a = WiFi.getTxPower();
    Serial.print("TX power: ");
    Serial.println(a);
`
Check the getTxPower value to confirm that the setting is being applied.

closed 2025-01-16 by Parsaabasi