espressif/arduino-esp32 #7267: Support littleFS and fatFS for UpdateClass

vovagorodok, 2022-09-18:
### Related area

OTA updates

### Hardware specification

ESP32 DevKitC

### Is your feature request related to a problem?

I have created simple platformio library for update devices via Bluetooth LE:\
https://github.com/vovagorodok/ArduinoBleOTA \
~3kB/s upload speed achieved with SPIFFS.\
Here https://github.com/fbiego/ESP32_BLE_OTA_Arduino was made some tests, where ~12kB/s is achieved with fatFS (and firmware update take 1.2min instead of 5).\
My implementation bases on ArduinoOTA library for Wifi updates. ArduinoOTA use https://github.com/espressif/arduino-esp32/blob/master/libraries/Update/src/Update.h for that. As i see there no possibility to update software for fatFS/littleFS.
Any  plans to add it?

### Describe the solution you'd like

As I correctly understand here:
```
#define U_FLASH   0
#define U_SPIFFS  100
#define U_AUTH    200
```
can be added `U_FFAT` and `U_LITTLEFS` and handled in `UpdateClass::begin`

### I have checked existing list of Feature requests and the Contribution Guide

- [X] I confirm I have checked existing list of Feature requests and Contribution Guide.

me-no-dev, 2022-09-19:
Updater actually already works for FATFS. `U_SPIFFS` will try to get SPIFFS or FAT partition [here](https://github.com/espressif/arduino-esp32/blob/master/libraries/Update/src/Updater.cpp#L135-L146). I do not recall if LittleFS uses different partition spec, but if so, it just needs to be added in the same check and from then on the code will upload to that partition the image you have provided.

vovagorodok, 2022-09-19:
Tests made on FFAT (https://github.com/espressif/arduino-esp32/tree/master/libraries/FFat), not FatFS. There is a difference? FFAT and FatFS is the same?

vovagorodok, 2022-09-19:
This offset is correct ?
https://github.com/espressif/arduino-esp32/blob/2d6ca351a3b35fe0ce5d427a469e49ed7892c25a/libraries/Update/src/Updater.cpp#L140

As I see `default_ffat.csv` and `default.csv` has the same address maping

me-no-dev, 2022-09-20:
@lbernstone do you remember what was that offset for?

lbernstone, 2022-09-20:
It is how wear levelling works.  There's 1k offset to hold the usage bitmap.

vovagorodok, 2022-09-20:
Success, works fine. Bottleneck was on bluetooth communication (switch between Tx to Rx takes a long time). After changing protocol a bit ~12kB/s was achieved. Lets close issue

closed 2022-09-21 by vovagorodok

szerwi, 2022-12-25:
I believe this issue should be reopened, as it seems that LittleFS is not supported.
After sending the LittleFS image, I get error when trying to mount it:
```
./components/esp_littlefs/src/littlefs/lfs.c:1229:error: Corrupted dir pair at {0x0, 0x1}
E (22483) esp_littlefs: mount failed,  (-84)
E (22484) esp_littlefs: Failed to initialize LittleFS
```

Sending the same image using LittleFS plugin (https://github.com/lorol/arduino-esp32littlefs-plugin) works well.

lbernstone, 2022-12-25:
You should open a new issue since this had to do with adding different file types for upload, rather than fs failure, and OP closed the issue once fixed.