espressif/arduino-esp32 #7268: ESP32-S3 WIFI START-STOP-START KO with ESPAsyncWebServer (OK with ESP32)

06GitHub, 2022-09-19:
### Board

ESP32-S3

### Device Description

ESP32-S3 : AI Thinker ESP-S3-12K  FLASH 4Mb PSRAM 8Mb

Test has also been done (not shown) with :
ESP32-S2 : SAOLA WROVER FLASH 4Mb PSRAM 8Mb
ESP32 : LILYGO T8 V.1.7.1 4Mb PSRAM 8Mb

### Hardware Configuration

Nothing is attached to development board, excepted USB cable between computer and development board (ie bare development board)

### Version

v2.0.4

### IDE Name

Arduino IDE

### Operating System

Windows 10

### Flash frequency

80 Mhz

### PSRAM enabled

yes

### Upload speed

115200

### Description

On ESP32-S3, **WIFI is **KO** when triggering **starting (#1), stopping and (re)starting (#2)** Wifi with ESPAsyncWebServer**.

Same KO behaviour for ESP32-S2.

**Note this issue does not occur on ESP32** : in other words, it is possible to start/stop/(re)start wifi on ESP32. Maybe there is a regression on ESP32-S3 and ESP32-S2, or the way to stop/(re)start Wifi has changed on ESP32-S3 and ESP32-S2 as compared to ESP32 ?

Such sequence is needed for project because user should be able to activate/deactivate Wifi on his/her own. Since Wifi consumes resources/battery, Wifi should be off when/if user does not need it. Project relies on batteries for power.

On ESP32-S3 (and ESP32-S2), **the only way to reactivate Wifi after (re)start #2 is to reboot**, which is not user friendly.

**Tests have also been done with Wifi Station (not shown)** : same KO behaviour for ESP32-S3 and ESP32-S2, and OK behaviour for ESP32. So it seems that issue is not due to Wifi Access Point vs Wifi Station (since same issue occurs) but to ESP32-S3 and ESP32-S2.

**Screenshot 1**
[image]

**Screenshot 2**
[image]

### Sketch

```cpp
See sketch and html page below
------
SKETCH
------
// Wifi issue "start stop start",  esp32s3 ko, esp32s2 ko, esp32 ok, Sep 19 2022

// Debugging
[... 232 lines cut ...]
[ 91284][D][WiFiGeneric.cpp:929] _eventCallback(): Arduino Event: 10 - AP_START
[ 91286][D][ESP32-S3_wifistartstopstart_test.ino:119] wifiStart(): [TEST] Wifi : Init OK
[ 91294][I][ESP32-S3_wifistartstopstart_test.ino:121] wifiStart(): [TEST] softAPSSID: ESP32_AP
[ 91302][I][ESP32-S3_wifistartstopstart_test.ino:122] wifiStart(): [TEST] softAPIP: 192.168.4.1
[ 91310][I][ESP32-S3_wifistartstopstart_test.ino:123] wifiStart(): [TEST] mode: 2
[ 91318][I][ESP32-S3_wifistartstopstart_test.ino:163] loop(): [TEST] ----------------- :-/ ------------------->
[ 91327][D][ESP32-S3_wifistartstopstart_test.ino:164] loop(): [TEST] After Wifi start #2, ISSUE : Web page is no longer reachable on Wifi Access Point for ESP32-S3 (Wifi network no longer visible)
[ 91345][D][ESP32-S3_wifistartstopstart_test.ino:165] loop(): [TEST] See screenshot #2
[ 91352][D][ESP32-S3_wifistartstopstart_test.ino:166] loop(): [TEST] Not shown : ESP32-S2 is also KO, but ESP32 is OK
[ 91363][D][ESP32-S3_wifistartstopstart_test.ino:167] loop(): [TEST] Not shown : Same issue with Wifi Station
[ 91372][I][ESP32-S3_wifistartstopstart_test.ino:168] loop(): [TEST] ----------------------------------------->
[ 91382][D][ESP32-S3_wifistartstopstart_test.ino:171] loop(): [TEST] loop ...
[ 96389][D][ESP32-S3_wifistartstopstart_test.ino:171] loop(): [TEST] loop ...
[101389][D][ESP32-S3_wifistartstopstart_test.ino:171] loop(): [TEST] loop ...
```

### Other Steps to Reproduce

As stated above, issue :
- does **not** occur with ESP32
- occurs also with Wifi Station

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

Jason2866, 2022-09-19:
Is ESPAsyncWebServer really working with other MCUs than ESP32.
https://github.com/me-no-dev/ESPAsyncWebServer/issues/951

06GitHub, 2022-09-19:
As per my usage, ESPAsyncWebServer works nicely on ESP32-S2 and ESP32-S3, as long as one does not START-STOP-(RE)START WIFI.
For instance, START and STOP-START WIFI sequence behaves OK, ESPAsyncWebServer is up and running and pages are displayed correctly (no "Connection has timed out" error)

Jason2866, 2022-09-19:
@06GitHub Your sketch with a S3 (AI Thinker ESP-S3-12K FLASH 8Mb PSRAM 8Mb)
[image]

06GitHub, 2022-09-19:
@Jason2866  Thanks for spending some time to check this :-)

> Ahh, yes. It is offline after stop / start

So same issue occurs on your side, good information !

Jason2866, 2022-09-29:
Imho the ESPAsyncWebServer needs to be reviewed. There are open issues and unmerged PRs for.

mrengineer7777, 2022-10-14:
ESPAsyncWebServer is brilliant work.  Sadly it appears to have been abandoned by the owner.

me-no-dev, 2022-10-14:
hopefully soon

06GitHub, 2023-09-02:
**ESP32-S3 WIFI START-STOP-START is now OK** as per test (re)done today Sep 2 2023, see configuration below.

Wifi is up again after START-STOP-START sequence, no need to reboot ESP32-S3 after STOP.

Problem was encountered about 1 year ago with "arduino-esp32 core" version **2.0.4**. Behaviour is now OK with "arduino-esp32 core" version **2.0.9**.

=> **Issue Closed**

**Configuration**
```
MCU: ESP32-S3-WROOM-1-N16R8 (Flash 16Mb PSRAM 8Mb)
Arduino ESP32 Core :  v2.0.9
Arduino IDE : 1.8.19
Windows 10
```

closed 2023-09-02 by 06GitHub