espressif/arduino-esp32 #7283: ESP32 always reset

savagerex, 2022-09-21:
### Board

ESP32 S3

### Device Description

ESP32-S3-DevKitC-1

### Hardware Configuration

ESP32S3 + sensor(iis3dwb)

### Version

v2.0.4

### IDE Name

Arduino IDE

### Operating System

Windows 10

### Flash frequency

80Mhz

### PSRAM enabled

no

### Upload speed

921600

### Description

ESP32 always reset.

### Sketch

```cpp
#include "SPI.h"
#include "BoardConfig.h"
#include "Sensor.h"
#include "IIS3DWB.h"
#include <WiFi.h>

#if CONFIG_FREERTOS_UNICORE
#define ARDUINO_RUNNING_CORE 0
#else
#define ARDUINO_RUNNING_CORE 1
#endif

const char* ssid = "vivo Y21";  //SW1-WIFI6
const char* pass = "1234567890";   //wireless

const uint16_t port = 8234;
const char * host = "192.168.11.124";

WiFiClient client;

const char Version[] = "1.0.0";
const char CompileDate[] = __DATE__ " " __TIME__;
#define CSPIN    10

uint8_t Ascale = AFS_2G;

float aRes;                              // scale resolutions per LSB for the accel
float accelBias[3] = {0.0f, 0.0f, 0.0f}; // offset biases for the accel
int16_t IIS3DWBData[4] = {0};            // Stores the 16-bit signed sensor output
uint8_t IIS3DWBRawData[6] = {0};            // Stores the 16-bit signed sensor output
float ax, ay, az, accelTemp;             // variables to hold latest accel data values
uint8_t IIS3DWBstatus;
//long values[40];
uint8_t ui8txLogBuf[20];
uint16_t i, j, k, m, n;
uint16_t mm;
volatile bool IIS3DWB_DataReady = false;
int packet_counter = 1;

unsigned long SpiStartTime;
unsigned long SpiEndTime;

unsigned long UartStartTime;
unsigned long UartEndTime;

xQueueHandle xQueue;

IIS3DWB IIS3DWB(CSPIN);

// Pins used for the connection with the sensor
const int CHIP_SELECT_PIN = 10;  //7

uint8_t ascii_trans(uint8_t ui8Code) {
  if ((ui8Code == 0 || ui8Code > 0) && ui8Code <= 9) {
    ui8Code += 48;  //0x30;
  } else if (ui8Code >= 10 && ui8Code <= 15) {
    ui8Code += 55;  //0x37;
  }
  return ui8Code;
}

void IRAM_ATTR IntOneIsr() {
  IIS3DWB_DataReady = true;
}

void setup() {
  // put your setup code here, to run once:
  Serial.begin(115200);
  Serial1.begin(4608000, SERIAL_8N1, 20, 19);

  SPI.begin(GPIO_OUTPUT_SCLK, GPIO_INPUT_MISO, GPIO_OUTPUT_MOSI, GPIO_OUTPUT_CS);
  //SPI.beginTransaction(SPISettings(spiClk, MSBFIRST, SPI_MODE0));

  // Initalize the  data ready and chip select pins:
  pinMode(CSPIN, OUTPUT);
  digitalWrite(CSPIN, HIGH);

  Serial.println("IIS3DWB accel...");
  uint8_t c = IIS3DWB.getChipID();  // Read CHIP_ID register for IIS3DWB
  Serial.print("IIS3DWB "); Serial.print("I AM "); Serial.print(c, HEX); Serial.print(" I should be "); Serial.println(0x7B, HEX);
  Serial.println(" ");

  if (c == 0x7B) // check if all SPI sensors have acknowledged
  {
    Serial.println("IIS3DWB is online...");
    Serial.println(" ");

    // reset IIS3DWB to start fresh
    IIS3DWB.reset();

    // get accel sensor resolution, only need to do this once
    aRes = IIS3DWB.getAres(Ascale);

    IIS3DWB.selfTest();

    IIS3DWB.init(Ascale); // configure IIS3DWB

    IIS3DWB.offsetBias(accelBias);
    Serial.println("accel biases (mg)"); Serial.println(1000.0f * accelBias[0]); Serial.println(1000.0f * accelBias[1]); Serial.println(1000.0f * accelBias[2]);
    Serial.println(" ");
    delay(1000);
  }
  else
  {
    if (c != 0x6A) Serial.println(" IIS3DWB not functioning!");
    while (1) {};
  }

  // Give the sensor time to set up:
  delay(100);
  attachInterrupt(INTONE, IntOneIsr, RISING);

  delay(10);
    WiFi.begin(ssid,pass);

    Serial.print("Connecting.");
    while(WiFi.status() != WL_CONNECTED) {
        Serial.print(".");
        delay(500);
    }

    Serial.print("WiFi connected - IP address: ");
    Serial.println(WiFi.localIP());

  xQueue = xQueueCreate(100, sizeof(IIS3DWBRawData));

  xTaskCreatePinnedToCore(taskOne, "TaskOne", 10000, NULL, 1, NULL, 0);
  xTaskCreatePinnedToCore(taskTwo, "TaskTwo", 10000, NULL, 5, NULL, 1);
}

void taskOne( void * parameter ) {
  disableCore0WDT();
  for (;;) {
    if (IIS3DWB_DataReady) {
      IIS3DWB_DataReady = false;
      if (m == 0) {
        SpiStartTime = millis();
      }

      IIS3DWB.readAccelRawData(IIS3DWBRawData);

      xQueueSend(xQueue, &IIS3DWBRawData, 0);

      m++;
      if (m == 26667) {
        SpiEndTime = millis();
        Serial.print("**************SPI Sensor Time*************");
        Serial.println(SpiEndTime - SpiStartTime);
        m = 0;
      }
    }
  }
  vTaskDelete( NULL );
}

void taskTwo( void * parameter ) {
  uint8_t xQueueData[6] = {0};
  
  for (;;) {
    if (xQueueReceive(xQueue, (void *)&xQueueData, 0) == pdTRUE) {
      if (mm == 0) {
        UartStartTime = millis();
      }

      for (i = 0; i < 6; i++) {
        ui8txLogBuf[j] = ascii_trans((xQueueData[i] >> 4));
        ui8txLogBuf[j + 1] = ascii_trans((xQueueData[i] & 0x0F));
        j += 2;
      }

      if (client.connect(host, port)){
          if (packet_counter < 10) {
            client.write(ui8txLogBuf, 12);
          } else if (packet_counter == 10) {
            ui8txLogBuf[j] = '\r';
            ui8txLogBuf[j + 1] = '\n';
            client.write(ui8txLogBuf, 14);
            //m += 1;
            //j = 0;
            packet_counter = 0;
          }
          packet_counter++;
          j = 0;
      
      //if (packet_counter < 10) {
      //  Serial1.write("testtesttest");
      //} else if (packet_counter == 10) {
      //  Serial1.write("testtesttest\r\n");
      //  packet_counter = 0;
      //}
      //packet_counter++;

      mm++;
      if (mm == 26667) {
        UartEndTime = millis();
        Serial.print("**************UART Time*************");
        Serial.println(UartEndTime - UartStartTime);
        mm = 0;
        }
      }
    }
  }
  vTaskDelete( NULL );
}

void loop() {
}
```

### Debug Message

```plain
Connecting..WiFi connected - IP address: 192.168.11.124
Guru Meditation Error: Core  1 panic'ed (IllegalInstruction). Exception was unhandled.

Core  1 register dump:
PC      : 0x29782528  PS      : 0x00060830  A0      : 0x82003067  A1      : 0x3fca0780  
A2      : 0x3fc98a24  A3      : 0x3c0802d9  A4      : 0x0000202a  A5      : 0x00000028  
A6      : 0x00000000  A7      : 0x00000000  A8      : 0x8206fdec  A9      : 0x3fca0760  
A10     : 0x3fc98a24  A11     : 0x3c0802d9  A12     : 0x0000202a  A13     : 0x00000bb8  
A14     : 0x00060820  A15     : 0x00000001  SAR     : 0x0000000a  EXCCAUSE: 0x00000000  
EXCVADDR: 0x00000000  LBEG    : 0x40056f5c  LEND    : 0x40056f72  LCOUNT  : 0xffffffff  

Backtrace:0x29782525:0x3fca0780 |<-CORRUPTED

ELF file SHA256: 0000000000000000

Rebooting...
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0xc (RTC_SW_CPU_RST),boot:0x2b (SPI_FAST_FLASH_BOOT)
Saved PC:0x42002d1a
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fce3808,len:0x43c
load:0x403c9700,len:0xbec
load:0x403cc700,len:0x2a3c
SHA-256 comparison failed:
Calculated: 3611f330726ad4edf64f62982b0568332a5800bc59425cab90be5379db92c108
Expected: ffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffff
Attempting to boot anyway...
entry 0x403c98d8
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

Jason2866, 2022-09-21:
This is not needed and useless.
```
#if CONFIG_FREERTOS_UNICORE
#define ARDUINO_RUNNING_CORE 0
#else
#define ARDUINO_RUNNING_CORE 1
#endif
```

Check your used Gpios, you are using not available ones.

savagerex, 2022-09-22:
ESP32S3 still always reset.

SuGlider, 2022-09-22:
@savagerex - would you be able to isolate the part of the code that generates the `IllegalInstruction`?
It sounds different because it seems like you may be linking some code from another SoC into the software.

SuGlider, 2022-09-22:
@savagerex - I think that the issue is in this line:
```
#define CSPIN    10
```

GPIO10 is used for QIO Flash access.
This causes the `IllegalInstruction` reset.
You can use DIO instead in order to release GPIO 9 and 10 for being used in the sketch or change it to some other GPIO.

savagerex, 2022-09-22:
it's work. thank you!!

SuGlider, 2022-09-22:
ESP32 uses GPIO6, GPIO7, GPIO8, GPIO9, GPIO10, GPIO11 to access the flash in QIO (default). 
Those pins should not be used by the sketch.
Only when selected DIO mode in the Arduino IDE, it is possible to use GPIO9 and GPIO10.

SuGlider, 2022-09-22:
> it's work. thank you!!

Great! Can we close the issue?

savagerex, 2022-09-22:
ok

closed 2022-09-23 by me-no-dev