espressif/arduino-esp32 #7293: Linker cannot find function `uint32_t gpio_input_get_high(void);`

codewitch-honey-crisis, 2022-09-23:
### Board

esp32s3

### Device Description

I think it applies to anything on the esp32s3 line. Maybe the S2 as well? Though I haven't tested for that.

### Hardware Configuration

I'm using a Lilygo T-Display S3. It is an ESP32S3 with an attached ST7789 via an 8bit parallel i80 interface.

### Version

v2.0.3

### IDE Name

PlatformIO

### Operating System

Windows 11

### Flash frequency

40mhz

### PSRAM enabled

no

### Upload speed

921600

### Description

(framework-arduinoespressif32 @ 3.20003.220626 (2.0.3))

function uint32_t gpio_input_get_high(void);
in esp32s3/rom/gpio.h

is not found by the linker, as though it is not being compiled in.

### Sketch

```cpp
#include <Arduino.h>
#include "esp32s3/rom/gpio.h"
void setup() {
    Serial.begin(115200);
    Serial.println(gpio_input_get_high());
}
void loop() {
}
```

### Debug Message

```plain
CONFIGURATION: https://docs.platformio.org/page/boards/espressif32/esp32-s3-devkitc-1.html
PLATFORM: Espressif 32 (5.0.0+sha.d92c478) > Espressif ESP32-S3-DevKitC-1-N8 (8 MB QD, No PSRAM)
HARDWARE: ESP32S3 240MHz, 320KB RAM, 8MB Flash
DEBUG: Current (esp-builtin) On-board (esp-builtin) External (cmsis-dap, esp-bridge, esp-prog, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa)
PACKAGES:
 - framework-arduinoespressif32 @ 3.20003.220626 (2.0.3)
 - tool-esptoolpy @ 1.30300.0 (3.3.0)
 - tool-mkfatfs @ 2.0.1
 - tool-mklittlefs @ 1.203.210628 (2.3)
 - tool-mkspiffs @ 2.230.0 (2.30)
 - toolchain-riscv32-esp @ 8.4.0+2021r2-patch3
 - toolchain-xtensa-esp32s3 @ 8.4.0+2021r2-patch3
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ deep, Compatibility ~ soft
Found 42 compatible libraries
Scanning dependencies...
Dependency Graph
|-- htcw_st7789 @ 1.1.6
|   |-- htcw_gfx @ 1.3.3
|   |   |-- htcw_bits @ 1.0.6
|   |   |-- htcw_data @ 1.0.6
|   |   |-- htcw_io @ 1.1.3
|   |   |   |-- htcw_bits @ 1.0.6
|   |   |   |-- FS @ 2.0.0
|   |   |   |-- SD @ 2.0.0
|   |   |   |   |-- FS @ 2.0.0
|   |   |   |   |-- SPI @ 2.0.0
|   |-- htcw_tft_io @ 1.1.3
Building in release mode
Compiling .pio\build\T-Display-S3\src\main.cpp.o
Linking .pio\build\T-Display-S3\firmware.elf
c:/users/gazto/.platformio/packages/toolchain-xtensa-esp32s3@8.4.0+2021r2-patch3/bin/../lib/gcc/xtensa-esp32s3-elf/8.4.0/../../../../xtensa-esp32s3-elf/bin/ld.exe: .pio\build\T-Display-S3\src\main.cpp.o:(.literal._Z5setupv+0x8): undefined reference to `gpio_input_get_high'
c:/users/gazto/.platformio/packages/toolchain-xtensa-esp32s3@8.4.0+2021r2-patch3/bin/../lib/gcc/xtensa-esp32s3-elf/8.4.0/../../../../xtensa-esp32s3-elf/bin/ld.exe: .pio\build\T-Display-S3\src\main.cpp.o: in function `setup()':
F:\Users\gazto\Documents\PlatformIO\Projects\tdisplay_gfx/src/main.cpp:4: undefined reference to `gpio_input_get_high'
collect2.exe: error: ld returned 1 exit status
*** [.pio\build\T-Display-S3\firmware.elf] Error 1
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

P-R-O-C-H-Y, 2022-10-14:
Hi @codewitch-honey-crisis, this function is not available for esp32c3 and esp32s3. Esp-idf linker script does not provide address for this function. But you don't really need this function. You can use the `digitalRead(pin)`, it will be even more faster than the ROM functions.
Is there any specific reason you want to use `gpio_input_get_high()`?

codewitch-honey-crisis, 2022-10-14:
My attempts to use digitalRead on these pins has been unsuccessful. For context, I'm driving an i8080 bus connected to an LCD in 8 bit parallel. I have generic code for doing this using the arduino framework but it does not work.

The only way I've found to get it work is read a whole bank of pins at once. gpio_input_get_low() works, so now on the S3 how do i get the high pins as a uint32_t, or in the alternative, get ALL the pins as a uint64_t?

P-R-O-C-H-Y, 2022-10-17:
For S3 and C3 you can use `uint32_t gpio_input_get(void);` to read all input pins.

codewitch-honey-crisis, 2022-10-17:
Looking for that function on google, because according to your prototype of it, it does not do what I need, and only returns the first 32 pins.

I need pins 33-64 (or 33 to whatever the high end is)

Your function returns an uint32_t. It would have to be uint64_t to return the data I need.

codewitch-honey-crisis, 2022-10-18:
Thanks! I’ll give that a try. From: Jan ProcházkaSent: Tuesday, October 18, 2022 4:00 AMTo: espressif/arduino-esp32Cc: honey the codewitch; MentionSubject: Re: [espressif/arduino-esp32] Linker cannot find function `uint32_t gpio_input_get_high(void);` (Issue #7293) As it seems there is no option for reading the higher pins 32+, I found another solution which may be suitable for you.You may try using DEDICATED GPIO BUNDLES (link to docs), you can give it a try.I have created a simple sketch using it for pins 4,5,6,7 and it worked perfectly. Can you test that out for your needs?#include "driver/dedic_gpio.h" dedic_gpio_bundle_handle_t bundleA = NULL; void setup() {  // put your setup code here, to run once:  Serial.begin(115200);   // configure GPIO  const int bundleA_gpios[] = {4, 5, 6, 7};  gpio_config_t io_conf = {      .mode = GPIO_MODE_INPUT,  };   for (int i = 0; i < sizeof(bundleA_gpios) / sizeof(bundleA_gpios[0]); i++) {    io_conf.pin_bit_mask = 1ULL << bundleA_gpios[i];    gpio_config(&io_conf);  }   // Create bundleA, input only   dedic_gpio_bundle_config_t bundleA_config = {    .gpio_array = bundleA_gpios,    .array_size = sizeof(bundleA_gpios) / sizeof(bundleA_gpios[0]),    .flags = {      .in_en = 1,    },  };      dedic_gpio_new_bundle(&bundleA_config, &bundleA);} void loop() {  Serial.println(dedic_gpio_bundle_read_in(bundleA));  delay(500);}

P-R-O-C-H-Y, 2022-11-15:
@codewitch-honey-crisis Hello, did you give it a try? Did it work for your case? If yes, we can close this issue as solved.

codewitch-honey-crisis, 2022-11-15:
Oh gosh, i have not yet. Work beckoned me and then I got busy and forgot all about it. I'm kind of buried right now so I'm not sure when I can get to it. How do you want to handle this? I'm not sure when I'll be able to circle back. Maybe this weekend?

P-R-O-C-H-Y, 2022-11-15:
@codewitch-honey-crisis Thanks for quick reply. When you have same free time to test it will be great :) It does not hurry, there is no change pending for this. Asking just to know, if we can close the issue :)

codewitch-honey-crisis, 2022-11-17:
You may as well close this as my graphics libraries fail to compile under the latest Arduino framework bits because of a missing sdkconfig.h that I don't use and don't include explicitly. I can't run it down, and this S3 is the only 8-bit LCD interface I have. So I'm stuck.

codewitch-honey-crisis, 2022-11-17:
I'm stalled. I got the bundle implemented but I'm getting no display output. It might be a timing issue, and I'm in no mood to dig into this with a logic analyzer. I'm done.

P-R-O-C-H-Y, 2022-11-21:
IDK if I can help you with that. So I am closing this for now. If you work on it and still need any help feel free to reopen this issue. Or if you make it work, leave a comment here. So we can have it as reference for future issues.

closed 2022-11-21 by P-R-O-C-H-Y