espressif/arduino-esp32 #7294: Abort while updating firmware from SD card

ifengchao, 2022-09-23:
### Board

ESP32 DEV Module

### Device Description

ESP32 Board with sd card and 3.5 inch tft display.

### Hardware Configuration

sd card & tft 3.5 inch display

works fine, i can display images on screen stored in sd card.

### Version

latest master (checkout manually)

### IDE Name

pio

### Operating System

windows10

### Flash frequency

40

### PSRAM enabled

no

### Upload speed

1152000

### Description

**esp_partition_erase_range** aborted. Tried sdk version v4.4.1 and v4.4.2, both crashed.
Partitions below:
# Name,   Type, SubType, Offset,  Size, Flags
nvs,      data, nvs,     0x9000,  0x5000,
otadata,  data, ota,     0xe000,  0x2000,
app0,     app,  ota_0,   0x10000, 0x100000,
spiffs,   data, spiffs,  0x110000,0x2F0000,

### Sketch

```cpp
code in setup() function:
SD.begin(...);
File firmware =  SD.open(FW_FILE,"r");
Update.begin(firmware.size(), U_FLASH);
Update.writeStream(firmware);
```

### Debug Message

```plain
Below is the serial log:

fw found!
Try to update!
CALLBACK:  Update process at 0 of 1010672 bytes...
partition name app0, offset = 0x10000, 0x100000

abort() was called at PC 0x40082116 on core 1

Backtrace:0x40083801:0x3ffb26100x4008d445:0x3ffb2630 0x400929c9:0x3ffb2650 0x40082116:0x3ffb26d0 0x400f8403:0x3ffb2720 0x400ee0c5:0x3ffb2740 0x400ea1b6:0x3ffb2760 0x400ea43f:0x3ffb2780 0x400d6879:0x3ffb27a0 0x400d9fd3:0x3ffb27f0 0x400f1f3e:0x3ffb2820 

  #0  0x40083801:0x3ffb2610 in panic_abort at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/esp_system/panic.c:402
  #1  0x4008d445:0x3ffb2630 in esp_system_abort at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/esp_system/esp_system.c:128
  #2  0x400929c9:0x3ffb2650 in abort at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/newlib/abort.c:43 (discriminator 3)
  #3  0x40082116:0x3ffb26d0 in esp_flash_erase_region at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/spi_flash/esp_flash_api.c:541 (discriminator 5)
  #4  0x400f8403:0x3ffb2720 in esp_partition_verify at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/spi_flash/partition.c:390 (discriminator 1)
  #5  0x400ee0c5:0x3ffb2740 in WebServer::_parseRequest(WiFiClient&) at C:/Users/PP/.platformio/packages/framework-arduinoespressif32/libraries/WebServer/src/Parsing.cpp:181
  #6  0x400ea1b6:0x3ffb2760 in UpdateClass::_writeBuffer() at C:/Users/PP/.platformio/packages/framework-arduinoespressif32/libraries/Update/src/Updater.cpp:198
  #7  0x400ea43f:0x3ffb2780 in UpdateClass::writeStream(Stream&) at C:/Users/PP/.platformio/packages/framework-arduinoespressif32/libraries/Update/src/Updater.cpp:352
```

### Other Steps to Reproduce

1) put a file firmware.bin in the root of SD card.
2) power on the board.

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

ifengchao, 2022-09-25:
After some efforts, I found that it is related to the partition of flash.
If used the default partition table, firmware updating works fine. if i customize the partition table with 

> nvs, data, nvs, 0x9000, 0x5000,
otadata, data, ota, 0xe000, 0x2000,
app0, app, ota_0, 0x10000, 0x100000,
spiffs, data, spiffs, 0x110000,0x2F0000,

There's no app1 partition, firmware updating will abort in function **_writeBuffer** -> **esp_partition_erase_range** .

1) I wonder if we cloud not update "app0"  while we booted from "app0" ?
2) Hope someone knows and cloud tell me how to debug under the idf under PIO.
I have tried adding some debug code in  file the esp_flash_api.c located in ".platformio/packages/framework-espidf/components/spi_flash/esp_flash_spi.c", but it seem this file will not be compiled when i rebuild the project.

ifengchao, 2022-09-26:
I have tested web update via web broswer, the same problem.
Can someone give me a hint how can i debug the idf sdk under pio or arduino IDE ?

ifengchao, 2022-09-27:
After some googling, I started to reliased that the arduino-esp32 probably using the libspi_flash.a static library, so I can't debug it.
@me-no-dev Sorry to bother you, but cloud you give me some hints on this in your spare time. :)
Maybe i should compile libspi_flash.a myself?

me-no-dev, 2022-09-27:
If you want to update the firmware you need more than one OTA partition. It can not overwrite your current app

ifengchao, 2022-09-27:
> If you want to update the firmware you need more than one OTA partition. It can not overwrite your current app

Thank you very much for your help saving me a lot of time, I will look for other solutions.

closed 2022-09-27 by ifengchao